wmbusmeters/wmbusmeters #514: nanoCUL firmware for Techem

dzikiBOB, 2022-04-10:
Hi guys, 
Cannot find myself how to check firmware so could you please tell me if standard nanoCUL 868 usb stick from smarthomekomponente_de  need some firmware upgrade to read meter like here [https://i.ebayimg.com/images/g/51YAAOSwemVazO0~/s-l400.jpg](url)
It is Techem radio3

weetmuts, 2022-04-28:
The cul firmware is problematic, since it cannot handle full length telegrams. The standard firmware is limited
to <60 byte long telegrams, but there is a firmware improvement here: https://weetmuts.github.io/wmbusmeterswiki/nanoCUL.html that can handle slightly longer telegrams.

So, I do not know if it works with your meter. Try a cheap rtlsdr first, then if you want to reduce the cpu-load you can buy an im871a for example.

BIBOLV, 2023-01-02:
Closing issue due to inactivity.

closed 2023-01-02 by BIBOLV