wmbusmeters/wmbusmeters #518: Service needs restart regularly to work properly

BillGOH, 2022-04-22:
First of all, thank you very much for this great software!

I am using it on a Raspberry Pi 4B with openHABian (Raspberry Pi OS (bullseye)) and openHAB 3.2 with a NanoCUL USB Stick (https://www.smart-home-komponente.de/nano-cul/nano-cul-868-extern/#cc-m-product-8485019276). I can successfully read my Diehl Izar watermeter.

However, I discovered that WMBusmeters (ver. 1.7.0) stops working after resetting the stick every 23 hours. I therefore created a cronjob to restart the WMBusmeters service every 22 hours. Yet, it did not work reliably. I then created two cronjobs every 12 hours to restart the service. Now it seems to work.

Any ideas why the service seems to stop working after 12-23 hours? So far I could not discover anything suspicious in the log file (did not use the debug-level, however, because the file gets very large over a long period of time).

Thanks for help an ideas!

weetmuts, 2022-04-22:
There is a feature that by default restarts the underlying wmbus collection code every 23 hours.

From the README you can see that it can be set to a different value with: 
`--resetafter=<time> reset the wmbus dongle regularly, default is 23h`

This was introduced because the hardware dongles sometimes flip out after running for too long!?!?!?!
But it also exercises the reset logic, which is quite complicated. To flush any bugs. :-)

Can you run with: --resetafter=2m 
to see if you can provoke the problem?

I have fixed a bug in this code for version 1.4.0, but you might have found another bug.

weetmuts, 2022-04-22:
To debug this problem I need a --debug log of a reset where the nanocul does not get up and running again.
We should be able to see in the log what it is sending to the nanocul and what responses it gets.

BillGOH, 2022-04-22:
Thank you for your quick response!

This my log (looks good to me on first glance):

`wmbusmeters --resetafter=2m --debug /dev/ttyUSB0:cul:t1 MeinZaehler izar 15133235 NOKEY                    (wmbusmeters) version: 1.7.0-7-g869123d
(config) using device: /dev/ttyUSB0:cul:t1
(config) number of meters: 1
(lookup) with file/hex "/dev/ttyUSB0" cul
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars()           modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars()           modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect cul)
(serial /dev/ttyUSB0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary "5620312E3637206E616E6F43554C3836385F723536380D0A"
(cul) probe response "V 1.67 nanoCUL868_r568<0D><0A>"
(serialtty) closed /dev/ttyUSB0 (detect cul)
(cul) are you there? yes
(serial) EVENT thread interrupted
(main) opening /dev/ttyUSB0:cul:t1
Started config cul on /dev/ttyUSB0 listening on t1
(cul) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars()           modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars()           modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (cul)
(main) regular reset of cul  on /dev/ttyUSB0 will happen every 120 seconds
(cul) set link mode t
(serial /dev/ttyUSB0) sent "6272740A0D"
(serial) EVENT thread interrupted
(serial) received binary "544D4F44450D0A"
(cul) checkCULFrame "TMODE<0D><0A>"
(cul) no leading 'b' so it is text and no frame
(cul) received "TMODE"(serial /dev/ttyUSB0) sent "5832310A0D"
(wmbus) no alarm (expected activity) for cul
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) received binary "623239343441353131373830373335333231333135393331434131313131313030313335443131343538"
(cul) checkCULFrame "b2944A511780735321315931CA1111100135D11458"
(cul) no eol found yet, partial frame
(serial) received binary "3142353930373036313546363839414435354546353137464441353130314335343634433046443042393341303846          4634373444384331383131330D0A"
(cul) checkCULFrame "b2944A511780735321315931CA1111100135D114581B59070615F689AD55EF517FDA5101C5464C0FD0B93A08FF474D8C181          13<0D><0A>"
(cul) checkCULFrame RSSI_RAW=19
(cul) checkCULFrame LQI=64
(wmbus) trimming frame A "2944A511780735321315931CA1111100135D114581B59070615F689AD55EF517FDA5101C5464C0FD0B93A08FF474D8          C1"
(wmbus) ff a dll crc 0-9 931c ok
(wmbus) ff a dll crc mid 12-27 d55e ok
(wmbus) ff a dll crc mid 30-45 d8c1 ok
(wmbus) trimming frame A "2944A511780735321315931CA1111100135D114581B59070615F689AD55EF517FDA5101C5464C0FD0B93A08FF474D8          C1"
(wmbus) trimmed 6 dll crc bytes from frame a and ignored 0 suffix bytes.
(wmbus) trimmed frame A "2944A511780735321315A1111100135D114581B59070615F689AF517FDA5101C5464C0FD0B93A08FF474"
(cul) received full T1 frame
(meter) no meter handled  checking 1 templates.
(diehl) preprocess necessary SWAPPING
(diehl) Pre-processing: swapping address field
(wmbus) parseDLL @0 42
(wmbus) parseELL @10 32
(wmbus) parseNWL @10 32
(wmbus) parseAFL @10 32
(wmbus) parseTPL @10 32
(meter) MeinZaehler: for me? 15133235 in
(meter) MeinZaehler: yes for me
(meter) created "MeinZaehler" "izar" "15133235" encrypted
(meter) used meter template MeinZaehler  izar to match 15133235
(meter) started meter 1 (MeinZaehler 15133235 izar)
(diehl) preprocess necessary SWAPPING
(diehl) Pre-processing: swapping address field
[... 562 lines cut ...]
(wmbus) parseELL @10 32
(wmbus) parseNWL @10 32
(wmbus) parseAFL @10 32
(wmbus) parseTPL @10 32
(telegram) TPL CI=a1
(izar) Decoded PRIOS data: 4B13B405000D270500BF2A447E276C7A288284C50A96FC7FA8A518
(izar) 000   : 29 length (41 bytes)
(izar) 001   : 44 dll-c (from meter SND_NR)
(izar) 002   : a511 dll-mfct (DME)
(izar) 004   : 35321315 dll-id (15133235)
(izar) 008   : 78 dll-version
(izar) 009   : 07 dll-type (Water meter)
(izar) 010   : a1 tpl-ci-field (Mfct specific)
(izar) 011 C?: 711100135D63A6344F5498B5853E95F5651E10EAD8BCB01AAB0438193903BA mfct specific
MeinZaehler     15133235                000000  373.779 m3      337.677 m3      2021-10-31      8.5 y   no_alarm       n          o_alarm 8 s     0       2022-04-22 20:27.07
^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(serial) TIMER thread interrupted
(wmbus) disconnected /dev/ttyUSB0 cul
(serialtty) closed /dev/ttyUSB0 (cul)
(serial) event loop stopped!
(wmbus) deleted cul
(serial) closing 1 devices
`

hemandk, 2022-06-16:
Was there any progress on this? I seem to have the same problem, that the nano cul stopped working after 23h.

agrauballe, 2022-12-31:
I think I have encountered the same issue
Got the log below, now trying to let "resetafter=365d" as I assume somehow the nano cul does not survive the reset.
Needs more testing though, as I just recently got my setup running.
@weetmuts is "resetafter=365d" a valid option?

(wmbus) resetting /dev/ttyUSB0:cul[]
(wmbus device) timeout request id 1
(cul) setting link mode(s) c1 is not supported for this cul device!
Expected cul on /dev/ttyUSB0 but did not find it! Ignoring tty!

weetmuts, 2022-12-31:
I think resetafter=0d will turn it off completely.... :-) but yes 365d should set a year before a reset.

Reset will close the file descriptor /dev/ttyUSB0, this might or might not drop the DTR (data terminal ready) pin which might or might not trigger the serial device attached to know that the computer no longer wants to talk to it.... but assuming that the dongle does not care, it will know nothing.  

Wmbusmeters then waits 3 seconds after which wmbusmeters re-opens the file descriptor and sends the initialization commands again to the dongle.

If the dongle has gone haywaire and there is no check for DTR, then the dongle will be haywire even after the reset. Perhaps there is a lower level usb-reset that could power off/on the dongle? This would for sure reboot it.
Can you check if there is such a thing?

weetmuts, 2022-12-31:
If you both experience this and it happens after 23h then you should be able to provoke it after 2 minutes with
a --resetafter=2m can you both collect --debug logs @agrauballe @BillGOH (again thank you) but attach the log files instead of pasting them into the posts. This makes it easier for me to understand them.

Can you try this with the latest version of wmbusmeters on github. Thanks!

agrauballe, 2023-01-01:
@weetmuts, I tried to put --resetafter=2m and it does seem too work even after reset, I also think that my issue is actually not correlated to the reset. It occurs before the reset. So somehow the Nano Cul becomes unresponsive.
Well keep debugging and check when it occurs and what can bring it back.
Currently I have set reset to 55m after to try whether the reset can keep it healthy.

agrauballe, 2023-02-14:
@weetmuts and all, just an update from my side.
I have concluded that the "--resetafter" feature is actually a good thing and might recover the Nano Cul device in some case. So there is nothing that needs to be improved on wmbusmeters side. However the Nano Cul is troublesome at times. First I have figured out from others that it needs to have the original bootloader exchanged to the optiboot one. This makes it possible to recover the device via usbreset, rather than having to hot plug or reboot. 
But even after this the Nano Cul sometimes just stops and I have to do the usbreset cmd. I have made a bash script which runs by a cron job each 10 minutes to check that there are new readings, if not the script will stop wmbus meters, do usbreset and start wmbusmeters. After introducing this I keep receiving values but see the script is executed sometimes several times a day and sometimes after some days. But anyway it solves my issues.
Script is here for anybody as inspiration:

crontab line: 
*/10 * * * * /home/pi/docker/wmbusmeters/check_last_update.sh >> /home/pi/docker/wmbusmeters/log.txt

```bash
#!/bin/bash

file_timestamp="$(stat -c '%Y' /home/pi/docker/wmbusmeters/config/logs/meter_readings/MyTapWater)"
current_time="$(date +"%s")"

let delta_time=$current_time-$file_timestamp

if [ $delta_time -gt 600 ]
then
  cd /home/pi/docker/wmbusmeters/
  echo $(date) - Value: $delta_time >> log.txt
  /usr/local/bin/docker-compose down
  sleep 10
  usbreset 0403:6001
  sleep 5
  /usr/local/bin/docker-compose up -d
fi
```

vmath54, 2023-04-01:
Hi,

Same problem : from 11:00 p.m., wmbusmeters no longer reports information.
I have a nanoCUL, with firmware nanoCUL_MBUS_bufsize300.hex
On an odroid c4, armbian, and locally compiled wmbusmeters.
wmbusmeters is running as a service

after a usbreset on the nanoCUL, wmbusmeters returns the information again ; without having to restart the wmbusmeters service

**########### config files ############**

```
# cat /etc/udev/rules.d/95-VM_domotic.rules
...
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", SYMLINK+="ttyUSB-wmbus"
```

```
# cat /etc/wmbusmeters.conf
loglevel=normal
device=/dev/ttyUSB-wmbus:cul:t1
#resetafter=<time> reset the wmbus dongle regularly, default is 23h
logtelegrams=false
format=json
ignoreduplicates=true
#meterfiles=/var/lib/wmbusmeters/meter_readings
#meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
#regularshell=5   # pas encore implemente. Voir https://github.com/wmbusmeters/wmbusmeters/issues/591
shell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/"$METER_ID" -m "$METER_JSON" -u xxxxx -P yyyyy
alarmshell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters_alarm -m "$ALARM_TYPE $ALARM_MESSAGE" -u xxxxx -P yyyyy

```
```
# cat /etc/wmbusmeters.d/mywater.conf
name=mywater
id=zzzzzzzz
key=NOKEY
driver=dme_07
calculate_total_l=total_m3

```
**############### logs ####################**

```
# tail /var/log/wmbusmeters/wmbusmeters.log
...
(wmbusmeters) started meter 1 (mywater 25201732 dme_07)
[2023-03-31_19:23:13] (memory) rss 3854336 peak 3.67 MiB
[2023-03-31_23:00:05] (wmbus) resetting /dev/ttyUSB-wmbus:cul[]
[2023-03-31_23:00:08] (wmbus) reset completed /dev/ttyUSB-wmbus:cul[]

```
// --- no data since 2023-03-31_23:00 ---

```
# ls -al /dev/ttyUSB-wmbus
lrwxrwxrwx 1 root root 7 mars  27 11:57 /dev/ttyUSB-wmbus -> ttyUSB1

# usbreset 0403:6001          // at 2023-04-01_13:56
Resetting FT232R USB UART ... ok
```

// --- and wmbusmeters returns the information again ...

```
# tail /var/log/messages
Apr  1 13:56:08 localhost kernel: [474553.317396] ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1
Apr  1 13:56:08 localhost kernel: [474553.317468] ftdi_sio 1-1.2:1.0: device disconnected
Apr  1 13:56:08 localhost kernel: [474553.445103] usb 1-1.2: reset full-speed USB device number 7 using xhci-hcd
Apr  1 13:56:08 localhost kernel: [474553.595534] ftdi_sio 1-1.2:1.0: FTDI USB Serial Device converter detected
Apr  1 13:56:08 localhost kernel: [474553.595689] usb 1-1.2: Detected FT232RL
Apr  1 13:56:08 localhost kernel: [474553.596528] usb 1-1.2: FTDI USB Serial Device converter now attached to ttyUSB2

# ls -al /dev/ttyUSB-wmbus
lrwxrwxrwx 1 root root 7 avril  1 13:56 /dev/ttyUSB-wmbus -> ttyUSB2

# tail /var/log/wmbusmeters/wmbusmeters.log
...
[2023-03-31_19:23:13] (memory) rss 3854336 peak 3.67 MiB
[2023-03-31_23:00:05] (wmbus) resetting /dev/ttyUSB-wmbus:cul[]
[2023-03-31_23:00:08] (wmbus) reset completed /dev/ttyUSB-wmbus:cul[]
(wmbusmeters) started meter 1 (mywater 25201732 dme_07)
[2023-04-01_13:56:11] Started config cul on /dev/ttyUSB-wmbus listening on t1

```

closed 2023-07-02 by github-actions[bot]