wmbusmeters/wmbusmeters #519: Michael Rac MBWREP Mini

roth-m, 2022-04-22:
On my setup an Izare Re 868 Pulse is linked to a Michael Rac MBWREP Mini (https://michaelrac.com/alle-produkte/stationary-radio-range-extender/mbwrep-mini/?lang=en)
![Produkt_06_2-scaled-1]

(meter) meter: meter detection could not find driver for id: 2675xxxx mfct: (MPR) Michael Rac, Germany (0x3612) type: Reserved for switching devices (0x24) ver: 0x91
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(wmbus) WARNING! no key to decrypt payload! Permanently ignoring telegrams from id: 26751781 mfct: (MPR) Michael Rac, Germany (0x3612) type: Reserved for switching devices (0x24) ver: 0x91

Here are some telegrams:

T1;1;1;2022-04-22 13:13:56.000;13;13;2675xxxx;0x3e4412368117752691247aff00302589186700c9195b23c693fd9d15fa693ef1ef975c6fa9c15c8bb2c26c4fbebe8f26fb3fad5884733b5a8aaa133a3bedb2
T1;1;1;2022-04-22 13:13:56.000;14;14;2675xxxx;0x3e4412368117752691247aff00302589186700c9195b23c693fd9d15fa693ef1ef975c6fa9c15c8bb2c26c4fbebe8f26fb3fad5884733b5a8aaa133a3bedb2
T1;1;1;2022-04-22 13:14:29.000;10;11;2675xxxx;0x3e4412368117752691247a00003025f661d1b7c7c2f7977f97488f43ed83ff08fad64f0d07a9519ab398f91a7a0b98e0e9ccdfdc05ea6afe3df32fed5f0562
T1;1;1;2022-04-22 13:14:29.000;11;9;2675xxxx;0x3e4412368117752691247a00003025f661d1b7c7c2f7977f97488f43ed83ff08fad64f0d07a9519ab398f91a7a0b98e0e9ccdfdc05ea6afe3df32fed5f0562
T1;1;1;2022-04-22 13:15:02.000;10;11;2675xxxx;0x3e4412368117752691247a01003025402623a4be415c2a672d3a4bb98998890b4b9fe8e7bddf9b92da032180e67cda27f20e9506d9f95428686249706c6fee
T1;1;1;2022-04-22 13:15:02.000;14;12;2675xxxx;0x3e4412368117752691247a01003025402623a4be415c2a672d3a4bb98998890b4b9fe8e7bddf9b92da032180e67cda27f20e9506d9f95428686249706c6fee
T1;1;1;2022-04-22 13:15:35.000;11;9;2675xxxx;0x3e4412368117752691247a0200302594e600682e7db761d0993b7164bb44803d3e2f45d92a906f4fb0360319a47ac6225cf962b796b515b4f998c84e3ebbab
T1;1;1;2022-04-22 13:15:35.000;9;10;2675xxxx;0x3e4412368117752691247a0200302594e600682e7db761d0993b7164bb44803d3e2f45d92a906f4fb0360319a47ac6225cf962b796b515b4f998c84e3ebbab
T1;1;1;2022-04-22 13:16:08.000;9;8;2675xxxx;0x3e4412368117752691247a0300302544fa130bce59ca9df8497063252988a3608f64ed6c328d9a1324bebcf3b8b7645c5e1adb933b820766f176fca9bddcc7
T1;1;1;2022-04-22 13:16:08.000;7;10;2675xxxx;0x3e4412368117752691247a0300302544fa130bce59ca9df8497063252988a3608f64ed6c328d9a1324bebcf3b8b7645c5e1adb933b820766f176fca9bddcc7
T1;1;1;2022-04-22 13:16:41.000;8;10;2675xxxx;0x3e4412368117752691247a040030254751bdac853577461362eb68a34d0cd0a656d6f2e4b993191985d22b452ada8ab2f51e2183dbd12b7c1726a6da96dae6
T1;1;1;2022-04-22 13:17:14.000;9;9;2675xxxx;0x3e4412368117752691247a0500302572d06c7949be04c0f8593c21816fefb21bacd4448bd30eecec527d4413d1e6eed57cd64d92bd1898ab743ab9713485b4
T1;1;1;2022-04-22 13:17:47.000;8;9;2675xxxx;0x3e4412368117752691247a060030252697bff8e764d63987f50383581131641978ea225727c11506952344f184eae63f894772d0608317f12f0322bacdc147
T1;1;1;2022-04-22 13:17:47.000;10;10;2675xxxx;0x3e4412368117752691247a060030252697bff8e764d63987f50383581131641978ea225727c11506952344f184eae63f894772d0608317f12f0322bacdc147
T1;1;1;2022-04-22 13:18:20.000;9;7;2675xxxx;0x3e4412368117752691247a07003025d79dfdec569f1dd54a656f1a07c61878e546af1c8e580cbd3ac71b41365776c4580eae5e160c299f3d6415b8e143b37e
T1;1;1;2022-04-22 13:18:20.000;10;8;2675xxxx;0x3e4412368117752691247a07003025d79dfdec569f1dd54a656f1a07c61878e546af1c8e580cbd3ac71b41365776c4580eae5e160c299f3d6415b8e143b37e

weetmuts, 2022-05-25:
Thank you for the report. However the telegrams are encrypted. Do you have a key?

roth-m, 2022-05-25:
No, but I believe it's a default one as this company uses defauly keys for Sharky and Diehl water meter. Could it be possible that the data emited by the Diehl Pulse are just encapsulated  by the MBWREP Mini?!

BIBOLV, 2023-01-02:
Closing issue due to inactivity. If you manage to get decrypted telegrams, please reopen issue and post them here to continue to work on supporting them.

closed 2023-01-02 by BIBOLV