wmbusmeters/wmbusmeters #521: wmbusmeters v1.6 works - wmbusmeters:latest do dont work for me

zibous, 2022-04-25:
If I use the last current Version `DOCKERIMAGE=weetmuts/wmbusmeters`, the NanoCul is no longer recognized. 

With version 1.6 everything works. 
```json
{
   "media":"water",
   "meter":"izar",
   "name":"watermeter",
   "id":"18444343",
   "prefix":"",
   "serial_number":"000000",
   "total_m3":368.284,
   "last_month_total_m3":360.246,
   "last_month_measure_date":"2022-04-01",
   "remaining_battery_life_y":11.5,
   "current_alarms":"no_alarm",
   "previous_alarms":"no_alarm",
   "transmit_period_s":8,
   "manufacture_year":"0",
   "timestamp":"2022-04-25T11:02:25Z",
   "device":"cul",
   "rssi_dbm":-59
}
```
I have to change something during configuration ?

**Version 1.6:**
`weetmuts/wmbusmeters:release-1.6.0@sha256:80e6ca17ceae4d113a5e1b8c85c9350d29768ca974c753bb5feb6bf75dbc4cd1`

**Version latest**
**Error: No wmbus device detected, waiting for a device to be plugged in.**

[_wmbusmeters_logs.txt](https://github.com/weetmuts/wmbusmeters/files/8553901/_wmbusmeters_logs.txt)

```ini
## default settings
loglevel=debug
device=auto:t1
device=/dev/ttyACM0
logtelegrams=true
format=json
## logging DO NOT WORK - disabled
# #meterfiles=/docker/wmbusmeters/log/meter_readings
meterfilesaction=append
## logging DO NOT WORK - disabled
## logfile=/docker/wmbusmeters/log/wmbusmeters.log
## additonal settings
meterfilesnaming=name
meterfilestimestamp=hour
ignoreduplicates=true
alarmtimeout=1h
alarmexpectedactivity=mon-sun(00-23)
## MQTT Brocker settings
shell=/usr/bin/mosquitto_pub -h localhost -p 1883 -u wmbusMaster -P 82R6T7tsKRz2rDoPj2Z -t tele/watermeter/data -m "$METER_JSON"
alarmshell=/usr/bin/mosquitto_pub -h 10.1.1.94  -u  wmbusMaster -P 82R6T7tsKRz2rDoPj2Z -t   tele/watermeter/alarm -m "$ALARM_TYPE $ALARM_MESSAGE"
```

Docker Install:
```bash
#!/bin/bash
source .env

# ----------------------------------
# sudo sh wmbusmeters.sh
# ----------------------------------
DOCKERIMAGE=weetmuts/wmbusmeters:release-1.6.0
# DOCKERIMAGE=weetmuts/wmbusmeters
CONTAINERLABEL=wmbusmeters
APPSDATA=$PWD${DOCKER_APPSDIR}${CONTAINERLABEL}

echo "Create persistent data folder and log folder"
mkdir -p ${APPSDATA} >/dev/null 2>&1

echo "Try to remove previuos installation..."
docker stop ${CONTAINERLABEL} >/dev/null 2>&1
docker rm ${CONTAINERLABEL} >/dev/null 2>&1

echo "Install Docker container ${CONTAINERLABEL}."
docker run --detach \
  --privileged \
  --name=${CONTAINERLABEL} \
  --restart=always \
  --volume /etc/localtime:/etc/localtime:ro \
  --volume ${APPSDATA}:/wmbusmeters_data \
  --volume /dev/:/dev/ \
  ${DOCKERIMAGE}

echo "Docker container ${CONTAINERLABEL} ready."

```

weetmuts, 2022-04-28:
Can you run with --debug and capture the log? We need to see the communication with the cul.

zibous, 2022-04-28:
@weetmuts 
Yes, but how can i set --debugg with a docker installation ?

```bash
docker run --detach \
  --privileged \
  --name=${CONTAINERLABEL} \
  --restart=always \
  --volume /etc/localtime:/etc/localtime:ro \
  --volume ${APPSDATA}:/wmbusmeters_data \
  --volume /dev/:/dev/ \
  weetmuts/wmbusmeters --debugg

--> docker: Error response from daemon: OCI runtime create failed: container_linux.go:380: starting container process caused: exec: "--debug": executable file not found in $PATH: unknown.
```

weetmuts, 2022-04-28:
Ah, you already have loglevel=debug in the conf file. Please capture the log when you unplug/replug the nanocul, and post it here.

zibous, 2022-04-28:
@weetmuts 
ok.

**Log latest for weetmuts/wmbusmeters@sha256:7b254f46d78b0ed8fbf0bfcd87ffd9f24692f03fa9dea6a86c5b8330474e02b6**

[_wmbusmeters_latest_logs.txt](https://github.com/weetmuts/wmbusmeters/files/8580468/_wmbusmeters_latest_logs.txt)

Log Version 1.6:
[_wmbusmeters_v16_logs.txt](https://github.com/weetmuts/wmbusmeters/files/8580436/_wmbusmeters_v16_logs.txt)

weetmuts, 2022-04-28:
Yes, you are quite right the auto detection of the cul fails and I can reproduce it.
Can you change your device to explicitly name cul:

device=/dev/ttyS5:cul:t1
or just
device=cul:t1 
if the tty keeps changing.

Perhaps you should remove the second device ACM0 as well.

weetmuts, 2022-04-28:
Hmmm, but often it actually does work. Anyway, buy specifying cul it will start quicker and use the right driver directly.

zibous, 2022-04-28:
@weetmuts 
This is working for the latest docker image:
```
## default settings
loglevel=debug
device=auto:t1
# device=/dev/ttyACM0
device=/dev/ttyS5:cul:t1    <--- This is working /dev/ttyACM0 not !
logtelegrams=true
format=json
## logging
# #meterfiles=/docker/wmbusmeters/log/meter_readings
meterfilesaction=append
## logfile=/docker/wmbusmeters/log/wmbusmeters.log
## additonal settings
meterfilesnaming=name
meterfilestimestamp=hour
ignoreduplicates=true
alarmtimeout=1h
alarmexpectedactivity=mon-sun(00-23)
## MQTT Brocker settings
shell=/usr/bin/mosquitto_pub -h 10.1.1.94 -p 1883 -u **** -P **** -t tele/watermeter/data -m "$METER_JSON"
alarmshell=/usr/bin/mosquitto_pub -h 10.1.1.94  -u **** -P **** -t   tele/watermeter/alarm -m "$ALARM_TYPE $ALARM_MESSAGE"
```

```
# ls -l /dev/serial/by-id
lrwxrwxrwx 1 root root 13 Apr 28 09:28 usb-SHA_CUL868-if00 -> ../../ttyACM0

# ls /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 Apr 28 19:05 /dev/ttyACM0
# ls /dev/ttyS5
crw-rw---- 1 root dialout 4, 69 Apr 28 19:05 /dev/ttyS5
```

Log:
[_wmbusmeters_logs.txt](https://github.com/weetmuts/wmbusmeters/files/8584900/_wmbusmeters_logs.txt)

How to do ??

I would like to use a BASH file instead of the MQTT messages,  because I want to send the 
messages to 2 MQTT brokers and INFLUXDB.  

If I enter a BASH file instead of the SHELL that is in the config directory,  `--volume ${APPSDATA}:/wmbusmeters_data \`
it will not be found and will not be executed.  

How can I do this with a Docker installation ?

weetmuts, 2022-04-28:
Sorry, I am not an expert on the docker stuff. It seems like the protective security measures of docker makes it hard for you to execute anything that is not already installed in the docker image.

I actually think it is easier to install wmbusmeters as a daemon without docker. 
Assuming you are running outside of docker, I presume you just do
shell=/bin/bash /somewhere/mybash.sh "$METER_JSON"

zibous, 2022-04-29:
@weetmuts 

 Negativ, it looks like that `shell` can'nt execute a bash with docker image.
 
 ```log
debug message:
 /bin/sh: /usr/bin/bash: not found
 (shell) /bin/sh: return code 127
 (shell) /bin/sh exited with non-zero return code: 127
 
 shell=/usr/bin/bash /docker/apps/wmbusmeters/etc/publish.sh "$METER_JSON"
 
 /docker/apps/wmbusmeters
    ├── etc
    │   ├── publish.sh
    │   ├── wmbusmeters.conf
    │   └── wmbusmeters.d
    └── logs
        └── meter_readings

ls /usr/bin/bash
-rwxr-xr-x 1 root root 1.4M Oct  7  2021 /usr/bin/bash*

ls /docker/apps/wmbusmeters/etc/publish.sh
-rwxr--r-- 1 root root 397 Apr 26 13:48 /docker/apps/wmbusmeters/etc/publish.sh*

```

weetmuts, 2022-04-29:
@BIBOLV can you help here?

BIBOLV, 2022-05-01:
Firstly - I can confirm that with latest docker image, simple device declaration does not work any more with following error:
```shell
(serial) check if /dev/im871a can be accessed
(serial) tty /dev/im871a can be accessed
(verbose) ignoring device /dev/im871a
(lookup) with file/hex "/dev/im871a" unknown
```
I had to change /dev/im871a to /dev/im871a:im871a[serial]:c1,t1 to restore functionality. @weetmuts maybe alert about breaking change should be added to change list?

 Secondly - it is possible to run scripts in docker, but couple of thing to remember:
-  "**B**ourne **a**gain **sh**ell" is not available in wmbusmeters docker container, just "Bourne **sh**ell", so /bin/bash should be changed to /bin/sh. Also remember about "Shebang line" in script itself. 
- Location of script, so if you use following data location `--volume ${APPSDATA}:/wmbusmeters_data` in docker startup and your script is located in  `/docker/apps/wmbusmeters/etc/` and `DOCKER_APPSDIR=/docker/apps/` then path to script from inside of docker would be `/wmbusmeters_data/etc/publish.sh`. So changing line to `shell=/bin/sh /wmbusmeters_data/etc/publish.sh "$METER_JSON"` should work.

zibous, 2022-05-02:
@BIBOLV 
Thanks 👍 , sh is now working.

I still have another problem with logging. I can't find the right setting for the logs to be created.

```
## default settings
loglevel=debug
device=/dev/ttyS5:cul:t1
logtelegrams=true
format=json

## logging
meterfiles=/wmbusmeters/logs/meter_readings
logfile=/wmbusmeters/logs/wmbusmeters.log

meterfilesaction=append
meterfilesnaming=name
meterfilestimestamp=hour
ignoreduplicates=true

## MQTT Brocker settings
shell=/bin/sh /wmbusmeters_data/etc/publish.sh "$METER_JSON"

alarmtimeout=1h
alarmexpectedactivity=mon-sun(00-23)
shell=/bin/sh /wmbusmeters_data/etc/alarm_message.sh "$METER_JSON"
```

Errormessage:
```
(wmbusmeters) using log file /wmbusmeters/logs/wmbusmeters.log
Could not open log file /wmbusmeters/logs/wmbusmeters.log
```

BIBOLV, 2022-05-02:
Same as with script `/wmbusmeters_data/` not `/wmbusmeters/`

zibous, 2022-05-03:
@BIBOLV 

> * Same as with script /wmbusmeters_data/ not /wmbusmeters/

Yes correct, thanks 👍 

**Script ("Bourne shell")** 
> * Location of script, so if you use following data location `--volume ${APPSDATA}:/wmbusmeters_data` in docker startup and your script is located in  `/docker/apps/wmbusmeters/etc/` and [...]

Script ("Bourne shell") is executed, but the ENV arguments are not passed. When I run the scripts on the console, content and mqtt Message will be created.

**Docker Install**
```bash
#!/bin/bash
source .env

# ----------------------------------
# sudo sh wmbusmeters.sh
# ----------------------------------
DOCKERIMAGE=weetmuts/wmbusmeters
CONTAINERLABEL=wmbusmeters
APPSDATA=$PWD${DOCKER_APPSDIR}${CONTAINERLABEL}

echo "Create persistent data folder and log folder"
mkdir -p ${APPSDATA}

echo "Try to remove previuos installation..."
docker stop ${CONTAINERLABEL} >/dev/null 2>&1
docker rm ${CONTAINERLABEL} >/dev/null 2>&1

echo "Install Docker container ${CONTAINERLABEL}."
docker run --detach --interactive \
  --privileged \
  --name=${CONTAINERLABEL} \
  --restart=always \
  --volume /etc/localtime:/etc/localtime:ro \
  --volume ${APPSDATA}:/wmbusmeters_data \
  --volume /dev/:/dev/ \
  ${DOCKERIMAGE}

echo "Docker container ${CONTAINERLABEL} ready."
```

**Settings**
```ini
## -----------------------------------------------
## default settings for wnbusmeters docker latest
## -----------------------------------------------
loglevel=debug
device=auto:t1
# device=/dev/ttyACM0
device=/dev/ttyS5:cul:t1

## logging
meterfiles=/wmbusmeters_data/logs/meter_readings/
logfile=/wmbusmeters_data/logs/wmbusmeters.log
logtelegrams=true
format=json
meterfilesaction=append
meterfilesnaming=name
meterfilestimestamp=hour
ignoreduplicates=true
alarmtimeout=1h
alarmexpectedactivity=mon-sun(00-23)

## publish data and alarm
shell=/bin/sh /wmbusmeters_data/etc/publish.sh "$METER_JSON"
shell=/bin/sh /wmbusmeters_data/etc/alarm_message.sh "$ALARM_MESSAGE"
```

**Script**
```bash
#!/bin/sh

TIMESTAMP=`date +%Y-%m-%dT%H:%M:%S%z`
if [ -z "$1" ]
then
	echo "Argument payload is empty"
    echo "Error, no payload" > ./alarm_message.json
else
	/usr/bin/mosquitto_pub -h 10.1.1.94  -p 1883  -u smarthome -P Tfib5WD5eA4PkU64FD5 -t tele/watermeter/data -m "$1"
	/usr/bin/mosquitto_pub -h 10.1.1.217 -p 4883  -u smarthome -P TCs5h2Vv6gDgrFUP6N8 -t tele/watermeter/data -m "$1"
	/usr/bin/mosquitto_pub -h 10.1.1.94 -p 1883  -u smarthome -P Tfib5WD5eA4PkU64FD5 -t tele/watermeter/state -m "{\"timestamp\": $TIMESTAMP, \"state\": \"Online\"}"
    /usr/bin/mosquitto_pub -h 10.1.1.217 -p 4883  -u smarthome -P TCs5h2Vv6gDgrFUP6N8 -t tele/watermeter/state -m "{\"timestamp\": $TIMESTAMP, \"state\": \"Online\"}"
	echo "$1" > ./watermeter_data.json
fi
```
Docker Log
```
(config) "device" "auto:t1"
(config) "# device=/dev/ttyACM0" ""
(config) "device" "/dev/ttyS5:cul:t1"
(config) "## logging" ""
(config) "meterfiles" "/wmbusmeters_data/logs/meter_readings/"
(config) "logfile" "/wmbusmeters_data/logs/wmbusmeters.log"
(config) "logtelegrams" "true"
(config) "format" "json"
(config) "meterfilesaction" "append"
(config) "meterfilesnaming" "name"
(config) "meterfilestimestamp" "hour"
(config) "ignoreduplicates" "true"
(config) "alarmtimeout" "1h"
(config) "alarmexpectedactivity" "mon-sun(00-23)"
(config) "## publish and alarm" ""
(config) loading meter file /wmbusmeters_data/etc/wmbusmeters.d/diehl
(config) name=watermeter
(config) type=izar
(config) key=<notprinted>
(wmbusmeters) using log file /wmbusmeters_data/logs/wmbusmeters.logArgument payload is empty
Argument payload is empty
```

Log:

[wmbusmeters.log](https://github.com/weetmuts/wmbusmeters/files/8608408/wmbusmeters.log)

BIBOLV, 2022-05-03:
Change also watermeter_data.json to full path and check what is written there. I suspect that you will need to change $1 to $@

ptorrent, 2022-05-05:
same problem here with this command:

wmbusmeters --useconfig=/weble/supervision/data/14 --device=/dev/ttyUSB0 --listento=c1,t1 --debug

Any idea?

ptorrent, 2022-05-05:
Solved by setting:

wmbusmeters --useconfig=/weble/supervision/data/14 --device=/dev/ttyUSB0:im871a --listento=c1,t1

Sorry..

zibous, 2022-05-06:
@BIBOLV 

> Try to output all passed variables to file with $@ for testing purposes. File should be with full path. Tested and it is working for me.

Yes the  `$@` is working, but i do not get any payload to the mqtt brocker ?
- The `$STATEMESSAGE` is working.
- The `watermeter_data.json` will be created and is working.
- **But `$PAYLOAD` is allways 0 bytes**

```bash
#!/bin/sh
TIMESTAMP=`date +%Y-%m-%dT%H:%M:%S%z`
STATEMESSAGE="{\"timestamp\": $TIMESTAMP, \"state\": \"Online\"}"
PAYLOAD="$@"
## -----------------------------------------------
echo "Timestamp: $TIMESTAMP"
## -----------------------------------------------
echo "Publish RB3a 1 - Watermeter Data : $PAYLOAD"
## -----------------------------------------------
/usr/bin/mosquitto_pub -h 10.1.1.94  -p 1883  -u **** -P **** -t tele/watermeter/data  -m "$PAYLOAD"
## -----------------------------------------------
echo "Publish MQTT 2 - Watermeter Data : $PAYLOAD"
## -----------------------------------------------
/usr/bin/mosquitto_pub -h 10.1.1.217 -p 4883  -u ***** -P **** -t tele/watermeter/data  -m "$PAYLOAD"
## --------------------------------------------------
echo "Publish state to MQTT Brocker 1: $STATEMESSAGE"
## --------------------------------------------------
/usr/bin/mosquitto_pub -h 10.1.1.94  -p 1883  -u ***** -P **** -t tele/watermeter/state -m "$STATEMESSAGE"
## -----------------------------------------------
echo "Publish state to MQTT Brocker 2"
## -----------------------------------------------
/usr/bin/mosquitto_pub -h 10.1.1.217 -p 4883  -u **** -P **** -t tele/watermeter/state -m "$STATEMESSAGE"
## -----------------------------------------------
## save the payload as json data
## -----------------------------------------------
echo "$@" > /wmbusmeters_data/watermeter_data.json
echo "End script - $TIMESTAMP"
``` 

MQTT Message:
```log
1651817328: Received PUBLISH from 05749086-8A9D-7411-1DD0-E273B163B11A (d0, q0, r0, m0, 'tele/watermeter/data', ... (392 bytes))
1651817328: Received PUBLISH from 9F2A4B5D-BD95-5214-FF27-CF3E035D2DD2 (d0, q0, r0, m0, 'tele/watermeter/state', ... (58 bytes))
1651817328: Received PUBLISH from 97860C07-DBD0-C66B-6E62-A6A26FA279C0 (d0, q0, r0, m0, 'tele/watermeter/data', ... (0 bytes)) <-----
1651817328: Received PUBLISH from C2A305FA-A63B-7960-AA0D-9ACA2BD8DF8B (d0, q0, r0, m0, 'tele/watermeter/alarmstate', ... (58 bytes))
1651817371: Received PUBLISH from 1E22FADA-7A35-2097-D1DC-E8B8AE793557 (d0, q0, r0, m0, 'tele/watermeter/data', ... (392 bytes))
1651817371: Received PUBLISH from 9441E66D-3486-B6FA-C132-A60DD905CC05 (d0, q0, r0, m0, 'tele/watermeter/state', ... (58 bytes))
1651817371: Received PUBLISH from A2F0EC6C-58DE-AD8F-3F58-0561DAEA5091 (d0, q0, r0, m0, 'tele/watermeter/data', ... (0 bytes))  <-----
1651817371: Received PUBLISH from 0FDF7948-D022-8929-C431-4AB1A7C22964 (d0, q0, r0, m0, 'tele/watermeter/alarmstate', ... (58 bytes))
```

What can that be or  what am I doing wrong? ?

BIBOLV, 2022-05-06:
Is something written in /wmbusmeters_data/watermeter_data.json ?

zibous, 2022-05-06:
@BIBOLV 

> Is something written in /wmbusmeters_data/watermeter_data.json ?
yes. 

Mea culpa.
My mistake was that I used the same topic `tele / watermeter / data` for the alarm_message and it is usually empty.
Now everything works as intended.

Thanks for your help and time.

closed 2022-05-06 by zibous