wmbusmeters/wmbusmeters #523: Wmbusmeters techem

amy92000, 2022-04-27:
Hello
 I have two techem hot and cold water meters I have a problem I can't see my meters on Auto T1 or Auto C1
 I followed the tutorial that a French made on YouTube.
 When I do the scan I see several techem counters in my building but not mine I give you the references of my counters I also put the photo of the RTL key that I use.
 My second problem I can't create mytapwater file in opt/etc/wmbusmeters.d/
 If however you can help me and tell me how I do to create the file I am a beginner.
 thank you in advance.
[image]
[image]
[image]
[image]

weetmuts, 2022-04-28:
Why don't you unplug your antenna and place the rtlsdr dongle right next to your meter.
Then any readings should come from your meter. :-)
 
Or connect a really short, bad antenna, if unplugging it completely silences the incoming telegram.

Sometimes the meters only send during certain hours during the week. S

amy92000, 2022-04-28:
Please tell me how i should do to create MyTapWater folder what command should i do?  because it finds zero files in opt/etc/wmbusmeters.d

weetmuts, 2022-04-29:
How did you install wmbusmeters?

amy92000, 2022-05-01:
i put this command git clone https://github.com/weetmuts/rtl-wmbus.git
i am on raspberry 4

cloom, 2022-11-01:
Hello @amy92000 

Did you figure it out at the end? I have the same meter Techem Radio 4 and I can get the data just fine, maybe I can help.
I am using a [Nooelec NESDR SMArt v4](https://www.nooelec.com/store/nesdr-smart-sdr.html).

I was wondering if you also get only a precision of 0.1 cubic meter over radio?

gotschi, 2022-11-10:
@cloom can you maybe help me?

I have 2 Radio 4's and I don't see the data, could it be that they are encrypted?

What is the proceeding upon discovering a telegram of it?
Thank you!

cloom, 2022-11-10:
@gotschi it helps to put commands or know where you get the output from :)

I am no expert and not home but here's what I can tell you:
Based on the manufacturer, the type and the driver selected by wmbusmeters it seems you are capturing information from your heating systems and some of your neighbors as you have 3 different IDs you get telegrams from (12952740, 12952739 and 12952737).

Do you have this device for your heating system?
https://www.engelmann.de/en/sensostar-c/

To see all telegrams received by the dongle I am using the following command:
```
$ wmbusmeters --logtelegrams rtlwmbus
```
I would advise to  use the logfile option to store everything in a file (here: mylogfile.txt). You can then let the command run for a while and then see how many telegrams you received and from which devices:
```
$ wmbusmeters --logtelegrams --logfile=mylogfile.txt rtlwmbus
```
As an example of the output I can get, here's a capture of a telegram from my similar heating system (but not encrypted):
```
$ cat mylogfile.txt
[...a lot of output...]
Received telegram from: 67659047
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat meter (0x43)
                   ver: 0x39
                device: rtlwmbus[00000001]
                  rssi: 116 dBm
                driver: compact5
telegram=|_37446850479065673943A2109F2BD61C0058DD030000000140000118A003000000000000000000C00110F0908507C860D14E3199FC52451F|+581
[...and more output...]
```
And then I can use whether the command line:
```
$ wmbusmeters --analyze _37446850479065673943A2109F2BD61C0058DD030000000140000118A003000000000000000000C00110F0908507C860D14E3199FC52451F
Auto driver  : compact5
Best driver  : compact5 04/49
Using driver : compact5 00/00
000   : 37 length (55 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 6850 dll-mfct (TCH)
004   : 47906567 dll-id (67659047)
008   : 39 dll-version
009   : 43 dll-type (Heat meter)
010   : a2 tpl-ci-field (Mfct specific)
011 C?: 109F2BD61C0058DD030000000140000118A003000000000000000000C00110F0908507C860D14E3199FC52451F mfct specific
059 C!: d61c energy used in previous billing period (7382.000000 KWH)
063 C!: dd03 energy used in current billing period (989.000000 KWH)

{
    "media":"heat",
    "meter":"compact5",
    "name":"",
    "id":"67659047",
    "total_kwh":8371,
    "current_kwh":989,
    "previous_kwh":7382,
    "timestamp":"2022-11-02T19:19:17Z"
}
```
or the website [wmbusmeters.org](https://wmbusmeters.org/analyze/37446850479065673943A2109F2BD61C0058DD030000000140000118A003000000000000000000C00110F0908507C860D14E3199FC52451F) to analyze the telegram received. 

Do you use the antenna? Maybe you can get close to your Techem Radio4. I don't have the exact output for my Techem but I get a different ID which is sending the telegram than the one written on them, but it's something like:
```
Received telegram from: XXXXXXXX
          manufacturer: (TCH) Techem Service (0x5068)
          concerning: ZZZZZZZ

```
Where ZZZZZZZ is the ID written on the counters. I have noticed mine are sending the information every 4min but only send update their value once per day around 3AM :(
The value is also in hL so 0.1 cubic meter precision.

I hope that helps

gotschi, 2022-11-11:
Thank you very much @cloom !

So `12952739` is my floor heating Techem S3,
this device ( I guess similar to what you posted)
[image]

Unfortunately I have no idea how to get a key for decryption?
I put it here but yeah, seems I need to get the key. ( Can I maybe read the LED on it with a light sensor? Do you by chance know how that behaves?)

https://wmbusmeters.org/analyze/_A944C5143927951200047A81009025ABEA435A69A6DAB50AB0B76181476F700A4AC01072D016E6033B173983FC9DCD4906F1FE4683C8D5714937DEDD4B1E5AFE5B130E587A477A7EED904D4AB7B1B985AF2AD1EA6FEAAA196BEE4C32C06C4D20A766769A3C5FA30ED156368547552F19697B7772BBB770FF80E3A39F2695CA0C8BEB0E5235DF907930213DEE860957585D902D135CF2BF2606DB07DA7DB56D03FD0C05010002FD0B0111

I could not get Telegrams from my Water sensors yet but they are both Radio 4's (cold/warm water)...

cloom, 2022-11-11:
What you can do is check your Techem Radio 4 id, let's take the one from @amy92000 as an example:
[image]

The ID is `73255592`. You can check in the logfile you have if the ID of your Techem Radio 4 are seen:
```
$ grep 73255592 mylogfile.txt
```

If you have nothing... then it seems you're not getting data from your counters. 

Is the antenna plugged on your dongle? If not you'll have to put the dongle next to the Techem Radio 4, or like it was [suggested](https://github.com/weetmuts/wmbusmeters/issues/523#issuecomment-1112331628), unplug the antenna (to not capture other signals) and put the key next to the counter.

I have seen [another thread](https://github.com/weetmuts/wmbusmeters/issues/46) where the data from the heating system can be retrieved using optical communication, but that's a different issue than the one opened here which is why you and @amy92000 cannot see your Techem Radio 4 data.

gotschi, 2022-11-11:
Yeah I've took note of my IDs earlier but only the Floor heating counter shows up, none of the Radio 4 :/
12218753 Cold Water
66977478 Warm Water

I have a antenna connected to the sdr, it shows all the other floor heating counters in other apartmens so it should also detect the Radio 4 I believe but yet no ID shows up for mine

I've requested Techem for information on the keys for them, lets see what they reply.
In any case, thank you for rubber ducking with me ;)

MartinBachmannHD, 2022-11-11:
...I just yesterday gave up to catch data from Techem radio 4 / vario 4 Heat Cost Allocators with my CUL Hardware on 'T' (maybe with other Hardware?). I was able to receive many, many telegrams from other Techems (most likely older Techem hardware), but all my quite new ones never showed up, no matter which tricks I tried. So I assume the new Hardware revision is more protected / proprietary?
[image]

machinekoder, 2023-01-05:
According to the manual, the `radio 4` devices send data only twice a month. So long-term logging is required to get some data.

closed 2023-04-19 by github-actions[bot]

gusi1994, 2023-09-03:
@gotschi Did you get any reply from techem regarding the key to decrypt the messages?

gotschi, 2023-09-04:
> @gotschi Did you get any reply from techem regarding the key to decrypt the messages?

no

Sheol27, 2024-11-20:
> ...I just yesterday gave up to catch data from Techem radio 4 / vario 4 Heat Cost Allocators with my CUL Hardware on 'T' (maybe with other Hardware?). I was able to receive many, many telegrams from [...]

Did you find a solution ? Currently i have the same device, i'm able to receive the telegrams but can't decode it with wmbusmeter

MartinBachmannHD, 2024-11-21:
Some time has passed since then... If I remember correctly I was able to read the messages but could not de-crypt them. Also our apartment-management company and techem all refused to give me the key (I was told I would be able with this to de-crypt everyone's messages - Datenschutz!!!!). Unless I find a message-key-hacking tool I stopped all activities. I think Techem offers a portal solution to provide you access to your data, but the apartment management company has to subscribe (and pay....) I assume.