wmbusmeters/wmbusmeters #524: Generic driver to support also unknown wmbus devices

rcliebig, 2022-04-28:
Hi, is there a way to transform telegrams for unknown devices without a specific driver?
Eg i would like to see something generic eg: `{"type":"gas", "id":"123456", "value_1":8736,12, "unit_1":"m3", "value_2":3,021, "unit_2": "V", ... }` ?

Thanks for your nice piece of software.

rcliebig, 2022-04-29:
Thank you for your insights about the general future of the wmbusmeters tool.
Though i think providing a way to extract data (eg via json transformation) of the telegrams that are not vendor specific a nice addon.
I came to this idea as i stumbled upon the iobroker library to analyze the packets i am sending from my own developed sensor One can try it out here without any installation: https://npm.runkit.com/iobroker.wireless-mbus

Your tool is closing the bridge from vendor devices to my smart home system :-)
Thanks for all your efforts!

weetmuts, 2022-04-29:
Thanks! Are you sending wmbus packets from your sensor? Have you tried "wmbusmeters --analyze ..."
or the online service of the same: https://wmbusmeters.org

It should be really easy for you to add a driver for your sensor. Look at this one for example:
https://github.com/weetmuts/wmbusmeters/blob/master/src/driver_ultrimis.cc

BIBOLV, 2023-01-02:
Closing due to inactivity. Please reopen if you want to continue to work on topic.

closed 2023-01-02 by BIBOLV