wmbusmeters/wmbusmeters #528: Move deb package meter readings from /var/log/ to /var/lib/?

petterreinholdtsen, 2022-04-30:
The default configuration file included in the deb build create /var/log/wmbusmeters/meter_readings/.  It is generated by the scripts/install_default_configuration.sh script.  Files in /var/log/ are normally rotated and it is ok to remove old versions after a while to clear some space.  I suspect meter readings are more valuable to keep than system log information, and wonder if it perhaps would be better to place it under /var/lib/ instead.  Do you have an opinion on this?

The FHS, https://refspecs.linuxfoundation.org/FHS_2.3/fhs-2.3.html , have this to say about /var/log/ and /var/lib/:

- /var/lib/ :
  This hierarchy holds state information pertaining to an application or the system. State information is data that programs modify while they run, and that pertains to one specific host. Users must never need to modify files in /var/lib to configure a package's operation.

  State information is generally used to preserve the condition of an application (or a group of inter-related applications) between invocations and between different instances of the same application. State information should generally remain valid after a reboot, should not be logging output, and should not be spooled data.

  An application (or a group of inter-related applications) must use a subdirectory of /var/lib for its data. There is one required subdirectory, /var/lib/misc, which is intended for state files that don't need a subdirectory; the other subdirectories should only be present if the application in question is included in the distribution. [[38]](https://refspecs.linuxfoundation.org/FHS_2.3/fhs-2.3.html#FTN.AEN2295)

  /var/lib/<name> is the location that must be used for all distribution packaging support. Different distributions may use different names, of course.

- /var/log/:
  This directory contains miscellaneous log files. Most logs must be written to this directory or an appropriate subdirectory.

It is of course a question of interpretation if the collection of meter readers is a log or an archive.  I find it to be an archive to be kept, not a log to be rotated.

weetmuts, 2022-04-30:
You are quite right. It should be pretty safe to change the default location to /var/lib since 
existing installations will not be changed, since the directory is specified in the conf file.

I have pushed a fix.

weetmuts, 2022-05-01:
Fix pushed. https://github.com/weetmuts/wmbusmeters/commit/b34d53c65f07bbb55ea68fb870acb1ff100e3563

closed 2022-05-01 by weetmuts