wmbusmeters/wmbusmeters #529: Daemon in deb lack USB device access by default

petterreinholdtsen, 2022-04-30:
I just set up wmbusmeters using the deb, and discovered that the daemon running as user wmbusmeters lacked access to the USB dongle with the radio receiver.  It would be better if this worked out of the box.  The device in question (/dev/bus/usb/002/003 in my case) is owned by root:plugdev, while the daemon run as wmbusmeters:nogroup.  The ACLs reported by getfacl show that the lightdm user also got access.  I assume this will change once someone log in on the console, according to the TAG+="uaccess" semantics of udev.

This can be fixed in various ways.  One is to add a udev rule to change the device ownership, another is to add a ACL granting the wmbusmeters user access.   Not quite sure which is the best approach, but thought it best to get the issue recorded.

weetmuts, 2022-04-30:
Hmmm, I think that the installation script does add wmbusmeters to the groups plugdev and dialout.
(And uucp for bsd style systems which is the same as dialout.)
Check scripts/add_wmbusmeters_user.sh

But this might not translate automatically into the deb installation of the daemon?

petterreinholdtsen, 2022-04-30:
[Fredrik Öhrström]
> Hmmm, I think that the installation script does add wmbusmeters to the
> groups plugdev and dialout.  (And uucp for bsd style systems which is
> the same as dialout.)  Check scripts/add_wmbusmeters_user.sh

Right.  I am not sure it is a good way to handle it.  I suspect a udev
rule is more future proof.

> But this might not translate automatically into the deb installation
> of the daemon?

Probably just my improved build rules causing it.  I did not notice the
add_wmbusmeters_user.sh code and did not lift it to the postinst.  But I
suspect adding groups might be a less optimal approach.

-- 
Happy hacking
Petter Reinholdtsen

weetmuts, 2022-04-30:
So, I had udev rules, but they are not useful, since most usb wmbus devices
are just standard serial ports connected to a wmbus chip. We can only find
the device by talking to the serial port and probing.

We could perhaps have a udev rule to identify the rtlsdr dongles, though.

petterreinholdtsen, 2022-04-30:
[Fredrik Öhrström]
> So, I had udev rules, but they are not useful, since most usb wmbus devices
> are just standard serial ports connected to a wmbus chip. We can only find
> the device by talking to the serial port and probing.

Why could the udev rule not grant access to standard serial ports when
the package is installed?

> We could perhaps have a udev rule to identify the rtlsdr dongles,
> though.

Could work, yeah.

-- 
Happy hacking
Petter Reinholdtsen

weetmuts, 2022-05-01:
It is important that the user still can access the serial ports and the rtlsdr device even after the wmbusmeters deb has been installed. This is necessary for debugging and is also part of the system setup, you often run "wmbusmeters auto:c1,t1" to get a feeling of the local radio traffic. 

I think that it is ok for wmbusmeters to be added to the dialout group. Recreating and overriding the default serial port rules in /lib/udev is a serious business and will probably cause a lot of intricate problems.

However the plugdev group is perhaps too broad.  How would you suggest that the access to the rtlsdr dongle
is granted to wmbusmeters, without revoking access from the user?

weetmuts, 2022-05-01:
The rtl-sdr udev rules can be found here: /lib/udev/rules.d/60-librtlsdr0.rules

weetmuts, 2022-05-01:
The serial port (dialout) rules can be found here: /lib/udev/rules.d/50-udev-default.rules

petterreinholdtsen, 2022-05-01:
[Fredrik Öhrström]
> It is important that the user still can access the serial ports and
> the rtlsdr device even after the wmbusmeters deb has been
> installed. This is necessary for debugging and is also part of the
> system setup, you often run "wmbusmeters auto:c1,t1" to get a feeling
> of the local radio traffic.

Sure.  But this will be taken care of with the uaccess udev tag, so no
need for group membership there.

The issue at hand is how to grant the daemon access even when there is
no-one logged in on the machine.

> I think that it is ok for wmbusmeters to be added to the dialout
> group. Recreating and overriding the default serial port rules in
> /lib/udev is a serious business and will probably cause a lot of
> intricate problems.

My idea was not to override the default rules, but to amend to them.
But I had a look at the udev default rules and the udev source, only to
discover there is no generic mechanism in udev to use acls to grant
access to devices.  The only code for this is in the uaccess internal
mechanism, which do not seem to be usable for anything other than
granting the logged in user access.

> However the plugdev group is perhaps too broad.  How would you suggest
> that the access to the rtlsdr dongle is granted to wmbusmeters,
> without revoking access from the user?

My idea was to tell udev add a ACL using setfacl or similar, by
providing a udev rule somewhere in /lib/udev/rules.d/ (or perhaps under
/etc/) using the wmbusmeters package, allowing the wmbusmeters user
access to relevant devices.  After looking at the code, I am less sure
if it is going to work.

-- 
Happy hacking
Petter Reinholdtsen

closed 2023-02-20 by github-actions[bot]