wmbusmeters/wmbusmeters #530: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8)

c64emulator, 2022-05-01:
I get errormessage when I try to attach to the container on raspberry.
`The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8).`

When looking at the docker repo [https://hub.docker.com/r/weetmuts/wmbusmeters/tags](https://hub.docker.com/r/weetmuts/wmbusmeters/tags) I see following:
[image]

Seems that the declaration of the OS/arch is wrong for the images "latest-arm64v8" and "latest-armhf"?

BIBOLV, 2022-05-01:
For now, please, use previous version of working docker image, while I dig out my testing rig, find and fix the issues.

BIBOLV, 2023-01-02:
Docker image tags were fixed. Closing issue.

closed 2023-01-02 by BIBOLV