wmbusmeters/wmbusmeters #531: traffic capture on M-Bus

NeuerUser, 2022-05-03:
Hello
Sorry, I don't know if this is the right place to ask.
I have a GWF water meter that has been connected to a wmbus sender module by my water company for remote reading. I would also like to see my water consumption. So I asked them for the encryption key. They just told me that they cannot give it to me.

Now I wonder, if I could "listen" to the traffic between the wmbus sender module and the water meter, which are connected by cable running m-bus protocol. I guess the wired traffic is not encrypted, so that could work. I, however, do not know, if it is possible at all to "listen" to (or "capture") the traffic on the cable. 
I understand that there must not be two masters on one bus, so as the wmbus sender module works as a master to the water meter, I guess, I should not put another master onto the same bus. So, I could only put a slave module on the bus and connect it via TTL serial or USB to a microcontroller or a PC. But, I haven't found any information, if it is possible to "sniff" the data that is transferred between the sender module and the water meter via a slave module.

Can anyone tell me if that is possible? And can the "sniffing" be done with "wmbusmeters"? Or do I need a different sw solution?

Thanks in advance!

weetmuts, 2022-05-04:
I suppose it is possible (https://www.zeta-eng.ch/en/products/interfaces/m-bus-analyzer/)

Though, you probably only need to capture the message from the meter to the wmbus module.
The capture message probably starts with 68....68..... and can be decoded as is by wmbusmeters.
This is a normal serial 2400 baud E81 communication, but the bits are sent (from the meter to the controller) using changes in the current. (from the controller to the meter they are sent using voltage changes).

I have no idea how to actually build such a solution. Electronics is not my forte.

An example mbus message:

https://wmbusmeters.org/analyze/68F8F86808007200464470A7320404270000000974040970040C0E082303000C14079519000B2D0500000B3B0808000A5B52000A5F51000A6206004C14061818004C0E490603000C7800464470891071609B102D020100DB102D0201009B103B6009009A105B78009A105F74000C22726701003C22000000007C2200000000426C01018C2006000000008C3006000000008C80100600000000CC200600000000CC300600000000CC801006000000009A115B64009A115F63009B113B5208009B112D020100BC0122000000008C010E490603008C2106000000008C3106000000008C811006000000008C011406181800046D310ACA210F21040010A0C116

weetmuts, 2022-06-19:
Closing this for now! Please open a new issue when you have captured some more mbus telegrams and need  help decoding them.

closed 2022-06-19 by weetmuts