wmbusmeters/wmbusmeters #532: HA Integration new pollinterval key is not working

autega, 2022-05-03:
Thanks for implementing mbus to wmbusmeters.
I´m getting the following error:

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Syncing wmbusmeters configuration ...
Registering meters ...
Adding meter-0001 ...
Generating MQTT configuration ... 
Broker core-mosquitto will be used.
Running wmbusmeters ...
Found invalid key "pollinterval" in meter config file
Started config MAIN=mbus on /dev/ttyUSB0

My config:

data_path: /config/wmbusmeters
conf: |-
  loglevel=normal
  device=MAIN=/dev/ttyUSB0:mbus:2400
  #donotprobe=/dev/ttyACM2
  logtelegrams=true
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=Strom
    driver=auto:mbus
    pollinterval=10s
    id=01000000
    key=NOKEY
mqtt: {}

Using "pollinterval" at cli is working as indented.

Thanks for your work

weetmuts, 2022-06-25:
Hmm, I just tested it on my setup and it works. Could it be that I have to do a new HA release by
bumping the version on ha-addon/config.json ? So that your setup uses the latest version of the github source?

weetmuts, 2022-06-25:
I just made a release 1.8.0 and updated the ha-addon version. Please test when it has rebuilt, that 
is in fact working!

weetmuts, 2022-08-28:
Does it work?

For the meter driver, please specify the exact meter driver instead of auto.

The reason is that auto-detection might switch to new improved drivers, which 
might not be backwards compatible. So for stability specify the meter driver name:
like driver=xyz:mbus

autega, 2022-11-01:
Sorry for the late reply. Everything is now working.
There was also some complication with two old mbus devices sharing the same secondary address.
Unfortunately it is not possible to change the id on these devices.
Is there a chance for wmbusmeters to use the primary address instead?

Should I open a new issue for that?

Thanks for your work.

weetmuts, 2022-11-01:
@autega Great that you got it working! 

Actually wmbusmeters already supports the primary address. :-)
Use id = 17 for wmbusmeters to use the primary address 17 instead.

 Please test if you have the time.

closed 2022-11-14 by autega