wmbusmeters/wmbusmeters #534: BMETERS Polska, Poland

ptorrent, 2022-05-05:
Hey therE !

I'm not able to get values from this meter, this is what I have with analyze:

```
wmbusmeters --analyze simulation_mymeter.txt
Auto driver  : not found!
Best driver  : aventieshca 03/03
Using driver : aventieshca 03/03
000   : 2e length (46 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : b009 dll-mfct (BMP)
004   : 98610368 dll-id (68036198)
008   : 53 dll-version
009   : 08 dll-type (Heat Cost Allocator)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : f2 tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 0020 tpl-cfg 2000 (synchronous )
015   : 2F skip
016   : 2F skip
017   : 03 dif (24 Bit Integer/Binary Instantaneous value)
018   : 6E vif (Units for H.C.A.)
019 C!: 000000 ("current_consumption_hca":0)
022  ?: 0F manufacturer specific data 100043106A7D2C4A078F12202CB1242A06D3062100210000

{
    "media":"heat cost allocation",
    "meter":"aventieshca",
    "name":"",
    "id":"68036198",
    "current_consumption_hca":0,
    "consumption_at_set_date_hca":0,
    "consumption_at_set_date_2_hca":0,
    "consumption_at_set_date_3_hca":0,
    "consumption_at_set_date_4_hca":0,
    "consumption_at_set_date_5_hca":0,
    "consumption_at_set_date_6_hca":0,
    "consumption_at_set_date_7_hca":0,
    "consumption_at_set_date_8_hca":0,
    "consumption_at_set_date_9_hca":0,
    "consumption_at_set_date_10_hca":0,
    "consumption_at_set_date_11_hca":0,
    "consumption_at_set_date_12_hca":0,
    "consumption_at_set_date_13_hca":0,
    "consumption_at_set_date_14_hca":0,
    "consumption_at_set_date_15_hca":0,
    "consumption_at_set_date_16_hca":0,
    "consumption_at_set_date_17_hca":0,
    "error_flags":"",
    "timestamp":"2022-05-05T13:10:27Z"
}

```

ptorrent, 2022-05-05:
By the way, is the a way to get list all drivers ?

Kosygor, 2022-05-05:
> By the way, is the a way to get a list of all drivers ?

It is right [here ](https://github.com/weetmuts/wmbusmeters/tree/master/src).

ptorrent, 2022-05-05:
If I set this is the meter config file:

```
name=a
id=78328752
driver=auto
```

I will have this as output:

```
telegram=|2E44B0095287327853087A44000020_2F2F036E0000000F100001998D7D2CED08AB117D2C0000000000000000000000|+742
{"media":"heat cost allocation","meter":"auto","name":"78328752","id":"78328752","timestamp":"2022-05-05T18:08:21Z","device":"im871a[00102864]","rssi_dbm":-44
```

AND 

if I set this in meter file:

```
name=a
id=78328752
driver=aventieshca
```

Output will be:
```
telegram=|2E44B0095287327853087A46000000_2F2F0F0000000000000000000000000000000000000000000000007D2C202C2F|+76
{"media":"heat cost allocation","meter":"aventieshca","name":"78328752","id":"78328752","current_consumption_hca":0,"consumption_at_set_date_hca":0,"consumption_at_set_date_2_hca":0,"consumption_at_set_date_3_hca":0,"consumption_at_set_date_4_hca":0,"consumption_at_set_date_5_hca":0,"consumption_at_set_date_6_hca":0,"consumption_at_set_date_7_hca":0,"consumption_at_set_date_8_hca":0,"consumption_at_set_date_9_hca":0,"consumption_at_set_date_10_hca":0,"consumption_at_set_date_11_hca":0,"consumption_at_set_date_12_hca":0,"consumption_at_set_date_13_hca":0,"consumption_at_set_date_14_hca":0,"consumption_at_set_date_15_hca":0,"consumption_at_set_date_16_hca":0,"consumption_at_set_date_17_hca":0,"error_flags":"","timestamp":"2022-05-05T18:12:31Z","device":"im871a[00102864]","rssi_dbm":-44}
```

Seems that "auto" is not doing his job correctly

Kosygor, 2022-05-05:
Which bmeters meter do You have ?

ptorrent, 2022-05-05:
this one : [https://bmetersuk.com/products/hydroclima-rfm/](https://urldefense.proofpoint.com/v2/url?u=https-3A__bmetersuk.com_products_hydroclima-2Drfm_&d=DwQFAw&c=euGZstcaTDllvimEN8b7jXrwqOf-v5A_CdpgnVfiiMM&r=h7wUln9dpMtdBZJWrO0KnEl0Q16oHEtdY1OZOC5uZU8&m=Q0p7Mrr4Ek1pMXJ4UTXJ3wdGVfIipaV2vK_0n4AHd_k&s=6P_K5VvMCkdCQNw_4-io96D3g0vedCo3_9ZlMrmqSjo&e=)

Kosygor, 2022-05-05:
My bad You have specified it earlier! 
Only other bmeters meter is hydrocalm3
Test This this:
```
name=a
id=78328752
driver=hydrocalm3
key=0
```
and this 

```
name=a
id=78328752
driver=rfmamb
key=0
```
I am no pro but this should give You "something". Also there is high chance that data frames are encrypted. If it is true Your operator should give You AES-key (they will be probably confused ;))

weetmuts, 2022-05-05:
Thanks @ptorrent ! I pushed a almost complete driver, hydroclima.
This telegram contains one wmbus compatible data point (the hca reading, which is 0 in this telegram).
This one is printed with any of the hca drivers, which is why you could have used the aventies driver.

Then there is manfacturer specific data following. Fortunately BMeters has actually 
provided me with a document that describes the format of the extra data!  Though 
I do not understand the exact meaning of some stuff.

If you run analyze again, you will get:
```
000   : 2e length (46 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : b009 dll-mfct (BMP)
004   : 98610368 dll-id (68036198)
008   : 53 dll-version
009   : 08 dll-type (Heat Cost Allocator)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 00 tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 2000 tpl-cfg 0020 ( )
015   : 2F skip
016   : 2F skip
017   : 03 dif (24 Bit Integer/Binary Instantaneous value)
018   : 6E vif (Units for H.C.A.)
019 C!: 000000 ("current_consumption_hca":0)
022  ?: 0F manufacturer specific data 100043106A7D2C4A078F12202CB1242A06D3062100210000
023   : *** 10 frame identifier OK
024 C!: *** 0043 status
026 C!: *** 106A time
028 C!: *** 7D2C date 2c7d
030 C!: *** 4A07 ("average_ambient_temperature_c":18.66)
032 C!: *** 8F12 ("max_ambient_temperature_c":47.51)
034 C!: *** 202C max date 2c20
036 C!: *** B124 num measurements 9393
038 C!: *** 2A06 ("average_ambient_temperature_last_month_c":15.78)
040 C!: *** D306 ("average_heater_temperature_last_month_c":17.47)
042 C!: *** 2100 indication u 3.300000
044 C!: *** 210000 total indication u 3.300000

{
    "media":"heat cost allocation",
    "meter":"hydroclima",
    "name":"",
    "id":"68036198",
    "current_consumption_hca":0,
    "average_ambient_temperature_c":18.66,
    "max_ambient_temperature_c":47.51,
    "average_ambient_temperature_last_month_c":15.78,
    "average_heater_temperature_last_month_c":17.47,
    "timestamp":"2022-05-05T18:44:51Z"
}
```

So it remains to understand how to translate the time bits to an hour:min
and the date bits to a date day/month.
As an example for time: 0x5208 → 11:40
and for date: 0x14d4 → 31 VII 2010
Perhaps you can help me figure out this?

Then there is the indication u whatever that is, the num readings and the status.
Any hints on what they can be is very appreciated.

weetmuts, 2022-05-05:
@ptorrent @Kosygor The meaning of the top three lines of analyze:
```
Auto driver  : not found!
Best driver  : aventieshca 03/03
Using driver : aventieshca 03/03
```
is to tell you that there was no knowledge in wmbusmeters of the mfct,ver,type combo
that identifies the meter driver. Ie auto will not work. However analyze then runs ALL drivers
on the telegram and counts how many bytes of content has been understood. 

Since the telegram only contains 3 bytes of wmbus compliant content at offset 19,
any hca driver that picks those will get a 3 out of 3 byte score. So aventies is the first test so that is the pick.

Then the final report is done using the selected driver. 
If you run with the latest wmbusmeters you now get:
```
Auto driver  : hydroclima
Best driver  : hydroclima 26/26
Using driver : hydroclima 00/00
```

ptorrent, 2022-05-05:
@weetmuts  Awseome ! thanks :)

```
wmbusmeters --analyze simulation_mymeter.txt
Auto driver  : hydroclima
Best driver  : hydroclima 26/26
Using driver : hydroclima 00/00
000   : 2e length (46 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : b009 dll-mfct (BMP)
004   : 52873278 dll-id (78328752)
008   : 53 dll-version
009   : 08 dll-type (Heat Cost Allocator)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 44 tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 0020 tpl-cfg 2000 (synchronous )
015   : 2F skip
016   : 2F skip
017   : 03 dif (24 Bit Integer/Binary Instantaneous value)
018   : 6E vif (Units for H.C.A.)
019 C!: 000000 ("current_consumption_hca":0)
022  ?: 0F manufacturer specific data 100001998D7D2CED08AB117D2C0000000000000000000000
023   : *** 10 frame identifier OK
024 C!: *** 0001 status
026 C!: *** 998D time
028 C!: *** 7D2C date 2c7d
030 C!: *** ED08 ("average_ambient_temperature_c":22.85)
032 C!: *** AB11 ("max_ambient_temperature_c":45.23)
034 C!: *** 7D2C max date 2c7d
036 C!: *** 0000 num measurements 0
038 C!: *** 0000 ("average_ambient_temperature_last_month_c":0)
040 C!: *** 0000 ("average_heater_temperature_last_month_c":0)
042 C!: *** 0000 indication u 0.000000
044 C!: *** 000000 total indication u 0.000000

{
    "media":"heat cost allocation",
    "meter":"hydroclima",
    "name":"",
    "id":"78328752",
    "current_consumption_hca":0,
    "average_ambient_temperature_c":22.85,
    "max_ambient_temperature_c":45.23,
    "average_ambient_temperature_last_month_c":0,
    "average_heater_temperature_last_month_c":0,
    "timestamp":"2022-05-05T20:38:59Z"
}
```

Kosygor, 2022-05-07:
> Fortunately BMeters has actually
> provided me with a document that describes the format of the extra data!

Is there any chance that You also have docs concerning BMeters Hydrocal-2 ?

weetmuts, 2022-05-07:
@Kosygor Hmm, I have docs for Hydrocal-MR ver 1.8 perhaps that is close enough?
The current hydrocalm3 driver was written before I had the docs, so the driver is to be improved.

Please open a new issue for that meter and post some telegrams and we can check it out.

BIBOLV, 2023-01-02:
Closing issue as solution was provided.

closed 2023-01-02 by BIBOLV