wmbusmeters/wmbusmeters #536: Not receiving any telegrams from a Sensus iPerl water meter

zredlined, 2022-05-07:
New member to the RTL/wmbus community here. I have a nesdr smart 2832U SDR, connected to a Ubuntu VM running on Windows 11/WSL2 via usbipd (I know). From everything I can see, wmbus and rtl_sdr are running correctly. My goal is to capture water usage from my Sensus iPerl water meter to detect leaks with our sprinkler system.

Any tips or tricks would be appreciated. I've let it run for about 2 hours yet and no telegrams received from any devices. I'm not sure how to check if wireless mbus or what protocol my meter is using, and also how to debug whether I'm running the right command (other than searching #issues). I've called the city (nothing) and also looked up the reference specs that I can find- looks like the iPerl supports wmbus and another protocol called sensusRF or flexnet.

Thanks!

Discovering the device via lsusb
 ```
~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Here's the command I'm running to launch the scan:
`~# sudo wmbusmeters --debug rtlwmbus:c1,t1`

Program output
```
~# sudo wmbusmeters --debug rtlwmbus:c1,t1
(wmbusmeters) version: 1.7.0-65-g3282c19
(config) number of meters: 0
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:   -f frequency_to_tune_to [Hz]
    [-s samplerate (default: 2048000 Hz)]
    [-d device_index (default: 0)]
    [-g gain (default: 0 for auto)]
    [-p ppm_error (default: 0)]
    [-b output_block_size (default: 16 * 16384)]
    [-n number of samples to read (default: 0, infinite)]
    [-S force sync output (default: async)]
    filename (a '-' dumps samples to stdout)

<<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 18874 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 18876 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(main) opening rtlwmbus
Started auto rtlwmbus[00000001] listening on c1,t1
(rtlwmbus) using command: rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 00000001
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serial) EVENT thread interrupted
(serialcmd) opened /bin/sh pid 18882 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) TIMER thread interrupted
(serial) closing 1 devices
(shell) sending SIGTERM to process group 18882
(serial) EVENT thread interrupted
(serial) event loop stopped!
(bgshell) sent sigterm, now waiting for child 18882 to exit.
(bgshell) 18882 terminated due to signal 15
(wmbus) disconnected 00000001 rtlwmbus
(serialcmd) closed /bin/sh pid=18882 fd=5 (rtlwmbus)
(wmbus) deleted rtlwmbus
(serial) closing 1 devices
```

TiddlyWiddly, 2022-08-16:
Hi, I'm not OP but I have the same water meter and same dongle and same issue. From what I can tell, I think my meter runs at 433Mhz. I ran the command you gave as "rtl_sdr -f 433M -s 1600000 - 2>/dev/null | rtl_wmbus -p s" but no joy. 

Pic:

[image]

BIBOLV, 2023-01-02:
Closing due to inactivity for now. Please reopen if you need further help.

closed 2023-01-02 by BIBOLV

guibar, 2024-11-23:
I have the same problem. From what I can understand, there might be a telegram per hour sent with the wmbus protocol. Other messages are sent every 15 seconds using the SensorRF proprietary protocol. As far as I can tell, there is no easy way to decode those messages.