wmbusmeters/wmbusmeters #537: Apator 17-2

akanegally, 2022-05-07:
Hello, 
I have the water meter apator at-wmbus-17-2.

On my meter (cf [my_meter.jpg](https://www.amazon.fr/clouddrive/share/Rp80OYVayyr4ZXKlkdpTd5M17xFtcn6ENczHMmk1dDj)), the serial is 1353735.
So I think the id of my telegram is 0014A807.

I try to capture the telegrams of my meter throught this command
./wmbusmeters --debug auto:t1 watermeter apator162 0014a807 NOKEY

but It doesn't seems to work. Those telegrams are maybe encrypted.
Here are all the telegram of my meter ([my_meter.log](https://www.amazon.fr/clouddrive/share/cLRgC5BqblpGeVRAIzv4S5UpoeZDW3iWtcZzEkHtA05)) with this command :

I tried to ask for the encryption key but the hotline of my provider was clueless.
Is there still a way to get my consumption ?

Many thanks for your help.

weetmuts, 2022-05-26:
Thanks, your meter does not seem to be encrypted.
What is the expected value, is it perhaps 0.823667 m3 ?

weetmuts, 2022-05-26:
Ok, this is a bit unclear. Can you run this command:
wmbusmeters --logtelegrams auto:t1 2>&1 | tee my.log

Then start using some water so that the meter value increases. After you think
you have 10 telegrams from your meter when you are using water, and you
have read off the starting value and the end value from the meter itself. 

Then post the lines in my.log that look like telegram=|...|
here! Thanks!

weetmuts, 2022-05-26:
And post the start and ending meter readings here as well. :-)

akanegally, 2022-05-27:
Hello Frédéric
I m currently on hollidays since the 14 may
Be sure I will check my meter when I come back the 3 June.

BIBOLV, 2023-01-02:
Closing due to inactivity. Please reopen issue if you manage to capture more telegrams.

closed 2023-01-02 by BIBOLV