wmbusmeters/wmbusmeters #542: build error std::numeric_limits

Mk-arc, 2022-05-13:
I cant seem to build the latest github version:
```
make
Building 1.7.0-69-g6fdc6d2
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Ibuild src/meters.cc -c -E > build/meters.o.src
g++ -Os -g -DFUZZING=false -fPIC -std=c++11 -Wall -Werror=format-security -Ibuild src/meters.cc -MMD -c -o build/meters.o
src/meters.cc: In member function ‘virtual double MeterCommonImplementation::getNumericValue(FieldInfo*, Unit)’:
src/meters.cc:1722:21: error: ‘numeric_limits’ is not a member of ‘std’
 1722 |         return std::numeric_limits<double>::quiet_NaN(); // This is translated into a null in the json.
      |                     ^~~~~~~~~~~~~~
src/meters.cc:1722:36: error: expected primary-expression before ‘double’
 1722 |         return std::numeric_limits<double>::quiet_NaN(); // This is translated into a null in the json.
      |                                    ^~~~~~
src/meters.cc:1722:36: error: expected ‘;’ before ‘double’
 1722 |         return std::numeric_limits<double>::quiet_NaN(); // This is translated into a null in the json.
      |                                    ^~~~~~
      |                                    ;
src/meters.cc:1722:42: error: expected unqualified-id before ‘>’ token
 1722 |         return std::numeric_limits<double>::quiet_NaN(); // This is translated into a null in the json.
      |                                          ^
make: *** [Makefile:126: build/meters.o] Error 1
```

weetmuts, 2022-05-13:
Great! Let me know if it compiled properly and I will push the fix.

Mk-arc, 2022-05-13:
Yes it finished building and works!

weetmuts, 2022-05-13:
Super! Fix pushed.

closed 2022-05-13 by weetmuts