wmbusmeters/wmbusmeters #544: Can't start add-on after reinstall

piio, 2022-05-14:
Hi,

I had to remove and add again wmbusmeters addon, but now I have problem with starting it:

s6-overlay-suexec: fatal: can only run as pid 1

piio, 2022-05-14:
I found similar case:
https://community.home-assistant.io/t/latest-base-image-with-s6-overlay-v3-breaks-add-on/421570

It look's like there was some changes with base image:
https://developers.home-assistant.io/blog/2022/05/12/s6-overlay-base-images/

bratmat, 2022-05-14:
Same problem

weetmuts, 2022-05-15:
Thanks @piio I followed your second link and added "init" : false 
to the config.json and pushed. I checked that the run.sh is executable. Please test if this is enough.

piio, 2022-05-15:
I have removed and installed addon and it runs correctly, so I think that this problem is resolved. Thanks.

closed 2022-05-15 by weetmuts