wmbusmeters/wmbusmeters #546: Support for Qundis Qheat 5.5 US

flozsc, 2022-05-16:
I have a QUNDiS Qheat 5.5 US meter, which identifies as Landis Gyr, however. I tried it with qheat driver, but no success.

Here's the data I got in the log:
(meter) Waermeverbrauch: meter detection did not match the selected driver qheat! correct driver is: unknown!
(meter) Not printing this warning again for id: 70835484 mfct: (LUG) Landis Gyr, Germany (0x32a7) type: Heat meter (0x04) ver: 0x0a
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|9644A732845483700A047A89000020_0274F000046D2C14D0250C0682910000446D3B17BF2C4C068251000084016D3B17DE248C010605900000CC0106988500008C020629770000CC0206226600008C030682510000CC0306933600008C040602260000CC0406691800008C050633140000CC0506900900008C060618020000CC0606000000008C0706000000003C22000000000F001000|+69
telegram=|9644A732845483700A047A8A000020_02746601046D2C14D0250C0682910000446D3B17BF2C4C068251000084016D3B17DE248C010605900000CC0106988500008C020629770000CC0206226600008C030682510000CC0306933600008C040602260000CC0406691800008C050633140000CC0506900900008C060618020000CC0606000000008C0706000000003C22000000000F001000|+186
telegram=|9644A732845483700A047A8B000020_0274DC01046D2C14D0250C0682910000446D3B17BF2C4C068251000084016D3B17DE248C010605900000CC0106988500008C020629770000CC0206226600008C030682510000CC0306933600008C040602260000CC0406691800008C050633140000CC0506900900008C060618020000CC0606000000008C0706000000003C22000000000F001000|+303

Is there any chance to get this added, please?

weetmuts, 2022-05-21:
Thanks! When I run analyze on the telegram it does not look much like the qheat driver.

https://wmbusmeters.org/analyze/9644a732845483700a047a890000200274f000046d2c14d0250c0682910000446d3B17Bf2c4c068251000084016d3B17de248c010605900000cc0106988500008c020629770000cc0206226600008c030682510000cc0306933600008c040602260000cc0406691800008c050633140000cc0506900900008c060618020000cc0606000000008c0706000000003c22000000000f001000:qheat

We can add a new driver for this one. Can you help out in guessing what the red values mean?
Is it 14 months of old measurements? What is the duration? Etc.

flozsc, 2022-05-22:
Hi, thanks for your response!

Meanwhile, I did play around with the source code and built something that works for me: https://github.com/flozsc/wmbusmeters/blob/8e90d7a83acb967fab225a0b34628f580d76de2c/src/driver_qheat_55_us.cc

I assume that there'd be certain quality standards to apply before merging this with your codebase and please bear with me being a total newbie when it comes to create pull requests, etc.

- For reference, I attached a photo of my device.
- I currently implemented read-out of storagenr 1 & 2, as they had datetime information to relate the data to.
- According to the [data sheet](https://qundis.de/wp-content/uploads/2021/01/T330_DST2_QHEA_EN0_USLG5_V411_260121_FINAL.pdf), the other values are monthly readings. I can add them as attributes.
- Is there anything else you would need me to do before creating a pull-request? Any guidance?

[image]

flozsc, 2022-05-22:
I added the past 13 monthly readings as well: https://github.com/flozsc/wmbusmeters/blob/249e97d55027bce3c776bea16113351a96969045/src/driver_qheat_55_us.cc

weetmuts, 2022-05-25:
Great work! But you can now make the driver even simpler! 

I have managed to remove the need for local variables in the class. 
(And the class can be named Driver if an empty namespace { surrounds everything.)

 Check driver_c5isf.cc for an example!

flozsc, 2022-05-26:
Cool! Much easier indeed!

See: https://github.com/flozsc/wmbusmeters/blob/a7008a9365265c93484f8cd2ac39852e387a20bc/src/driver_qheat_55_us.cc

Seems to be working well in my environment. Would you consider it to be ready for a pull request?

andrasj, 2022-09-25:
I've got simiar issues with a Qunids QHeat 5.5 US. (running from homeassistant Addon on RPi, not really sure how to pull to test new versions :(
Below some additional telegrams.
Is there any chance the current temperature is available from these messages?

> No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 71466672
          manufacturer: (LUG) Landis Gyr, Germany (0x32a7)
                  type: Heat meter (0x04)
                   ver: 0x0a
                device: rtlwmbus[00000001]
                  rssi: 151 dBm
                driver: unknown!
telegram=|_9644A732726646710A047AE800002002747801046D2715D9290C0668100000446D000000004C060000000084016D3B17DF288C010609030000CC0106950000008C020600000000CC0206000000008C030600000000CC0306000000008C040600000000CC0406000000008C050600000000CC0506000000008C060600000000CC0606000000008C0706000000003C22000000000F001000|+60
Received telegram from: 71466675
          manufacturer: (LUG) Landis Gyr, Germany (0x32a7)
                  type: Heat meter (0x04)
                   ver: 0x0a
                device: rtlwmbus[00000001]
                  rssi: 139 dBm
                driver: unknown!
telegram=|_9644A732756646710A047AE90000200274F401046D2615D9290C0639120000446D000000004C060000000084016D3B17DF288C010696020000CC0106780000008C020600000000CC0206000000008C030600000000CC0306000000008C040600000000CC0406000000008C050600000000CC0506000000008C060600000000CC0606000000008C0706000000003C22000000000F001000|+170
Received telegram from: 71466672
          manufacturer: (LUG) Landis Gyr, Germany (0x32a7)
                  type: Heat meter (0x04)
                   ver: 0x0a
                device: rtlwmbus[00000001]
                  rssi: 140 dBm
                driver: unknown!
telegram=|_9644A732726646710A047AEA00002002747002046D2715D9290C0668100000446D000000004C060000000084016D3B17DF288C010609030000CC0106950000008C020600000000CC0206000000008C030600000000CC0306000000008C040600000000CC0406000000008C050600000000CC0506000000008C060600000000CC0606000000008C0706000000003C22000000000F001000|+305
Received telegram from: 71466675
          manufacturer: (LUG) Landis Gyr, Germany (0x32a7)
                  type: Heat meter (0x04)
                   ver: 0x0a
                device: rtlwmbus[00000001]
                  rssi: 149 dBm
                driver: unknown!
telegram=|_9644A732756646710A047AEB0000200274EC02046D2615D9290C0639120000446D000000004C060000000084016D3B17DF288C010696020000CC0106780000008C020600000000CC0206000000008C030600000000CC0306000000008C040600000000CC0406000000008C050600000000CC0506000000008C060600000000CC0606000000008C0706000000003C22000000000F001000|+415

flozsc, 2022-09-26:
Let me sit down tonight or latest tomorrow and make a formal pull request out of it. Then it will be easy for @weetmuts to merge it.

andrasj, 2022-09-26:
Would be wonderful, but I'm not in a hurry, take your time.
Any Idea about the temperatures measured by the devices? According to the manual LOOP1 contains the current temperature, flow and energy info. Any idea if they are/can be encoded in the dataframes?

flozsc, 2022-09-27:
Created the PR for my change. However, the temperatur and flow are not provided. Only consumption.

The contents are listed in [Datasheet](https://qundis.com/wp-content/uploads/2022/09/Heat-meter-Q-heat-5.5-US_DST2_QHEA_EN0_USLG5_V411_06092022.pdf), p. 5, "Data telegram content"

weetmuts, 2022-09-28:
I added the two remaining fields. Now it decodes all data except the last few mfct specific bytes.

https://wmbusmeters.org/analyze/9644a732845483700a047a890000200274f000046d2c14d0250c0682910000446d3B17Bf2c4c068251000084016d3B17de248c010605900000cc0106988500008c020629770000cc0206226600008c030682510000cc0306933600008c040602260000cc0406691800008c050633140000cc0506900900008c060618020000cc0606000000008c0706000000003c22000000000f001000

BIBOLV, 2023-01-02:
Closing issue as solution was provided.

closed 2023-01-02 by BIBOLV