wmbusmeters/wmbusmeters #547: Kamstrup Pressure Meters (Driver)

LarsBorris, 2022-05-19:
Hello Frederik,

I was hoping that I could read the values of a Kamstrup PressureMeter using your application, sadly it does not work.
Is there any chance that you could add the driver? I am able to supply you with everything you need. Please let me know.
Thanks!

weetmuts, 2022-05-19:
Sure can! Just run:
wmbusmeters --logtelegrams auto:t1 MyPressureMeter amiplus 12345678 <key>
or NOKEY if it is not encrypted.
The json will not be correct, but that is fine.
Just post the lines with: telegram=|.......|
here. These are the decrypted telegrams, that I can use to create a new driver.

LarsBorris, 2022-05-19:
Hello Frederik,

Thank you so much.
Here you go:
telegram=|27442D2C1703007701188D28FE48E39322E8F179DE735DD3_660066006600E007883AA7F738BA0800|+2737
telegram=|27442D2C1703007701188D28FF6CE39322903479DE73B844_6600670066002200803A7EB890B70800|+2836
telegram=|32442D2C1703007701188D280080E39322DB8F78_22696600126967000269660005FF091954A33A05FF0A99BD823A02FD170800|+2935
telegram=|27442D2C1703007701188D280194E393226EC679DE735657_660067006600962B913A21B9423A0800|+3034
telegram=|27442D2C1703007701188D2802B8E393223EE179DE731E25_6600660066008F638F3A74CF923A0800|+3133
telegram=|27442D2C1703007701188D2803DCE393227BA279DE7398D9_660066006600CD2E743A5B596F3A0800|+3232
telegram=|27442D2C1703007701188D2804E0E393227A1879DE73974E_660067006600A83F8F3A1AAF5E3A0800|+3331
telegram=|27442D2C1703007701188D280504E493226CAD79DE73D2FF_660066006600523D823A4E596F3A0800|+3429

weetmuts, 2022-05-21:
Thanks! I pushed a driver. You can test it here:
https://wmbusmeters.org/analyze/32442d2c1703007701188d280080e39322dB8f7822696600126967000269660005ff091954a33a05ff0a99Bd823a02fd170800

Does the pressure look right?
Do you know what the two real numbers are?
There seems to be an error bit set. Do you know what it means, can you figure out what the other bits mean? From a manual or something. :-)

LarsBorris, 2022-05-25:
Hello Frederik!

I had a chance to test the driver.
Here are the results:
The PressureMeter shows 0.01 bar since it is currently sitting on my desk.

Here are the contents of my log files:
**wmbusmeters.log:**
(wmbusmeters) logging started 2022-05-19 23:14:36 using _
[2022-05-19_23:14:36] Started config amb8465[00000000] on /dev/ttyUSB0 listening on c1,t1
(wmbusmeters) waiting for telegrams
[2022-05-19_23:14:38] (memory) rss 1908736 peak 1.82 MiB
(wmbusmeters) started meter 1 (PressureSensor_77000316 77000316 auto)
[2022-05-19_23:15:00] (meter) newly created meter (PressureSensor_77000316 77000316 kampress) did not handle telegram!
(wmbusmeters) started meter 2 (PressureSensor_77000317 77000317 auto)
[2022-05-19_23:16:08] (meter) newly created meter (PressureSensor_77000317 77000317 kampress) did not handle telegram!
telegram=|32442D2C1603007701188D2817CCF6952213AE78_22696600126966000269660005FF09C142CB3A05FF0AAE5042BA02FD170800|+122
telegram=|27442D2C1703007701188D289208F29522C8A379DE73E249_6500660066002ECC933A70995C3A0800|+186
telegram=|27442D2C1603007701188D2818E0F69522659E79DE733B56_660066006600184FE53ACDB9C13A0800|+220
telegram=|27442D2C1703007701188D28932CF295229DA679DE734E2A_6500660066000A07923AE477A73A0800|+280
telegram=|27442D2C1603007701188D281904F7952205A179DE73169D_650066006600FE0ECA3A2403883A0800|+318
telegram=|27442D2C1703007701188D289440F29522BA0C79DE73339F_65006600660099D0C63A1025D8BA0800|+374
telegram=|27442D2C1603007701188D281A18F795227A0679DE732CE3_6500660066001607923A14D4A0BA0800|+416
telegram=|27442D2C1703007701188D289554F295224ED579DE73188A_650066006600E80EA43A6B97A3BA0800|+468

**PressureSensor_77000316:**
name=PressureSensor_77000316
id=77000316
key=################################

**PressureSensor_77000316-77000316_2022-05-19:**
{"media":"pressure","meter":"kampress","name":"PressureSensor_77000316","id":"77000316","pressure_bar":1.02,"max_pressure_bar":1.02,"min_pressure_bar":1.01,"status":"ERROR_FLAGS_0","timestamp":"2022-05-19T21:21:32Z","device":"amb8465[00000000]","rssi_dbm":-98}

As you can see from the data, the pressure_bar is not correct.
Otherwise it seems to work!

Thank you so much for all your work.

LarsBorris, 2022-05-25:
Hello!

I just realised that the display and the wmbus measurement is not the same. The display is set to show relative pressure and the wmbus measures the actual pressure. So the driver works! Thanks! Always a pleasure.

closed 2022-05-25 by LarsBorris

weetmuts, 2022-05-25:
Great! There are two more proprietary floating point values in the telegram, is there anything else in the meter that
could refer to those two?

weetmuts, 2022-05-25:
Also, do you have a list of the error conditions the meter can report? I would like to make the status reporting better.

weetmuts, 2022-05-25:
I think I found the error flags in the documentation, bit 8 means LOW pressure, which makes sense since your sensor is on your desk. The floating point values are: 
0.001252
-0.001248

Do they ring a bell?