wmbusmeters/wmbusmeters #548: which link mode ?

maijo366, 2022-05-22:
I have wmbusmeters in homeassistant running. 

When i write this in the config:

data_path: /config/wmbusmeters
conf: |-
  loglevel=debug
  device=/dev/ttyAMA0
  logtelegrams=true
  format=json
  meterfiles=/config/wmbusmeters/meter_readings
  meterfilesaction=overwrite
  meterfilesnaming=name
  meterfilestimestamp=day
  logfile=/config/wmbusmeters/wmbusmeters.log
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters: []
mqtt: {}

This are the logfiles.
Wmbus devices found but no meters supplied. You must supply which link modes to listen to. Eg. auto:c1

An when i write device=auto:c1,t1

I become this in the log:

(wmbusmeters) version: 1.7.0-72-gca264b3
(config) store meter files in: "/config/wmbusmeters/meter_readings"
(config) number of meters: 0
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) check if /dev/ttyAMA0 can be accessed
(serial) tty /dev/ttyAMA0 can be accessed
(serial) before config: /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,3,254) modem(DTR|RTS|CTS)
(serial) EVENT thread interrupted
(serial) after config:  /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,109,207,217,127,248) modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0

Please help i am a newbe. Google no answer ;-)

weetmuts, 2022-05-25:
Which wmbus dongle are you using?

maijo366, 2022-05-26:
I use the: TenYua RTL2832U+R820T2 Digital DVB-T SDR + DAB+ FM HDTV TV Tuner Receiver Stick USB 2.0.

weetmuts, 2022-05-26:
Check that you have /usr/bin/rtl_sdr and /usr/bin/rtl_wmbus installed.
Change: device=/dev/ttyAMA0
to: device=rtlwmbus:c1,t1,s1

The default setting for rtlsdr/rtlwmbus is to listen to c1,t1 and s1 at the same time.
You can limit this to only c1,t1 (if you need to reduce cpu usage) using a special CMD(...) check the README
for more info.

maijo366, 2022-05-28:
Thanks for the quick reply. 
It works now. I have received data from my water meter. see log.

Received telegram from: 77634257
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Cold water meter (0x16) encrypted
                   ver: 0x1b
                device: rtlwmbus[00000001]
                  rssi: 30 dBm
                driver: multical21
telegram=|_23442D2C574263771B168D20CC405E3221763E0468BD6ABBE316CEEA56B1B76B105648E2|+705

However, when I create the meter files, it does not recognize my key. see log.

(wmbus) WARNING! decrypted payload crc failed check, did you use the correct decryption key? cdf7 payload crc (calculated 94dc) Permanently ignoring telegrams from id: 77634257 mfct: (KAM) Kamstrup Energi (0x2c2d) type: Cold water meter (0x16) ver: 0x1b
(meter) newly created meter (Hauptwasserzaehler_MC21 77634257 multical21) did not handle telegram!

i received a .kem file from the water supplier. how can i read the file or the key. This is the file.

weetmuts, 2022-07-01:
```
cp 1122334455.kem 1122334455.zip
unzip 1122334455.zip -> afile

cd utils

javac XMLExtract.java
java -cp . XMLExtract [kemfilepassword] afile

weetmuts, 2022-08-28:
Can we close this now?

maijo366, 2022-08-29:
Yes, thanks for your help.

I received the key from my water supplier. Now it works

closed 2022-08-30 by weetmuts