wmbusmeters/wmbusmeters #549: Remove temperature difference sharky_774

weetmuts, 2022-05-25:
Hi @Germar @don-vip !

I am refactoring the drivers into a new very clean declarative format. (Check driver_iperl.cc for example.)
However, when I was refactoring sharky_774 I noticed that there is a value that is calculated, 
the "temperature_difference_c"

I would like to remove this field, since it is sort of outside of the scope of wmbusmeters
to do calculations on the data.

The next layer of data analysis should be able to easily calculate such things.

Wmbusmeters should focus on extracting, translating, normalizing  values passed in the telegram
but not actually do any further calculations.

So, my question to you, are you using temperature_difference_c, can I remove it?

don-vip, 2022-05-25:
Ah, I wrote support for Sharky 775 then this pull request added support for Sharky 774:
https://github.com/weetmuts/wmbusmeters/pull/425
If 774 transmits the difference as 775 does, then the same code can be used :) I agree with you Wmbusmeters should not compute values by itself.

DomAtHome, 2022-05-25:
Yeah, it can be deleted. I've based the code for Sharky 774 on 775. The 774 doesn't send the temp difference but it could be calculated in the class so I implemented it.

Totally agree that the temp difference should be calculated on data analysis layer.

weetmuts, 2022-12-29:
The difference is now removed from the driver and the whole driver has been rewritten to the latest format. However, wmbusmeters now also supports calculation of formulas. So the field can be recreated with:

```
wmbusmeters --format=json --ppjson --calculate_temperature_difference_c='flow_temperature_c-return_temperature_c' 3E44A5110564495841047A700030052F2F_0C06846800000C13195364000B3B0400000C2B110100000A5A17050A5E76020AA61800004C0647630000426CBF25 MyMeter sharky774 58496405 NOKEY
```

The formula calculator tracks the units so if you try to add incompatible units then the resulting value will be null.

weetmuts, 2022-12-29:
Note that since the field disappeared, the default fields would change anyway, so I simplified the
default fields significantly. Please use --selectfields=.... to pick the fields you want, if you use --format=fields

For json there is no change except that it disappeared.

closed 2022-12-29 by weetmuts