wmbusmeters/wmbusmeters #550: MQTT discoverable? Feature request

Bluhme1, 2022-05-26:
Using Home Assistant 
I have made a manual sensor in configuration. yaml for my Multical 21 watermeter and everything works fine
Would it be possible to make the sensor discoverable? What should be added to the shell=mosquito_pub command.
Thank you 
/Lars

flozsc, 2022-05-26:
Do I get you right that you'd like wmbusmeters to use [MQTT Discovery](https://www.home-assistant.io/docs/mqtt/discovery/) ?
I thought about the same but ended up using configuration.yaml.

However, I believe this would be a cool addition to make it more plug&play and energy metering more accessible. I'll try to fiddle with this a bit.

Bluhme1, 2022-05-26:
Looking forward 👏👏

flozsc, 2022-05-26:
Initial thought is to modify ha-addon/run.sh

- use reasonable default for state topic compatible with discovery (how to deal with custom prefix?)
- read each meter's configuration and publish setup message to configuration topic

However, I see issues with 
- value_template: meter configuration doesn't reveal information which JSON key to use as "main" information.
- unit_of_measurement: same here. We don't know yet, what unit we'll get.

I think the add-on configuration already is quite complex already and adding this information would make it even harder to understand.

Bluhme1, 2022-05-27:
You are probably right.
Isn’t there another way of doing this.
Looking at the documentation for MQTT discovery there is a possibility of adding to the Mosquito_pub command line. And we use that command as a shell= xx in the configuration of the add-on. Couldn’t we just modify this? I don’t know how as I don’t fully understand the documentation but maybe you do?
https://www.home-assistant.io/docs/mqtt/discovery/
/Lars

jnxxx, 2022-07-27:
I have made an attempt to implement this.
It is currently placed here: https://github.com/jnxxx/wmbusmeters/tree/master/ha-addon
If you find it useful I will make a pull request, to see if it can make it into the original repository.

From run.sh is calls an additional bash script.
This script reads the configured meters, and in particular the driver, ex. multical21.
Then it looks up a template file for the driver/type, substitutes some values and then publishes the config data in MQTT.

For it to work:
1. Meter must be defined with a specific driver (not auto, which isn't recommended for production anyway)
2. A template file must have been defined for the driver type.
3. MQTT discovery must be enabled in the config file.

The template file contains information which tells HA how to interpret and display the payload. Thus, one need to be defined for driver type. Currently only multical21 is defined, as it is the only meter I have a key for.

[image]

[image]

[image]

Bluhme1, 2022-07-27:
Looks great. I am in the house at the beach right now. But I will try as soon as I get home.
I have the Multical 21 but also the heat meter Kamstrup Multical 603.

jnxxx, 2022-07-28:
Well, as it runs in a container it is a bit cumbersome, you cannot just overwrite the files...

I think the easiet way would be to add the repository [https://github.com/jnxxx/wmbusmeters](https://github.com/jnxxx/wmbusmeters) like you added the original one.
Then stop the first instance, and copy its configuration file to the second instance. Enable MQTT discovery and start it.
Check the log to make sure MQTT discovery runs, and then you ought to see the Multical 21 device appear.

The templates folder is located in: `~/config/wmbusmeters/etc/mqtt_discovery`
when seen from VS code or the terminal.

You can then give it a try with a file for Multical 603, or you can send me an example of the payload.
I have this meter too, but await my key.

Bluhme1, 2022-07-28:
Great. I try when I get home during the weekend. We can perhaps work together with the template file for Kamstrup 603. I got the key from Odder Varmeforsyning and installed it already in the exiting add on.
And in Danish: tak for indsatsen. Endnu et skridt på vejen til at gøre energimåling mere brugervenligt og dermed være med til at fremme den grønne omstilling.

Bluhme1, 2022-07-31:
Returned home and installed and everything works fine for my Multical21. So much easier than the manual method. Therefore much more suitable for a broader user base and this is important when we wish to save on the resources of the Earth. The more people knowning their actual use of the resources the more likely it is that we can change our consumption in our daily life.
Thank you again for your effort. Hope that Fredrik (weetmuts) will accept your PR.
Concerning the multical603 (heat meter for district heating) I see a great potential, too.
Here is my payload from multical603:
{
    "media": "heat volume at inlet",
    "meter": "multical603",
    "name": "MyHeatMeter",
    "id": "80966656",
    "total_energy_consumption_kwh": 7078,
    "total_volume_m3": 264.4,
    "volume_flow_m3h": 0.032,
    "t1_temperature_c": 52.92,
    "t2_temperature_c": 27.75,
    "at_date": "",
    "current_status": "",
    "energy_forward_kwh": 15449,
    "energy_returned_kwh": 9250,
    "timestamp": "2022-07-30T19:09:21Z",
    "device": "im871a[00102280]",
    "rssi_dbm": -74
}

I , however, have a problem as Wmbusmeters doesn't recognize this meter and uses an "unknown" driver instead. See there log:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Syncing wmbusmeters configuration ...
Registering meters ...
Adding meter-0001 ...
Adding meter-0002 ...
Generating MQTT configuration ... 
Broker 192.168.86.152 will be used.
MQTT Discovery ...
 Adding meter: MyHeatMeter ...
  File /config/wmbusmeters/etc/mqtt_discovery/multical603.json not found.
 Adding meter: MyTapWater ...
  Add/update topic: homeassistant/sensor/wmbusmeters/58283912_external_temperature_c/config
  Add/update topic: homeassistant/sensor/wmbusmeters/58283912_flow_temperature_c/config
  Add/update topic: homeassistant/sensor/wmbusmeters/58283912_max_flow_m3h/config
  Add/update topic: homeassistant/sensor/wmbusmeters/58283912_rssi_dbm/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/58283912_status_burst/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/58283912_status_dry/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/58283912_status_leak/config
  Add/update topic: homeassistant/binary_sensor/wmbusmeters/58283912_status_reversed/config
  Add/update topic: homeassistant/sensor/wmbusmeters/58283912_target_m3/config
  Add/update topic: homeassistant/sensor/wmbusmeters/58283912_total_m3/config
MQTT Discovery cleanup...
Running wmbusmeters ...
Started auto im871a[00102280] on /dev/ttyUSB0 listening on c1
(meter) MyHeatMeter: meter detection did not match the selected driver multical603! correct driver is: unknown!
(meter) Not printing this warning again for id: 80966656 mfct: (KAM) Kamstrup Energi (0x2c2d) type: Heat volume at inlet meter (0x0c) ver: 0x35
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) newly created meter (MyHeatMeter 80966656 multical603) did not handle telegram!

I have reached out to Fredrik to have this meter recognizable in the Wmbusmeters Home Assistant add-on

Looking forward to hearing from you

/Lars

jnxxx, 2022-07-31:
Try to restart the addon now, and see if Multical 603 appears.

I am uncertain what energy_forward and returned is and how important they are. For now I have disabled them, but you can enable them from the UI if you like.

I noticed Multical 603 can send some more status codes than the driver currently reads. Among others I find it quite interesting for it to read leak and burst status.
I will make a feature request for this later.

Bluhme1, 2022-07-31:
It works 👍👍

Bluhme1, 2022-08-01:
Energy_forward (also called E8 in the Kamstrup guides) and energy_returned (called E9) can be very useful over time. The so-called "motivationsbidrag" is calculated once on year on the basis of the registers. So keeping an eye on these registers to make sure that the are as low as possible on a yearly basis can save you some money and of course also the resources of the Planet.
I have added a technical description of the Kamstrup Multical603 and the the registers are mentionen on page 100 including the math
[55122028_K1_DA.pdf](https://github.com/weetmuts/wmbusmeters/files/9232838/55122028_K1_DA.pdf)

jnxxx, 2022-08-01:
Thanks 👍
Unfortunately GUES in Bjerringbro will not hand out the key... 😟

Bluhme1, 2022-08-01:
I tried that in Odder also. But I am rather stubborn and insisted on a dialog with both the management and the board of Odder Forsyning with the arguments that knowing your data (which you own according to the GDPR legislation) can change behaviour and therefore help with the green transition.
Odder Forsyning only saw the data as a means to collect money and it took som time to persuade them to think otherwise.
I had to promise that I will pay for a new meter when we move away from the house. (It is rather silly but necessary).
You are welcome to use Odder Forsyning as an example that it can be done.
Wish you a good "fight"

convicte, 2023-01-08:
Hi @jnxxx, hi all,

I've made some attempts to implement the relatively popular IZAR driver into the MQTT auto-discovery, but I ran into a problem.

When I enable the MQTT auto-discovery in my config, the following error occurs:
```
MQTT Discovery ...
/mqtt_discovery.sh: line 44: kv[1]: unbound variable
[01:00:23] INFO: Running wmbusmeters ...
```
I've checked line 44, but was unfortunately unable to follow the code logic, and gave up. 
Would someone more familiar mind having a look at it?

Here my config, just in case:
[image]

I'll be opening a separate feature request for the addition of the IZAR

Thank you in advance.

jnxxx, 2023-01-09:
Check your defined meters.
Do any of them have a key/value without an empty value?

convicte, 2023-01-10:
Hi @jnxxx, 
Thank you for looking into it. I am a bit confused by the double negative statement.
What do you mean by 'without an empty value'?

My meters have an ID and the key for them is `NOKEY` per documentation.
They themselves work just fine in this configurations.
Should I change the `NOKEY` term to a different syntax?

Outside of it there are no other meters and/or values beyond what you see above.

Please excuse me if I missed your point, in which case I would greatly appreciate a clarification.

convicte, 2023-01-12:
Hi @jnxxx, would you mind offering some ideas or next troubleshooting steps.

I feel I am stuck and would really love to test the auto-discovery I've put together for this meter.

Thank you in advance!

jnxxx, 2023-01-13:
I cannot reproduce it.

I copied your template file, and configured a meter as follows:

```
data_path: /config/wmbusmeters
enable_mqtt_discovery: true
conf: >-
  loglevel=normal

  device=/dev/serial/by-id/usb-FTDI_2605056083001_WE_USB_Plug_2701E2E7-if00-port0:amb8465:c1

  logtelegrams=true

  format=json

  logfile=/dev/stdout

  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=water_meter
    driver=izar
    id=123
    key=NOKEY
mqtt: {}

```

When I start up:
[image]

The resulting device:
[image]

convicte, 2023-01-13:
[image]

Having used someone else's draft config, my meters had the `key` variable called as:

`key: NOKEY` and not `key=NOKEY`

It was working just fine until I started playing with the MQTT auto_discovery.
The reason you didn't spot it is I pixelated too far into the code, for the meter number.

Thank you for sharing this example, which made me realize there is an obvious difference starring me in the face.

BIBOLV, 2023-03-13:
Can we close this issue or there is still something to discuss?

closed 2023-03-13 by Bluhme1