wmbusmeters/wmbusmeters #551: Consultation regarding shell command in wmbusmeters.conf and Home Assistant MQTT discovery

VivantSenior, 2022-05-27:
Hi,

I started working on a bash script that will send [MQTT Discovery](https://www.home-assistant.io/docs/mqtt/discovery/) information to Home Assistant.

I'm running wmbusmeters on separate RPi which communicates with HA using MQTT (can't use HA addon because of the poor 868 MHz signal strength - I have tested few antennas).

My idea was to create a script with all of the HA discovery mqtt messages and then add it to the wmbusmeters `/etc/wmbusmeters.conf` file like this:
`shell=sudo bash /home/user/scripts/publish_data.sh "$METER_ID" "$METER_JSON"`

The configuration payload needs to be dynamic because I wanted the topic to look like this:
`homeassistant/sensor/[meter ID]_total_m3/config` and therefor would need to use variable `$METER_ID`. In addition I wanted to add [unique_id](https://www.home-assistant.io/integrations/sensor.mqtt/#unique_id) to be able to modify the sensors from the HA UI.

but I got information:
```(shell) exec "/bin/sh"
(shell) arg "-c"
(shell) arg "sudo bash /home/leszek/scripts/publish_data.sh "$METER_ID" "$METER_JSON""
(shell) env "METER_JSON={"media":"water","meter":"apator08","name":"MainWater","id":"XXX","total_m3":5361.65556,"timestamp":"2022-05-27T11:32:28Z","device":"cul","rssi_dbm":-37}"
(shell) env "METER_ID=XXX"
(shell) env "METER_NAME=MainWater"
(shell) env "METER_MEDIA=water"
(shell) env "METER_TYPE=apator08"
(shell) env "METER_TIMESTAMP=2022-05-27T11:32:28Z"
(shell) env "METER_TIMESTAMP_UTC=2022-05-27T11:32:28Z"
(shell) env "METER_TIMESTAMP_UT=1653651148"
(shell) env "METER_TIMESTAMP_LT=2022-05-27 13:32.28"
(shell) env "METER_DEVICE=cul"
(shell) env "METER_RSSI_DBM=-37"
(shell) env "METER_TOTAL_M3=5361.65556"
(shell) waiting for child 10538 to complete.
(shell) /bin/sh: return code 1
[2022-05-27_13:32:28] (shell) /bin/sh exited with non-zero return code: 1
```

Return code 1 means `Operation not permitted`.

publish_data.sh looks like this:
```
#!/bin/sh

host="mqtt IP / hostname here"
port=port_num_here
username="mqtt user here"
password="mqtt pass here"

METER_ID=$1
METER_JSON=$2

/usr/bin/mosquitto_pub -h $host -p $port -u $username -P $password -t "wmbusmeters/$METER_ID/state" -m "$METER_JSON"
```

When I run script itself (from different sudo account) it works fine but not from the wmbusmeters.conf. Based on [run.sh](https://github.com/weetmuts/wmbusmeters/blob/master/ha-addon/run.sh) I changed ownership of the publish_data.sh file to user `wmbusmeters` but it didn't help and done `chmod a+x`.

When I removed `/usr/bin/mosquitto_pub` line from the script the result was the same - `return code 1`.

@weetmuts, do you have any ideas what I'm doing wrong?

CC issue number: #550.

Best,
Viv

BIBOLV, 2022-05-28:
Hi,
Could not see anything wrong, so tested your script and it works, at least on Rocky, so this could be something debian specific.
[image]
But couple suggestions: 
- Get rid of sudo.
- Get rid of external script. Not sure what full functionality is intended/needed as I`m happy with manually creating HA sensors, but if multiple mqtt messages are needed to be sent, then it can be chained as in example in image,

VivantSenior, 2022-05-29:
@BIBOLV thanks for the reply. Script works if you run it by yourself but the error appears when you will add it to the ‘SHELL=‘ line in ‘/etc/wmbusmeters.conf

VivantSenior, 2022-05-29:
Closed the issue by mistake. Reopening.

closed 2023-04-05 by github-actions[bot]