wmbusmeters/wmbusmeters #556: Little help needed with "fields"

diddi68, 2022-06-07:
Hi,
I'm using wmbusmeters since about 2 month and it works great. Thank you for that. From the beginning I used "wmbusmeter.conf" with "format=json". Now I wanted to switch to "format=fields", with some values defined in "selectfields=" but I don't know the right shell command for submitting.
I tried
`shell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/$METER_ID -m "$METER_FIELDS"`
but without success. I still get the json-string.
What is the right command for submitting "fields"?

almerindo, 2022-08-25:
Why not just use it?
shell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/$METER_ID -m "field1=$METER_TIMESTAMP; field2=$METER_ID; field3=$METER_NAME; field4=$METER_TOTAL_M3"

BIBOLV, 2023-01-02:
@diddi68 were you able to get it running as you want?

diddi68, 2023-01-02:
Yes, but in a different way. I use one curl command for every single field.

BIBOLV, 2023-01-02:
Great, closing issue for now. Please reopen if any other question/issue.

closed 2023-01-02 by BIBOLV