wmbusmeters/wmbusmeters #557: Support for hydrus 2.0 on channel c1

rikke-rp, 2022-06-08:
Hello! I just received a diehl hydrus 2.0 water meter configured for c1 rather than t1, but it appeares that it is not supported. I would love to help making a driver for this meter. 

When I set hydrus as driver:
`
(meter) MyTapWater: meter detection did not match the selected driver hydrus! correct driver is: unknown!
(meter) Not printing this warning agin for id: 72909643 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x76
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination

telegram=|6344A5114396907276078C004E900F002C25F92200007573BC8BB905B68B7AD6704107102F2F|0C13000200000B3B0000000A5A44020A6627020C933C0000000084046D3B37DF258C041300020000AB043B0000009B043B00000004FD17000800002F2F2F|+34
telegram=|6344A5114396907276078C004F900F002C250E2300000C9782A359E2F0FF7AEB704107102F2F|0C13000200000B3B0000000A5A44020A6638020C933C0000000084046D3B37DF258C041300020000AB043B0000009B043B00000004FD17000800002F2F2F|+48
`

When I set auto as driver:
`
(meter) MyTapWater: meter detection could not find driver for id: 72909643 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x76
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination

telegram=|6344A5114396907276078C0050900F002C2511230000140C9741DFC84E497AEE704107102F2F|0C13000200000B3B0000000A5A44020A6638020C933C0000000084046D3B37DF258C041300020000AB043B0000009B043B00000004FD17000800002F2F2F|+35
`

BIBOLV, 2023-01-02:
Hi,
Were you able to get it running?

https://wmbusmeters.org/analyze/6344A5114396907276078C0050900F002C2511230000140C9741DFC84E497AEE704107102F2F0C13000200000B3B0000000A5A44020A6638020C933C0000000084046D3B37DF258C041300020000AB043B0000009B043B00000004FD17000800002F2F2F
Looks like waterstarm driver decodes consumption.

rikke-rp, 2023-01-03:
No I didn't. I had to focus on other things, and now the hydrus meter is in different hands. But a combination of waterstarm and hydrus driver gives the basic infomation :)

closed 2023-01-03 by BIBOLV