wmbusmeters/wmbusmeters #559: Zenner default AES keys would be needed

4K901, 2022-06-09:
Hello,

I'm testing a simple USB receiver where I can catch the telegrams of two Zenner watermeters from my household (type - Minomess wM-Bus C1 Mode and type EDC wireless M-bus module T1 Mode)

I tried to look trough the wmbusmeters project, but can not figure out how can I find out the default AES keys for the decryption.

Herebelow the raw, encrypted telegrams from the two meters:
ID 14340330:
3644496A4187025001377230033414496A010776002025F42F261F8EFA0D14A0B28DBFF3468906D97405FB723A26F884B344CA5199B12F

ID 15496853:
6644496A1483025514377253684915496A0007C30050054512796939A3AFECD7417FC33AC83CFE38520B8B92508FBC6210248E342F6950A61933F3D4258FBDD459FFE02B5DEB5169BD5AADB62905967D24317553991DFC29E22FE46AE3013C9F92F28832CFEAE0

Sorry I'm not an expert Github and not sure if I can ask this kind of question here. If this wasn't the right place, please just drop me a line where can I ask something like this.

Thank you in advane.

BIBOLV, 2022-06-09:
There should not be default encryption keys unless it was poorly configured. About keys you can read [here](https://github.com/weetmuts/wmbusmeters/issues/446) or check other issues in this project where it was discussed.

ElKaPa, 2022-06-10:
It's quite common for some metering companies to have a global AES key per customer or masterkey. In this case it could be true that your water company got a single key and that's why they will not share it.

They order them preconfigured from Zenner or parameterize them by themselves. It's impossible to get the key in this case.

BIBOLV, 2023-01-02:
Closing issue due to inactivity. Please reopen issue if you want to continue topic.

closed 2023-01-02 by BIBOLV

Uzlopak, 2026-05-24:
Hi,

the Zenner default key is `5A8470C4806F4A87CEF4D5F2D985AB18`