wmbusmeters/wmbusmeters #560: Can't read wmbus-08, but unismart works..

immek, 2022-06-09:
Hello,

First of all thank you for this great project. I wonder could you help me with my problem. I've got this dongle ("Found Rafael Micro R828D tuner" - https://imgur.com/a/8EFjq2o ) and using the command `wmbusmeters --logtelegrams rtl433:868.95M` I was able to read data from a couple of neighbor gas meters, my included (Apator Metrix / Unimeter). A few days ago I bought two Apator WMBUS-08 devices and installed them on my water meters. The problem is I can't get them to work. Why gas meter works and water meter doesn't?

Interesting that gas meters were not working with rtl_sdr + rtl-wmbus - only with rtl_433. I don't know the real difference. 

I also don't know if it matters, but the WMBUS-08 were bought as "used" but "working", taken off by some company during a replacemnt on some bigger housing estate.. Other people that bought these devices from the same seller, have rated them as "OK", so I guess something must be wrong in my setup..

weetmuts, 2022-06-09:
You can try tuning to different frequencies. Also how often is the apator-08 supposed to transmit messages?

immek, 2022-06-09:
I don't know how often.. maybe someone that has one working could share this information?

About other frequencies.. I've been using Gqrd to find some frequencies with peaks around 868.9 and I spend a lot of time trying to scan all that where "suspicious", but with no luck. I think that I'm missing something... Any idea / combination of parameters I should try?

immek, 2022-06-09:
Found here https://github.com/weetmuts/wmbusmeters/issues/81 , in the logs, that it can report even every few minutes.

closed 2022-06-14 by immek