wmbusmeters/wmbusmeters #561: MQTT status publish every telegram received

quadriz, 2022-06-11:
Hi, my setup: Raspberry Pi Os on Rpi4, RTL2839 and Apator Ultrimis. 
Wmbusmeters doesn't publish to MQTT every telegram received, just skipping already handled telegrams like:

> (rtlwmbus) received full frame
> (wmbus) skipping already handled telegram leng=47.

What can I do to set up wmbusmeters like publishing to MQTT every telegram received even if it has the same totalizer value?

wmbusmeter.conf:

> loglevel=debug
> device=rtlwmbus:868.95M
> logtelegrams=true
> format=json
> meterfiles=/var/log/wmbusmeters/meter_readings
> meterfilesaction=overwrite
> meterfilesnaming=name
> meterfilestimestamp=day
> logfile=/var/log/wmbusmeters/wmbusmeters.log
> shell=HOME=/home/pi mosquitto_pub -h host -p 1883 -u ust -P passw -t wmbusmeters/$METER_ID -m ">
> ignoreduplicates=true

/etc/wmbusmeters.d/meter:

> name=woda
> type=ultrimis
> id=myidxxxxx
> key=00112233445566778899AABBCCDDEEFF

weetmuts, 2022-06-12:
From the README add: ignoreduplicates=false

weetmuts, 2022-06-12:
Odd meter that uses encryption but allows equal reading to generate the same encrypted telegram.
Bad crypto since it leaks information about there is water consumption or not.

The problem with rtl_wmbus is that it sends the same telegram twice if it succeeds in multiple decoding strategies.
You can fix this by selecting a single decoding strategy and stick to that one.

weetmuts, 2022-06-17:
So are you really missing telegrams? The text "(wmbus) skipping already handled telegram leng=47"  is a normal side effect of using rtl_wmbus which produces multiple copies of the telegrams.

weetmuts, 2022-07-24:
Ping?

BIBOLV, 2023-01-02:
Closing due to inactivity.

closed 2023-01-02 by BIBOLV