wmbusmeters/wmbusmeters #562: Add support for Quindis 5,5 Qwater

hemandk, 2022-06-14:
Would it be possible to add Suppoer for this Quindis 5,5 Qwater?

weetmuts, 2022-09-21:
Thanks fix pushed! Please test!

hemandk, 2022-09-23:
How do I pull the latest version from GitHub?

Not a Linux expert, so I'm kinda stuck at how to update the repo on my raspberry pi.

weetmuts, 2022-09-23:
git pull
make clean
make

hemandk, 2022-09-24:
So I got it updated, but I still get the same error:

Do I need to call a different driver then the lse_07_17 ?

2022-09-24_12:10:21] Started auto rtlwmbus[00000001] listening on c1
(wmbusmeters) waiting for telegrams
[2022-09-24_12:10:22] (meter) BryggersCold: meter detection could not find driver for id: 36724533 mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x35
[2022-09-24_12:10:22] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2022-09-24_12:10:22] (meter) to add support for this unknown mfct,media,version combination
(wmbusmeters) started meter 1 (BryggersCold 36724533 auto)
[2022-09-24_12:10:22] (meter) BryggersCold: meter detection did not match the selected driver unknown! correct driver is: unknown!
(meter) Not printing this warning again for id: 36724533 mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x35
[2022-09-24_12:10:22] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2022-09-24_12:10:22] (meter) to add support for this unknown mfct,media,version combination
[2022-09-24_12:10:23] (memory) rss 3907584 peak 3.72 MiB
[2022-09-24_12:10:44] (meter) BryggersHot: meter detection could not find driver for id: 36633633 mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x35
[2022-09-24_12:10:44] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2022-09-24_12:10:44] (meter) to add support for this unknown mfct,media,version combination
(wmbusmeters) started meter 2 (BryggersHot 36633633 auto)
[2022-09-24_12:10:44] (meter) BryggersHot: meter detection did not match the selected driver unknown! correct driver is: unknown!
(meter) Not printing this warning again for id: 36633633 mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x35
[2022-09-24_12:10:44] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2022-09-24_12:10:44] (meter) to add support for this unknown mfct,media,version combination
(wmbusmeters) started meter 3 (KitchenHot 36724180 auto)
[2022-09-24_12:13:31] (meter) KitchenCold: meter detection could not find driver for id: 36724629 mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x35
[2022-09-24_12:13:31] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2022-09-24_12:13:31] (meter) to add support for this unknown mfct,media,version combination
(wmbusmeters) started meter 4 (KitchenCold 36724629 auto)
[2022-09-24_12:13:31] (meter) KitchenCold: meter detection did not match the selected driver unknown! correct driver is: unknown!
(meter) Not printing this warning again for id: 36724629 mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x35
[2022-09-24_12:13:31] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2022-09-24_12:13:31] (meter) to add support for this unknown mfct,media,version combination
[2022-09-24_12:27:57] (meter) KitchenHot: meter detection did not match the selected driver unknown! correct driver is: unknown!
(meter) Not printing this warning again for id: 36724180 mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x35
[2022-09-24_12:27:57] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2022-09-24_12:27:57] (meter) to add support for this unknown mfct,media,version combination

andrasj, 2022-09-25:
HI, I'm  new to wmbusmeters, but I'm trying to read 2 Qundis QWater 5.5 devices. (similar issues with qheat device, but I'll respond in the other [issue](https://github.com/weetmuts/wmbusmeters/issues/546))
I get similiar messages like mentionned earlier in this thread when trying ot use the qwater or lse_07_17 driver
Here are 2 messages I captured from the devices:
> Received telegram from: 51142025
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Warm Water (30°C-90°C) meter (0x06)
                   ver: 0x18
                device: rtlwmbus[00000001]
                  rssi: 126 dBm
                driver: qwater
telegram=|_394493442520145118067A4A0000200C13180000004C1300000000426CFFFFCC081318000000C2086CDF2802BB560000326CFFFF046D3215D929|+446
Received telegram from: 12290776
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Warm Water (30°C-90°C) meter (0x06)
                   ver: 0x18
                device: rtlwmbus[00000001]
                  rssi: 142 dBm
                driver: qwater
telegram=|_394493447607291218067A140000200C13080000004C1308000000426CBF2CCC081308000000C2086CDF2802BB560000326CFFFF046D3015D929|+458

does this help for testing/validation?

i144, 2022-09-28:
Hi Fredrik! These are some telegrams received from CUL, with RSSI/LQI bytes appended to them. I hope they will be useful.

b49449344078514511806B034780DFF5F3500820C00000E0007C113FF7D25FF16060000FFFF00000000DF272102004F03000080008000800080008000800F0007063D0000000000000000002F046D060CC328AA4A88F8
b3944934407851451180633237A0D0000200C13160600004C1300000028C800426CFFFFCC081321020000C2086CDFADEF2702BB560000326CFFFF046D060CC3280BC48105

b49449344965716511807794D780DFF5F350082FD00007F0007C113FFC79BFF33080000FFFF00000000DF277600002457000080008000800080008000800080009B240007000000000000002F046D030CC328255288FB
b49449344965716511807794D780DFF5F350082FD00000E0007C113FFA69CFF40080000FFFF00000000DF27760000412E000080008000800080008000800080009B240007000000000000002F046D050CC328BFF285EA
b39449344965716511807FA5A7AFE0000200C13410800004C1300000028DC00426CFFFFCC081376000000C2086CDFBFAE2702BB560000326CFFFF046D050CC328469481F5

b39449344168026511906FF407A110000200C13530000004C130000006BCF00426CFFFFCC081353000000C2086CDF6D5A2802BB560000326CFFFF046D0506DC29B3E58CF8
b4944934487411051180651E7780DFF5F350082580000EF0007C113FFC07BFF19000000FFFF00000000DF28190000B6FC00008000800080008000000000010000FCB5000000b1B44AC08811222200D076E5B7A0100000004131202000004FD080114FF3B02002B2A83E4
b4944934448172351190707E7780DFF5F3500826F0000810007C113FF85CBFF34000000FFFF00000000DF28340000102B00008000800080008000800080008003DC8B0000000000000000002F046D0606DC2977BA91E1
b4944934487411051180651E7780DFF5F3500825900007E0007C113FF4DD7FF19000000FFFF00000000DF28190000B6FC00008000800080008000000000010000FCB50000000000000000002F046D0E06DC29935F9AE4
b49449344078514511806B034780DFF5F350082DF0000610107C113FF4350FF54970000FFFF00000000DF2849510017EF00008000800080008000800F00070000BCE900000000000000EC012F046D1106DC2948AE8AE2
b49449344299529511907F2F4780DFF5F350082DB00000E0007C113FF72E6FF55340000FFFF00000000DF286322002691000080008000800080008000800080009B2480008004000000DE002F046D1106DC297BCF91E3

closed 2023-04-05 by github-actions[bot]