wmbusmeters/wmbusmeters #563: Build: missing libtinfo  -  undefined reference to symbol 'curs_set'

hvegh, 2022-06-15:
On some distros the ncurses library is split into: **libncurses** and **libtinfo** 

If you have this, you can fix temporary in the Makefile by adding **-ltinfo** in the linker stage:

Change Makefile:
```
...
$(BUILD)/wmbusmeters-admin: ...
           $(CXX) -o $(BUILD)/wmbusmeters-admin    .... -lncurses  ...
...

```
to:
```
...
$(BUILD)/wmbusmeters-admin: ...
           $(CXX) -o $(BUILD)/wmbusmeters-admin    .... -lncurses -ltinfo ...
...

```

weetmuts, 2022-09-23:
@jsuchal @hvegh I have temporarily reverted the addition of ltinfo.
How should we solve this? The configured could check if ltinfo is available.

hvegh, 2022-09-28:
The internet is full of issues regarding libncurses and libtinfo:
[https://dropbear.xyz/2011/09/30/ncurses-library-split/](https://dropbear.xyz/2011/09/30/ncurses-library-split/) is an explanation.

Not sure how to solve,  if libtinfo is present might be good indication it is split...

hvegh, 2022-10-19:
Gentoo builds libtinfo as a seperate library by default:

```
Calculating dependencies... done!
[ebuild   R    ] sys-libs/ncurses-6.3_p20220423:0/6::gentoo  USE="cxx gpm (split-usr) (tinfo) -ada -debug -doc -minimal -profile (-stack-realign) -static-libs -test -trace -verify-sig" 0 KiB

```
For Gentoo builds one should instruct to build ncurses with  tinfo. 
This needs the -tinfo flag to in the build system...

...
tried this but it does not seem to work... anyway the issue is resolved for me.

weetmuts, 2023-01-21:
This is on my TODO.

closed 2023-05-05 by github-actions[bot]