wmbusmeters/wmbusmeters #566: Stopped detecting amb8465 in 1.8.0

lazyoldbear, 2022-06-25:
In 1.6.0, upon start it said:
`Started config amb8465[00000000] on /dev/ttyUSB-Metis-2 listening on c1,t1`

On 1.8.0 it says:
`Expected amb8465 on /dev/ttyUSB-Metis-2 but found unknown instead, ignoring it!`

Config:
`device=/dev/ttyUSB-Metis-2:amb8465:c1,t1`

Tried all sorts of auto, does not see anything. 

Installed from this chart: https://github.com/k8s-at-home/charts/tree/master/charts/stable/wmbusmeters (version 1.2.0)

weetmuts, 2022-06-26:
Yes, indeed. Its the amb8465 again. It seems like I have always tested the auto detection when I plug in the amb8465
while wmbusmeters is running. This means that the detector code works on a freshly reset dongle. However
it seems like the amb8465 can get into a mode where the detector code fails thourougly.

If you start the daemon, then plug in the amb8465 does that work?

weetmuts, 2022-06-26:
Can you enable loglevel=debug and post the log here when you restart the docker?

lazyoldbear, 2022-06-26:
I did not try exactly plugging after starting, but I did try to plug it out and back in. My obvious suspicion would be around dashes in the device name, and also in ID specifying feature: perhaps it does not want to work without an explicit ID any longer?..

Anyway, I don't have "docker", but I'll try to bake a debug log now.

lazyoldbear, 2022-06-26:
```

(config) "# device=auto:c1" ""
(config) "device" "/dev/ttyUSB-Metis-2:amb8465:c1,t1"
(config) "# device=/dev/ttyUSB-Metis-2" ""
(config) "listento" "c1"
(config) "logtelegrams" "false"
(config) "format" "json"
(config) "# meterfiles=/wmbusmeters_data/logs/meter_readings" ""
(config) "# meterfilesaction=overwrite" ""
(config) "# meterfilesaction=overwrite" ""
(config) "# meterfilesnaming=name" ""
(config) "# meterfilestimestamp=day" ""
(config) "# logfile=/wmbusmeters_data/logs/wmbusmeters.log" ""
(config) "# shell=/usr/bin/mosquitto_pub -h mosquitto.elveli-mqtt -i "$MQTT_CLIENT_ID" -u "$MQTT_USER" -P "$MQTT_PASSWORD" -q 1 -t wmbusmeters/$METER_ID -m "$METER_JSON"" ""
(config) "# alarmshell=/usr/bin/mosquitto_pub -h mosquitto.elveli-mqtt -i "$MQTT_CLIENT_ID" -u "$MQTT_USER" -P "$MQTT_PASSWORD" -q 1 -t wmbusmeters_alarm -m "$ALARM_TYPE $ALARM_MESSAGE"" ""
(config) "shell" "sh /wmbusmeters_custom_bin/message.sh $METER_ID "$METER_JSON""
(config) "alarmshell" "sh /wmbusmeters_custom_bin/message_alarm.sh "$ALARM_TYPE" "$ALARM_MESSAGE""
(wmbusmeters) version: 1.8.0
(config) using device: /dev/ttyUSB-Metis-2:amb8465:c1,t1 
(config) number of meters: 1
(lookup) with file/hex "/dev/ttyUSB-Metis-2" amb8465
(serial) check if /dev/ttyUSB-Metis-2 can be accessed
(serial) tty /dev/ttyUSB-Metis-2 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB-Metis-2 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(195,95,184,98,51,157,205,2) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB-Metis-2 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(195,95,184,98,16) modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB-Metis-2 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyUSB-Metis-2) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465) reading response... 1
(serial) received binary "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF00030000FFFFFFFFFFFFFF"
(amb8465) not enough data for decoding ConfigAMB8465!
(amb8465) reading response... 2
(serial) received binary "0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF69"
(amb8465) proprely decoded ConfigAMB8465 response. Content: id=00000000 media=00 version=00 c_field=44 auto_rssi=00
(amb8465) found response at offset 0
(serialtty) closed /dev/ttyUSB-Metis-2 (detect amb8465)
(serial) EVENT thread interrupted
(amb8465) detect id=00000000 media=00 version=00 c_field=44 auto_rssi=00
(amb8465) are you there? yes 00000000
(main) opening /dev/ttyUSB-Metis-2:amb8465:c1,t1
Started config amb8465[00000000] on /dev/ttyUSB-Metis-2 listening on c1,t1
(amb8465) on /dev/ttyUSB-Metis-2
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB-Metis-2 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(195,95,184,98,51,157,205,2) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB-Metis-2 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(195,95,184,98,16) modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB-Metis-2 fd 3 (amb8465)
(amb8465) get config
(serial /dev/ttyUSB-Metis-2) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary "FF8A7A00780080710200000000FF"
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FF"
(amb8465) not enough bytes yet, partial command response 14 126.
(serial) received binary "FFFA00FFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) not enough bytes yet, partial command response 30 126.
(serial) received binary "FFFFFF0C3200026400FFFFFFFFFF01"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF01"
(amb8465) not enough bytes yet, partial command response 45 126.
(serial) received binary "0004000000FFFFFF0144000000000000"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF0144000000000000"
(amb8465) not enough bytes yet, partial command response 61 126.
(serial) received binary "0000FFFF0B060100FFFFFFFFFF0003"
[... 9 lines cut ...]
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF00030000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) not enough bytes yet, partial command response 107 126.
(serial) received binary "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF00030000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) not enough bytes yet, partial command response 122 126.
(serial) received binary "FFFFFF69"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200026400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF00030000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF [... cut ...]
(main) regular reset of amb8465  on /dev/ttyUSB-Metis-2 will happen every 82800 seconds
(amb8465) set link mode 09
(serial /dev/ttyUSB-Metis-2) sent "FF040109F3770000"
(serial) received binary "FF8401007A"
(amb8465) checkAMB8465Frame "FF8401007A"
(amb8465) received full command frame
(amb8465) set link mode completed
(amb8465) set link mode response "0100"
(wmbus) set timeout amb8465 to "3600" with expected activity "mon-sun(00-23)"
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
...
```

lazyoldbear, 2022-06-26:
But now it works. Both with debug and without. 
I can't reproduce the failure any longer.

However, these reports about sufficient data and whatnot. Could it be something with bitrate that just makes the issue floating?

lazyoldbear, 2022-06-26:
Just keeps working now. So far, bitrate/timing is my biggest suspect.

weetmuts, 2022-07-01:
I have been testing with my amb8465 down for some time and it seems to work.
Closing this for now. I will keep the other amb8465 issues open. Thanks for the report!

closed 2022-07-01 by weetmuts

lazyoldbear, 2022-07-29:
I hope it is acceptable to reopen it, for I see it again. Out of the blue (it worked, no one touched anything, and then again).
`Expected amb8465 on /dev/ttyUSB-Metis-2 but found unknown instead, ignoring it!
(verbose) ignoring device c1,t1
(lookup) with file/hex "/dev/ttyUSB-Metis-2" amb8465
(serial) check if /dev/ttyUSB-Metis-2 can be accessed
(serial) tty /dev/ttyUSB-Metis-2 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB-Metis-2 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(177,100,228,98,127,222,54,36) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB-Metis-2 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(177,100,228,98,16) modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB-Metis-2 fd 5 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyUSB-Metis-2) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465) reading response... 1
(serial) received binary ""
(amb8465) reading response... 2
(serial) received binary ""
(amb8465) reading response... 3
(serial) received binary ""
(amb8465) are you there? no.
(serialtty) closed /dev/ttyUSB-Metis-2 (detect amb8465)
(serial) EVENT thread interrupted
Expected amb8465 on /dev/ttyUSB-Metis-2 but found unknown instead, ignoring it!
(verbose) ignoring device c1,t1
(lookup) with file/hex "/dev/ttyUSB-Metis-2" amb8465
(serial) check if /dev/ttyUSB-Metis-2 can be accessed
(serial) tty /dev/ttyUSB-Metis-2 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB-Metis-2 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(177,100,228,98,127,222,54,36) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB-Metis-2 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(177,100,228,98,16) modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB-Metis-2 fd 5 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyUSB-Metis-2) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted`

lazyoldbear, 2022-07-29:
@weetmuts Please advise: should this be reopened, or new input merged elsewhere?

lazyoldbear, 2022-07-29:
Interestingly, reverted to 1.6.0, does not work still. 
It may not be 1.8.0 after all.

lazyoldbear, 2022-07-29:
Machine reboot helped, but that is so discouraging.