wmbusmeters/wmbusmeters #568: RTL-SDR v3 usb  "No wmbus device detected" on NUC with HA OS

Grisen80, 2022-07-02:
I have installed wmbusmeters as an add-on on my 5y+ old NUC running Home Assistant OS,  but  HA wont  recognize [my RTL-SDR v3 USB dongle](https://www.amazon.co.uk/s?i=merchant-items&me=AGX9N6DGNRN2Q&rh=p_4%3ARTL-SDR+Blog&dc&marketplaceID=A1F83G8C2ARO7P&qid=1608657156&ref=sr_nr_p_4_5). "No wmbus device detected, waiting for a device to be plugged in."

I have tried all usb ports on the NUC, unplugged  my zigbee and zwave dongles, multiple restarts but no luck. Nothing shows up under tty in the HA "Hardware" section.

I have verified that the dongle works by running wmbusmeters as a daemon on my Linux box and it picks up my watermeter signals ok

Is this a HA thing? Is the problem that HA does not support this particular dongle? Any help much appreciated. Thanks

tfitts, 2022-08-10:
I'm getting the same thing.  It looks like this might solve the issue.  I wonder if weetmuts can compile that in the ha addon?

https://github.com/weetmuts/wmbusmeters/issues/372

Grisen80, 2022-08-10:
Yes, that would be great. 

@weetmuts, would adding the rtl-sdr to the Home Assistant add-on as mentioned in [372](https://github.com/weetmuts/wmbusmeters/issues/372) be possible?

BIBOLV, 2022-08-12:
Hi,
rtl-wmbus and rtl_433 is already added to HA docker image, so this more looks like particular dongle and HA compatibility. I would suggest to check all possible logs when attaching dongle, maybe there are some clues, like issues with kernel modules.

tfitts, 2022-08-18:
I'm using the Nooelec NESDR Smart v4 Bundle - Premium RTL-SDR w/Aluminum Enclosure, 0.5PPM TCXO, SMA Input & 3 Antennas. RTL2832U & R820T2-Based Software Defined Radio.  The specs are below.  From what I've read this seems to be a fairly popular RTL-SDR device.  Do you know how I would verify that this isn't supported?

BUSNUM: '001'
DEVNAME: /dev/bus/usb/001/003
DEVNUM: '003'
DEVPATH: /devices/platform/soc/ffe09000.usb/ff500000.usb/xhci-hcd.1.auto/usb1/1-1/1-1.2
DEVTYPE: usb_device
DRIVER: usb
ID_BUS: usb
ID_FOR_SEAT: usb-platform-xhci-hcd_1_auto-usb-0_1_2
ID_MODEL: RTL2838UHIDIR
ID_MODEL_ENC: RTL2838UHIDIR
ID_MODEL_ID: '2838'
ID_PATH: platform-xhci-hcd.1.auto-usb-0:1.2
ID_PATH_TAG: platform-xhci-hcd_1_auto-usb-0_1_2
ID_REVISION: '0100'
ID_SERIAL: Realtek_RTL2838UHIDIR_00000001
ID_SERIAL_SHORT: '00000001'
ID_USB_INTERFACES: ':ffffff:'
ID_VENDOR: Realtek
ID_VENDOR_ENC: Realtek
ID_VENDOR_ID: 0bda
MAJOR: '189'
MINOR: '2'
PRODUCT: bda/2838/100
SUBSYSTEM: usb
TAGS: ':seat:'
TYPE: 0/0/0
USEC_INITIALIZED: '3771300'

Grisen80, 2022-08-18:
Thanks for you input @BIBOLV 

I don't understand what happened but I when I plugged in the dongle today to look for info in the logs (no restart or anything) wmbus recognizes the stick and started printing out received telegrams. 

Very strange

comperta, 2022-08-26:
Hello @Grisen80 ,
I have the similar issue.
Did you change anything in your conf?

thecem, 2022-08-27:
https://github.com/weetmuts/wmbusmeters/issues/588#issuecomment-1222063890

try this as YAML conf.

BIBOLV, 2023-01-02:
Closing as solution was provided.

closed 2023-01-02 by BIBOLV