wmbusmeters/wmbusmeters #569: No meters configured even though I have created multiple meter files in /etc/wmbusmeters.d

koldbrandt, 2022-07-05:
I am trying to parse some messages and forward them through mqtt. 
I do however continue to get "No meters configured. Printing id:s of all telegrams heard!" even though I have multiple meters in /etc/wmbusmeters.d. 
I have tried both with the wildcard '*' in id, after the 2 first numbers, which are the same for all my devices. I have also tried running the shell script mentioned in this issue https://github.com/weetmuts/wmbusmeters/issues/300 and ofcourse edited the the type to driver and put in the correct driver. 

A simplified config file:
`
loglevel=normal  
device=amb8465:t1,c1  
logtelegrams=true  
format=json  
meterfiles=/wmbusmeters_data/logs/meter_readings  
meterfilesaction=overwrite  
logfile=/var/log/wmbusmeters/wmbusmeters.log  
`

Please let me know if you need any more information.

koldbrandt, 2022-07-05:
I got it to work. 
It seems like it can't read the meterfiles if your config file is located in '/home/me/.config/wmbusmeters'. 
If I just edit the one located at '/etc/wmbusmeters.config', then it works without any problem. 

Should I close this issue or do you want to look into why it doesn't work when the config file is located in '/home/me/.config/wmbusmeters' ?

weetmuts, 2022-07-05:
The daemon looks for /etc/wmbusmeters.conf and /etc/wmbusmeters.d by default
if you want to change the location you have to change the wmbusmeters.service script to specify --useconfig=/home/me/.config/wmbusmeters when it starts wmbusmetersd.

If you start wmbusmeters (not the daemon) you can also specify --useconfig=/home/me/.config/wmbusmeters to load /home/me/.config/wmbusmeters/wmbusmeters.conf and meters from
/home/me/.config/wmbusmeters/wmbusmeters.d

koldbrandt, 2022-07-05:
Ah okay, I didn't get the last part that the meters had to be put into /home/me/.config/wmbusmeters/wmbusmeters.d - I am not sure how clear this is in the documentation. Thank you for the fast reply!

weetmuts, 2022-07-05:
Great! I improved the readme to explain it better. Good luck!

Jordan87, 2022-11-08:
Hi
What i'm doing wrong that my configuration for meter is missing after addon restart and i cant configure my meter?
From log I see:

> No meters configured. Printing id:s of all telegrams heard!
> Received telegram from: 2119e124
> manufacturer: (SAP) Sappel (0x4c30)
> type: Water meter (0x07)
> ver: 0x00
> device: cul
> rssi: -35 dBm
> driver: izar

I tried to add:
> meters:
> - |-
> name=MainWater
> driver=izar
> id=2119e124
> key=00000000000000000000000000000000

to files /config/wmbusmeters/etc/wmbusmeters.conf or /config/wmbusmeters/etc/wmbusmeters.d/ but after restart those changes are missing and same info in log is visible.

Thanks