wmbusmeters/wmbusmeters #570: Multical302 total energy consumption is always 0

erinaceinae, 2022-07-18:
Hi, 

I'm trying to use wmbusmeters to read my Multical 302 heat meter. I got an AES key from heat distributor, telegrams are decoded but total energy consumption is 0 when my meter shows a value ~59 GJ on a display. Current power consumption works fine I can see it changes when I use heat, also total volume grows. Is it possible to solve this problem?

`telegram=|25442D2C9635966730048D200F82465521DCB879E26D143A|4E1700141700A05D01DE26000000|+27
(multical302) 00: 25 length (37 bytes)
(multical302) 01: 44 dll-c (from meter SND_NR)
(multical302) 02: 2d2c dll-mfct (KAM)
(multical302) 04: 96359667 dll-id (67963596)
(multical302) 08: 30 dll-version
(multical302) 09: 04 dll-type (Heat meter)
(multical302) 0a: 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(multical302) 0b: 20 ell-cc (slow_resp sync)
(multical302) 0c: 0f ell-acc
(multical302) 0d: 82465521 sn (AES_CTR)
(multical302) 11: dcb8 payload crc (calculated dcb8 OK)
(multical302) 13: 79 tpl-ci-field (EN 13757-3 Application Layer with Compact frame (no tplh))
(multical302) 14: e26d format signature
(multical302) 16: 143a data crc
(multical302) 18: 4E1700
(multical302) 1b: 141700
(multical302) 1e: * A05D01 total volume (895.040000 m3)
(multical302) 21: * DE26 target date (2022-06-30 00:00)
(multical302) 23: * 0000 current power consumption (0.000000 kW)
(multical302) 25: * 00 info codes ()

(shell) env "METER_NAME=Heat"
(shell) env "METER_MEDIA=heat"
(shell) env "METER_TYPE=multical302"
(shell) env "METER_TIMESTAMP=2022-07-18T16:55:59Z"
(shell) env "METER_TIMESTAMP_UTC=2022-07-18T16:55:59Z"
(shell) env "METER_TIMESTAMP_UT=1658163359"
(shell) env "METER_TIMESTAMP_LT=2022-07-18 18:55.59"
(shell) env "METER_DEVICE=rtl433[77771111153705700]"
(shell) env "METER_RSSI_DBM=999"
(shell) env "METER_TOTAL_ENERGY_CONSUMPTION_KWH=0"
(shell) env "METER_CURRENT_POWER_CONSUMPTION_KW=0"
(shell) env "METER_TOTAL_VOLUME_M3=895.04"
(shell) env "METER_AT_DATE=2022-06-30 00:00"
(shell) env "METER_TOTAL_ENERGY_CONSUMPTION_AT_DATE_KWH=0"
(shell) env "METER_CURRENT_STATUS="

`

weetmuts, 2022-07-19:
Actually, it does not look like the telegram contains any total energy....

Can you run with --logtelegrams for a while to see if you can catch both the long version of the telegram
and the compact version? The one above is a compact version. There might also be yet another version
of the telegram with the total energy.

erinaceinae, 2022-07-19:
Hi 

I cached few long frames. I can see a total heat (in GJ) I consumed on webpage of the supplier (update once a month). So it's possible it might be another version of telegram with total energy.

`telegram=|2E442D2C9635966730048D200EB1795521651478|030F4E1700430F1417000314A55D01426CDE26022D000001FF2100|+43558
telegram=|25442D2C9635966730048D200FC0795521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+43574
telegram=|25442D2C9635966730048D2010C0795521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+43591
telegram=|25442D2C9635966730048D2012C1795521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+43623
telegram=|25442D2C9635966730048D2013D0795521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+43640
telegram=|25442D2C9635966730048D2014D0795521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+43656
telegram=|25442D2C9635966730048D2015D1795521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+43672
telegram=|2E442D2C9635966730048D2016D2795521651478|030F4E1700430F1417000314A55D01426CDE26022D000001FF2100|+43689
telegram=|25442D2C9635966730048D2017E0795521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+43705
telegram=|25442D2C9635966730048D2019E1795521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+43738
telegram=|25442D2C9635966730048D201AE1795521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+43754
telegram=|2E442D2C9635966730048D2026217A5521651478|030F4E1700430F1417000314A55D01426CDE26022D000001FF2100|+131
telegram=|25442D2C9635966730048D2027227A5521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+147
telegram=|25442D2C9635966730048D2028227A5521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+164
telegram=|25442D2C9635966730048D2029307A5521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+180
telegram=|25442D2C9635966730048D202A307A5521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+196
telegram=|25442D2C9635966730048D202B317A5521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+212
telegram=|25442D2C9635966730048D202C317A5521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+228
telegram=|25442D2C9635966730048D202D407A5521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+244
telegram=|2E442D2C9635966730048D202E417A5521651478|030F4E1700430F1417000314A55D01426CDE26022D000001FF2100|+261
telegram=|25442D2C9635966730048D2030427A5521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+293
telegram=|25442D2C9635966730048D2031507A5521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+309
telegram=|25442D2C9635966730048D2032507A5521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+325
telegram=|25442D2C9635966730048D2033517A5521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+341
telegram=|25442D2C9635966730048D2035607A5521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+373
telegram=|2E442D2C9635966730048D2036617A5521651478|030F4E1700430F1417000314A55D01426CDE26022D000001FF2100|+389
telegram=|25442D2C9635966730048D2037627A5521D47A79E26D0C5F|4E1700141700A55D01DE26000000|+406`

erinaceinae, 2022-07-19:
> Thanks, you are running an old version of wmbusmeters. In december 2021 I
> added support for handling meters that can either send KWh or MJ and
> multical 302 is one of those. With version 1.8.0 you get the total energy
> from the first telegram in your list. It automatically normalizes the
> energy to kwh, but you can get mj by adding --addconversions=MJ if you
> really want it.
> wmbusmeters --format=json --addconversions=MJ 2E442D2C9635966730048D200EB1795521651478030F4E1700430F1417000314A55D01426CDE26022D000001FF2100 MyHeatMeter auto ANYID NOKEY
> {"media":"heat","meter":"multical302","name":"MyHeatMeter","id":"67963596","total_energy_consumption_kwh":16572.222222,"total_energy_consumption_mj":59660,"current_power_consumption_kw":0,"total_volume_m3":895.09,"at_date":"2022-06-30 00:00","total_energy_consumption_at_date_kwh":16411.111111,"total_energy_consumption_at_date_mj":59080,"current_status":"","timestamp":"2022-07-19T07:40:07Z"}

Ok! Thank you for your help :)

weetmuts, 2022-07-24:
Can we close this now?

erinaceinae, 2022-07-24:
yes, I updated to newest version and it works, thank you :)

weetmuts, 2022-07-24:
Great!

closed 2022-07-24 by weetmuts