wmbusmeters/wmbusmeters #572: Need List of all founded Meters

joergbeicht, 2022-07-25:
My question:
Is it possible to get a list of all meter ids received without having to create a file in etc/wmbusmeters.d?
My subtenant also has a type=multitical21.
I thought 2. would give me ALL the counters.
But that doesn't happen. In the log file I find the counter of the subtenant.

Here are my previous files in etc/wmbusmeters.d

1.
name=basement-joerg
id=25575311
key=A1F854D742A371D18CD37875C4962064
driver=car
type=multiple21

2.
name=allothermometers
id=*
key=NOKEY
driver=car

to 1.
Sends me all meter data flawlessly
to 2.
Sends all meters that do not require an AES key.
Including many heating cost meters

This is my wmbus,eters.conf:
loglevel=debug
device=auto
donotprobe=/dev/ttyAMA0
listento=c1,t1,s1
logtelegrams=true
format=json
meterfiles=/wmbusmeters_data/logs/meter_readings
meterfilesaction=append
logfile=/wmbusmeters_data/logs/wmbusmeters.log
shell=/usr/bin/mosquitto_pub -h 192.168.178.22 -p 1883 -u pi -P 123456 -t wmbusmeters/$METER_ID -m "$METER_JSON"

Thanks for any info,
Georg

BIBOLV, 2022-08-12:
Hi,
Could you please explain a bit more your issue? I`m bit confused as you are stating that you want to get all available meters in your area and in begging of post you are saying that it does not work, but at the end that it does. Or it does not work as you would like? Or when you are running config 2, you do not see your meter that works in config 1, but others do?
Either way I would suggest to switch on debug and run for longer periods.

BIBOLV, 2023-01-02:
Closing due to inactivity.

closed 2023-01-02 by BIBOLV