wmbusmeters/wmbusmeters #573: /etc/systemd/system/multi-user.target.wants/wmbusmeters.service  is missing a line

Grothesk242, 2022-07-25:
/etc/systemd/system/multi-user.target.wants/wmbusmeters.service misses 
this line: 

ExecStartPre=-/bin/mkdir -p /var/log/wmbusmeters

When adding the line (or the missing directory manually) the service starts, otherwise it fails with this error: 

````
systemctl status wmbusmeters.service
● wmbusmeters.service - "wmbusmeters service"
     Loaded: loaded (/lib/systemd/system/wmbusmeters.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Mon 2022-07-25 23:17:35 CEST; 2min 36s ago
       Docs: https://github.com/weetmuts/wmbusmeters
             man:wmbusmeters(1)
        CPU: 61ms

Jul 25 23:17:35 mysterium systemd[1]: wmbusmeters.service: Scheduled restart job, restart counter is at 3.
Jul 25 23:17:35 mysterium systemd[1]: Stopped "wmbusmeters service".
Jul 25 23:17:35 mysterium systemd[1]: wmbusmeters.service: Start request repeated too quickly.
Jul 25 23:17:35 mysterium systemd[1]: wmbusmeters.service: Failed with result 'exit-code'.
Jul 25 23:17:35 mysterium systemd[1]: Failed to start "wmbusmeters service".
```

weetmuts, 2022-08-30:
You are absolutely right! Thanks! It seems like previously the /var/log/wmbusmeters directory was
installed automatically, but when the meter readings were moved from log to lib, the mkdir got lost.

I think I have fixed it now! Please test if you have the time.
https://github.com/weetmuts/wmbusmeters/commit/165389910dc05850f7584865f5ef02723b2d1176

BIBOLV, 2023-01-02:
Closing as issue was fixed.

closed 2023-01-02 by BIBOLV