wmbusmeters/wmbusmeters #576: More info code values available on Multical 603

jnxxx, 2022-07-31:
I noticed [here](https://koce1-kamstrup.ocecdn.oraclecloud.com/content/published/api/v1.1/assets/CONTCA9C63276A19491FAE5F1A7E254CD06B/native/55122029_L1_EN.pdf?channelToken=ed241bbb18f444908a8fc9ed97ca5d5b) in section 7.7.2 that Multical 603 has more info codes than the driver currently exposes.
In particular I find leak and burst information interesting.

So, I wanted to extend the driver with these and started out with the following:
```
#define INFO_CODE_IN_A1_LEAKAGE_IN_THE_SYSTEM 4096
#define INFO_CODE_IN_B1_LEAKAGE_IN_THE_SYSTEM 8192
#define INFO_CODE_IN_A2_LEAKAGE_IN_THE_SYSTEM 262144
#define INFO_CODE_IN_B2_LEAKAGE_IN_THE_SYSTEM 524288
#define INFO_CODE_V1_V2_BURST_WATER_LOSS 268435456
#define INFO_CODE_V1_V2_BURST_WATER_PENETRATION 536870912
#define INFO_CODE_V1_V2_LEAKAGE_WATER_LOSS 1073741824
#define INFO_CODE_V1_V2_LEAKAGE_WATER_PENETRATION 2147483648
```

And then planned to update MeterMultical603::status() accordingly.
That would be quite simple.

However, I notice the variable `info_codes_` is defined as uchar (8 bit) and read like this:
`  extractDVuint8(&t->dv_entries, "04FF22", &offset, &info_codes_);`

Doesn't "04" mean 32 bit?
If so, wouldn't it be correct to just change it as follows:
`  extractDVuint32(&t->dv_entries, "04FF22", &offset, &info_codes_);`
(and of course adjust the variable definition to uint32)
?

And second, how to test it?

thecem, 2022-08-19:
Pls send me your changes, so I would try to test. I need additional a actual telegram.

weetmuts, 2022-09-03:
Thanks! I took a look at 7.7.2 and added all the codes. Fix pushed.

I also refactored the multical603 driver to the new driver format. Thanks @thecem for the
same work on multical303. I took yours and modified the field names to get backwards compatibility
for the multical603 driver.

I have a vaque feeling that all the heat meters could be somehow handled by a single driver, now
that PrintProperty::OPTIONAL works. But such a merge will have to wait until all drivers have been 
rewritten to the new format and there is way to deprecate fields, since we need to consolidate the field naming.

Please test the new 603!

https://github.com/weetmuts/wmbusmeters/blob/master/src/driver_multical603.cc

thecem, 2022-09-03:
What should we use for field naming?
Own naming or vendor specific?

closed 2023-04-05 by github-actions[bot]