wmbusmeters/wmbusmeters #577: sendbasic device data with mosquttio without aes-key

joergbeicht, 2022-08-02:
First of all:
All devices found, e.g. my water meter, are transferred to the
wmbusmeters.log written e.g. like this:
Received telegram from: 25575311
           manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                   type: Cold water meter (0x16) encrypted
                    ver: 0x1b
                 device: im871a[00103086]
                   RSSI: -117dBm
                 driver: multical21
Question:
Can I transfer this data immediately via mosquitto_pub?
So far mosquitto_pub only sends when I create a file with key=my-aes-key under /etc/wmbusmeter.d

weetmuts, 2022-08-04:
Hmm, do you mean that you want the summary sent using the shell command?

BIBOLV, 2023-01-02:
Closing due to inactivity. Please reopen issue if you want to continue to work on topic.

closed 2023-01-02 by BIBOLV