wmbusmeters/wmbusmeters #579: Add support for GWF EnerCal F2 and similar devices

awor, 2022-08-05:
I am trying to connect a GWF EnerCal F2 to your solution. This heat meter is popular in the German speaking part of Europe and is sold under numerous brands :
https://www.google.com/search?q=enercal+F2
Example: https://mgs-owl.de/mediafiles/Sonstiges/Deltamess/Montageanleitung/Bedien_10254_10255_10256_10257_10259_10260_10261_10262_10263_10264_10985.pdf

Could you please add support for this? Here is the debug output:

root@wmbus:/home/libmbus/binwmbusmeters --format=json --logtelegrams --debug --pollinterval=10 MAIN=/dev/ttyUSB0:mbus Tempo piigth:MAIN:2400 00309924 NOKEY
(wmbusmeters) version: 1.8.0-2-gd2918c5
(config) using device: MAIN=/dev/ttyUSB0:mbus
(config) number of meters: 1
(meter) created Tempo piigth encrypted
(lookup) with file/hex "/dev/ttyUSB0" mbus
(lookup) driver: mbus
(main) opening MAIN=/dev/ttyUSB0:mbus
Started config MAIN=mbus on /dev/ttyUSB0
(mbus) on /dev/ttyUSB0
(serial) before config: /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars() modem(DTR|RTS)
(serial) after config: /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (mbus)
Sending NKE to mbus MAIN
(serial /dev/ttyUSB0) sent "1040004016"
(serial) EVENT thread interrupted
(main) regular reset of mbus on /dev/ttyUSB0 will happen every 82800 seconds
(wmbus) no alarm (expected activity) for mbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(meter) secondary addressing bus MAIN to address 00309924
(serial /dev/ttyUSB0) sent "680B0B6873FD5224993000FFFFFFFFAB16"
(serial) EVENT thread interrupted
(serial) received binary "E5"
(mbus) checkMBUSFrame
"E5"
(mbus) received E5 single byte frame.
(mbus) checkMBUSFrame
""
(mbus) less than 6 bytes, partial frame
(meter) polling Tempo 00309924 (secondary) with req ud2 bus MAIN
(serial /dev/ttyUSB0) sent "105BFD5816"
(serial) EVENT thread interrupted
(serial) received binary "68"
(mbus) checkMBUSFrame
"68"
(mbus) less than 6 bytes, partial frame
(serial) received binary "8E8E68"
(mbus) checkMBUSFrame
"688E8E68"
(mbus) less than 6 bytes, partial frame
(serial) received binary "08017224"
(mbus) checkMBUSFrame
"688E8E6808017224"
(mbus) not enough bytes, partial frame 8 148
(serial) received binary "993000"
(mbus) checkMBUSFrame
"688E8E6808017224993000"
(mbus) not enough bytes, partial frame 11 148
(serial) received binary "E61E08"
(mbus) checkMBUSFrame
"688E8E6808017224993000E61E08"
(mbus) not enough bytes, partial frame 14 148
(serial) received binary "040900"
(mbus) checkMBUSFrame
"688E8E6808017224993000E61E08040900"
(mbus) not enough bytes, partial frame 17 148
(serial) received binary "000004"
(mbus) checkMBUSFrame
"688E8E6808017224993000E61E08040900000004"
(mbus) not enough bytes, partial frame 20 148
(serial) received binary "068146"
(mbus) checkMBUSFrame
"688E8E6808017224993000E61E08040900000004068146"
[... 266 lines cut ...]
(piigth) 135 : C2 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
(piigth) 136 : 80 dife (subunit=0 tariff=0 storagenr=1)
(piigth) 137 : 00 dife (subunit=0 tariff=0 storagenr=1)
(piigth) 138 : 6C vif (Date type G)
(piigth) 139 C?: C127
(meters) render field operating_time(Time)[2] with dventry @58 key 0426 data 49880200
(meters) "operating_time_h":46.100278
(meters) render field on_time(Time)[3] with dventry @52 key 0422 data 49880200
(meters) "on_time_h":46.100278
(meters) render field meter_datetime(Text)[4] with dventry @76 key 046D data 2510DE27
(meters) "meter_datetime":"2022-07-30 16:37"
{"media":"heat","meter":"piigth","name":"Tempo","id":"00309924","operating_time_h":46.100278,"on_time_h":46.100278,"meter_datetime":"2022-07-30 16:37","temperature_c":null,"average_temperature_1h_c":null,"average_temperature_24h_c":null,"relative_humidity_rh":null,"relative_humidity_1h_rh":null,"relative_humidity_24h_rh":null,"timestamp":"2022-07-30T11:02:02Z","device":"MAIN","rssi_dbm":0}
(mbus) checkMBUSFrame
""
(mbus) less than 6 bytes, partial frame
^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(wmbus) disconnected /dev/ttyUSB0 mbus
(serialtty) closed /dev/ttyUSB0 (mbus)
(serial) TIMER thread interrupted
(serial) event loop stopped!
(wmbus) deleted mbus
(serial) closing 1 devices
root@wmbus:/home/libmbus/bin#

BIBOLV, 2022-08-05:
Sensostar driver can decode some values, could you verify that those are correct?
```
{
    "media":"heat",
    "meter":"sensostar",
    "name":"",
    "id":"00309924",
    "meter_timestamp":"2022-07-30 16:37",
    "total_kwh":411265,
    "power_kw":0,
    "flow_water_m3h":0,
    "forward_c":31,
    "return_c":29,
    "difference_c":2.2,
    "total_water_m3":35354.96,
    "current_status":null,
    "timestamp":"2022-08-05T20:54:34Z"
}
```

awor, 2022-08-06:
The json data you posted looks good, many thanks for your input @BIBOLV !

I tried this on my side, but it isn't returning any json in the console: How must I proceed to test on my end?

root@wmbus:~# wmbusmeters --format=json --logtelegrams --debug  --pollinterval=10 MAIN=/dev/ttyUSB0:mbus Tempo sensostar:MAIN:2400 00309924 NOKEY
(wmbusmeters) version: 1.8.0-2-gd2918c5
(config) using device: MAIN=/dev/ttyUSB0:mbus
(config) number of meters: 1
(lookup) with file/hex "/dev/ttyUSB0" mbus
(lookup) driver: mbus
(main) opening MAIN=/dev/ttyUSB0:mbus
Started config MAIN=mbus on /dev/ttyUSB0
(mbus) on /dev/ttyUSB0
(serial) before config: /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (mbus)
Sending NKE to mbus MAIN
(serial /dev/ttyUSB0) sent "1040004016"
(serial) EVENT thread interrupted
(main) regular reset of mbus  on /dev/ttyUSB0 will happen every 82800 seconds
(wmbus) no alarm (expected activity) for mbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop

awor, 2022-08-07:
Ah, this could be the problem: `(serial) internal error after select! errno=Illegal seek`
Hi @weetmuts, would it be possible for you to look into this error message? Maybe this is why polling is not working when using config files (command line works, see above):

root@wmbus:~# wmbusmeters --useconfig=/etc --debug
(config) "device" "MAIN=/dev/ttyUSB0:mbus:2400"
(config) "logtelegrams" "false"
(config) "format" "json"
(config) "meterfiles" "/var/lib/wmbusmeters/meter_readings"
(config) "meterfilesaction" "overwrite"
(config) "#logfile=/var/log/wmbusmeters/wmbusmeters.log" ""
(config) "logfile" "/dev/stdout"
(config) "shell" "/usr/bin/mosquitto_pub -h 172.22.1.151 -t wmbusmeters/"$METER_ID" -m "$METER_JSON""
(config) "" ""
(config) loading meter file /etc/wmbusmeters.d/EnerCal
(config) name=EnerCal
(config) driver=sensostar:mbus
(config) pollinterval=10s
(config) id=00309924
(config) overriding loglevel with debug
(wmbusmeters) using log file /dev/stdout
(wmbusmeters) version: 1.8.0-2-gd2918c5
(config) store meter files in: "/var/lib/wmbusmeters/meter_readings"
(config) using device: MAIN=/dev/ttyUSB0:mbus:2400
(config) number of meters: 1
(lookup) with file/hex "/dev/ttyUSB0" mbus
(lookup) driver: mbus
(main) opening MAIN=/dev/ttyUSB0:mbus:2400
Started config MAIN=mbus on /dev/ttyUSB0
(mbus) on /dev/ttyUSB0
(serial) before config: /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (mbus)
Sending NKE to mbus MAIN
(serial /dev/ttyUSB0) sent "1040004016"
(serial) EVENT thread interrupted
**`(serial) internal error after select! errno=Illegal seek`**
(main) regular reset of mbus  on /dev/ttyUSB0 will happen every 82800 seconds
(wmbus) no alarm (expected activity) for mbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop 

Many thanks in advance!

weetmuts, 2022-08-09:
Hmm, wmbusmeters never seek in the serial device. So is this a problem with the serial driver/device?

Which is the mbus device that you have connected to your /dev/ttyUSB0 ?

Can you also post the dmesg output when you plug-in the device?

weetmuts, 2022-08-09:
I have pushed a new driver enercal. You can check the decode here:
https://wmbusmeters.org/analyze/688e8e6808017224993000e61e080406000000040681460600041488f2350084401426e02600025B2600025f220002622a00042247880200042647880200043B00000000042c00000000046d2d0ede2784406e000000008480406e00000000c4800006B1450600c48000143Be43500c4c000142fd22600c4c0006e00000000c480406e00000000c280006cc1271f00000000f416

awor, 2022-08-09:
Many thanks for looking into this! At first glance, your new enercal driver is looking very good :) It's now returning the values/json where it previously failed. I still have some mqtt connection issue I need to investigate tomorrow. I will get back to you shortly. Until then, tack så mycket!

weetmuts, 2022-08-11:
Great! What is the name of the mbus master device that you use? I.e. the box that translates the serial data to mbus line levels?

Do you know why the  enercal has HCA values? And what are the subunits in the device used for?

awor, 2022-08-11:
I am using the same device that [atze09](https://github.com/atze09) is using in https://github.com/weetmuts/wmbusmeters/issues/435 which is this one:
https://www.aliexpress.com/item/4000787535232.html

Sorry, I am not familiar with HCA values and haven't got a clue what the subunits are used for. You can see what is available on the display of the unit by looking at pages 7-8 in the document listed here https://github.com/weetmuts/wmbusmeters/issues/579#issue-1330233871. Maybe it's one of the impulse registers? Once I connect wmbusmeters with my Home Assistant and get more familiar with the data, maybe I can give a better answer.

awor, 2022-08-12:
Okay, there seems to be a problem with the json payload, it contains a CR LF right after the meter name:
[image]
this is not valid and breaks the payload, hence it cannot be properly processed. I don't see anything I can do to cause or influence this. Here is my meter config in an editor:
[image]
when you have a moment, could you please look into this?
Generally, the parsing of the meter config seems to be a little flaky: Putting the parameters in a different order doesn't appear to work, also adding a CR LF at the end of the last line will break the config.

awor, 2022-08-27:
I had created my meter config file in WinSCP and Notepad++. I finally realized that Unix EOL is not CRLF but just LF. This caused all the havoc I described above. In NP++ I now set Unix (LF) in Edit > EOL Conversion, saved the file, and now all works as expected. It took me some time to figure this out. @weetmuts you might want to consider adding this warning in the documentation, i.e. that the config files will break if Windows style CRLF is used instead of Unix style LF line endings. Again, many thanks for your great work. A number of people are probably using your solution to optimize the energy used in their homes (like me). In other words, your great project is a contribution to saving the planet :-)

closed 2022-08-27 by awor

weetmuts, 2022-08-27:
Many thanks! Great that you found the problem! I think I will  fix the code so that it accepts CRLF as well. I just never expected anyone to edit the config files on windows! :-)