wmbusmeters/wmbusmeters #580: Receving wMBus meters telegrams via wMBus to LoRaWan gw

raszuk, 2022-08-09:
Hi, 

I am designing residential reading of wMBus water and heat meters. The issue is that the block of flats has 60 meters of concrete walls and max radius of best wMBus receiver I tested so far is 20 meters. 

So my next test is to use few of those extenders: https://wireless-solutions.de/products/wireless-m-bus-range-extender-with-lorawan/

Well they transmit via LoRaWan which on Raspberry Pi I can capture with SX1262 868MHz LoRa HAT. 

The question here is wmbusmeters capable of parsing such telegrams ? If so how to setup wmbusmeters.conf ? 

Many thx,
R.

BIBOLV, 2022-08-12:
Hi,
"First lora dongle is partially implemented - iu880b. This is still work in progress though." [Link](https://github.com/weetmuts/wmbusmeters/discussions/554#discussioncomment-3217381).
So you can get iu880b and help to get LoRa implemented in wmbusmeters or process LoRa msg`s in ChirpStack or other LoRa server and then forward received wmbus msg`s to wmbusmeters for processing.

BIBOLV, 2023-01-02:
Closing due to inactivity. Please reopen issue if you want to continue to work on topic.

closed 2023-01-02 by BIBOLV