wmbusmeters/wmbusmeters #581: Unable to run Wmbusmeters addon in Home Assistant - s6-overlay-suexec: fatal: can only run as pid 1

ma73u53, 2022-08-11:
Hi,
When I try to run Wmbusmeters (W-MBus to MQTT) addon in Home Assistant, I get the following error in log:
s6-overlay-suexec: fatal: can only run as pid 1

Tried to uninstall, reinstall the addon, but not helped. Addon is not able to run.
Also tried default yaml configuration.

HAOS:
Home Assistant 2022.8.3
Supervisor 2022.08.3
Operating System 8.4

Wmbusmeters (W-MBus to MQTT) version: 0.2.19
FC0012 tuner (Realtek RTL2838UHIDIR)

YAML configuration:
data_path: /config/wmbusmeters
conf: |-
  loglevel=normal
  device=auto:t1
  logtelegrams=false
  format=json
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME $METER_JSON
meters:
    name=ZimnaWoda
    driver=apator162
    id=01315701
    key=00000000000000000000000000000000
mqtt:
  host: 192.168.x.x
  port: 1883
  user: homeassistant
  password: xxx

Did you face the similar problem?
Thank you in advance.

BIBOLV, 2022-08-12:
Hi,
You are using old version - please update to 0.2.22, this issue should be fixed there.

emilleszczak2, 2022-11-06:
I have the same problem. I've just installed 1.9.0 and it's not starting on HA 2022.11.1. There is now a newer version available to install.

Any idea?

LukaszP2, 2022-11-17:
I have problem too...  
Wmbusmeters: 0.2.19
HA: 2022.11.3
Error: s6-overlay-suexec: fatal: can only run as pid 1

@BIBOLV  How to get  0.2.22?

LukaszP2, 2022-11-17:
add this as repsitory: https://github.com/weetmuts/wmbusmeter

emilleszczak2, 2022-11-17:
@LukaszP2 I have added the same repository and there is no 0.2.22

weetmuts, 2022-11-17:
@LukaszP2 you forgot the s at the end of the url. It is:
 https://github.com/weetmuts/wmbusmeters

emilleszczak2, 2022-11-17:
My problem is solved now. I deleted repository from HA and added it again.

BIBOLV, 2023-01-02:
Closing - issue was fixed.

closed 2023-01-02 by BIBOLV