wmbusmeters/wmbusmeters #584: Add support for Kamstrup Multical 303, how to?

thecem, 2022-08-16:
Log:

```
registering meters ...
Adding meter-0001 ...
Adding meter-0002 ...
Generating MQTT configuration ... 
Broker 192.168.1.7 will be used.
Running wmbusmeters ...
(config) "device" "rtlwmbus"
(wmbusmeters) version: 1.8.0-12-g9c4b9df
(config) using device: rtlwmbus 
(config) number of meters: 2
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers
Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-d device_index (default: 0)]
	[-g gain (default: 0 for auto)]
	[-p ppm_error (default: 0)]
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	filename (a '-' dumps samples to stdout)
<<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 139 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 140 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(main) found specified device (rtlwmbus) that matches detected device (:rtlwmbus[00000001]:0/0)
(main) opening rtlwmbus
Started config rtlwmbus[00000001] listening on none
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 00000001
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(serial) EVENT thread interrupted
(bgshell) arg "/usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serialcmd) opened /bin/sh pid 145 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) received ascii "T1;1;1;2022-08-16 11:49:50.000;135;137;82788281;0x5e442d2c8182788240047a180050258cbf442c002dd266452fcacee039a510ef69d0e72469a7f887f80ae919257d2d8e7f63de8d164d6506ac2a8def5cff71556305bea9c7ef45a015fe9ea49ffec85bd1cd6d6b6b0b1e355ac7370d43f48d<0A>T1;1;1;2022-08-16 11:49:50.000;126;134;82788281;0x5e442d2c8182788240047a180050258cbf442c002dd266452fcacee039a510ef69d0e72469a7f887f80ae919257d2d8e7f63de8d164d6506ac2a8def5cff71556305bea9c7ef45a015fe9ea49ffec85bd1cd6d6b6b0b1e355ac7370d43f48d<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2022-08-16 11:49:50.000;135;137;82788281;0x5e442d2c8182788240047a180050258cbf442c002dd266452fcacee039a510ef69d0e72469a7f887f80ae919257d2d8e7f63de8d164d6506ac2a8def5cff71556305bea9c7ef45a015fe9ea49ffec85bd1cd6d6b6b0b1e355ac7370d43f48d<0A>T1;1;1;2022-08-16 11:49:50.000;126;134;82788281;0x5e442d2c8182788240047a180050258cbf442c002dd266452fcacee039a510ef69d0e72469a7f887f80ae919257d2d8e7f63de8d164d6506ac2a8def5cff71556305bea9c7ef45a015fe9ea49ffec85bd1cd6d6b6b0b1e355ac7370d43f48d<0A>"
(rtlwmbus) received full frame
(meter) no meter handled  checking 2 templates.
(wmbus) parseDLL @0 95
(wmbus) parseELL @10 85
(wmbus) parseNWL @10 85
(wmbus) parseAFL @10 85
(wmbus) parseTPL @10 85
(meter) MainWater: for me? 82788281 in 34657054
(meter) MainWater: not for me: not my id
(meter) ODGMulti303: for me? 82788281 in 82788281
(meter) ODGMulti303: meter detection did not match the selected driver multical403! correct driver is: unknown!
(meter) Not printing this warning again for id: 82788281 mfct: (KAM) Kamstrup Energi (0x2c2d) type: Heat meter (0x04) ver: 0x40
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
[... 102 lines cut ...]

Config:
```
data_path: /config/wmbusmeters
conf: |-
  loglevel=debug
  device=rtlwmbus
  logtelegrams=true
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=ODGMulti303
    driver=multical403
    id=82788281
    key=NOKEY -> "actually not known, requested"
mqtt:
  host: sec
  port: 1883
  user: sec
  password: sec
```
The driver multical403 dosn't work but should: `ODGMulti303: meter detection did not match the selected driver multical403! correct driver is: unknown!`
What are the next steps to add the Kamstrup Multical 303

thecem, 2022-08-17:
Additional Information. (Total Energy is realy zero!)

It seems the driver multical603 is most fitting!

https://wmbusmeters.org/analyze/5e442d2c8182788240047af3005025a90c7eB357dd477fdf96495358deB74d67c02ddad4c97B94f2535eea82de2f8317f5efe7e2df9a0d58Bc630d817cdfdB83e4e32060720637cc8577ecd83BB53eBB13f77821d13fdf8165a1fB51681f46:multical603:75EDE0CBBB6E126764898645AA366568

```
Auto driver  : not found!
Best driver  : enercal 24/36
Using driver : multical603(driver should be upgraded) 00/00
000   : 5e length (94 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 2d2c dll-mfct (KAM)
004   : 81827882 dll-id (82788281)
008   : 40 dll-version
009   : 04 dll-type (Heat meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : f3 tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 5025 tpl-cfg 2550 (synchronous AES_CBC_IV nb=5 cntn=0 ra=0 hc=0 )
015   : 2f2f decrypt check bytes (OK)
017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
018   : 06 vif (Energy kWh)
019 C!: 00000000 total energy consumption (0.000000 kWh)
023   : 04 dif (32 Bit Integer/Binary Instantaneous value)
024   : 14 vif (Volume 10⁻² m³)
025 C!: EE000000 total volume (2.380000 m3)
029   : 04 dif (32 Bit Integer/Binary Instantaneous value)
030   : FF vif (Manufacturer specific)
031   : 07 combinable vif (Reserved)
032 C!: 3D000000 energy forward kwh (61)
036   : 04 dif (32 Bit Integer/Binary Instantaneous value)
037   : FF vif (Manufacturer specific)
038   : 08 combinable vif (Reserved)
039 C!: 3D000000 energy returned kwh (61)
043   : 02 dif (16 Bit Integer/Binary Instantaneous value)
044   : 59 vif (Flow temperature 10⁻² °C)
045 C!: 370A T1 flow temperature (26.150000 °C)
047   : 02 dif (16 Bit Integer/Binary Instantaneous value)
048   : 5D vif (Return temperature 10⁻² °C)
049 C!: 490A T2 flow temperature (26.330000 °C)
051   : 02 dif (16 Bit Integer/Binary Instantaneous value)
052   : 3B vif (Volume flow l/h)
053 C!: 0000 volume flow (0.000000 m3/h)
055   : 02 dif (16 Bit Integer/Binary Instantaneous value)
056   : FF vif (Manufacturer specific)
057   : 22 combinable vif (PerHour)
058 C?: 0000
060   : 02 dif (16 Bit Integer/Binary Instantaneous value)
061   : 6C vif (Date type G)
062 C!: D128 target date (2022-08-17 00:00)
064   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
065   : 06 vif (Energy kWh)
066 C?: 00000000
070   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
071   : 14 vif (Volume 10⁻² m³)
072 C?: 00000000
076   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
077   : 6C vif (Date type G)
078 C?: C128
080   : 2F skip
081   : 2F skip
082   : 2F skip
083   : 2F skip
084   : 2F skip
085   : 2F skip
086   : 2F skip
087   : 2F skip
088   : 2F skip
089   : 2F skip
090   : 2F skip
091   : 2F skip
092   : 2F skip
093   : 2F skip
094   : 2F skip

{
    "media":"heat",
    "meter":"multical603",
    "name":"",
    "id":"82788281",
    "total_energy_consumption_kwh":0,
    "total_volume_m3":2.38,
    "volume_flow_m3h":0,
    "t1_temperature_c":26.15,
    "t2_temperature_c":26.33,
    "at_date":"2022-08-17 00:00",
    "current_status":"",
    "energy_forward_kwh":61,
    "energy_returned_kwh":61,
    "timestamp":"2022-08-17T14:39:32Z"
}

Using: wmbusmeters: 1.8.0-10-g09fa9ae
09fa9ae7d4c73450e398ea05c6d373c3459f8f3b
```
And a lot more telegramms:

telegram=|_5E442D2C8182788240047A99005025BB0C09A5EFBCB0A1EB7A3EC5550DBAFEC436A6C3B3A44DCFCE0526555F20270C9228DF707EC3901A8EE066DB78BF0A12D23A95219BEA7B62B8A967621AF804E6492CB6E439D33AEFA8BF38B32232C676|+3288

telegram=|_5E442D2C8182788240047A9A005025AD2283C5F4F3F1D0373D62609BDF6E56930896FAB5D4DB5AFBE33FB9D32B4215BEA351CF9031B7DB9D9695F9C3F6AC5CF77B6DF03E560D1FD421327051CD064B0DFEB477257E21B705B7C10D30361D0A|+3304

telegram=|_5E442D2C8182788240047A9B005025B3B2DA1872E68EE2C7D08015422C7ABA8E47295863FA85017FF8F173FEFED78436489B2CA28F8D5739C38C9CFE3F95E5E07687338DFB4E00970D199A643F6500D3652496AA2E7F251CB3F88346BB6275|+3320

telegram=|_5E442D2C8182788240047A9C0050252764E3B7B74E432A084DA47A9265AF8723A4829EA28CB873FAEC6C206054366C5714E0907C27F37A27DE347E73D3DAD5FA24FB297D175CA639F86F3B8E5EBD8B66DEA0C958AFFA00794AE13F5AD5FBFE|+3335

telegram=|_5E442D2C8182788240047A9D005025A85EC1D893B2F2C3A21EDA8AA4638111C7EA84421B7BA11B4C4EC7F0C21F453F154A8AA93A06DEC02CBF70EB197364A9C23072CDE1ED8399940E5DFB298A8A6BC1340E77EC24045530B8CABF817BEE4A|+3351

telegram=|_5E442D2C8182788240047A9E0050259A8648D9E2A450ED1628F6B598ED90A7CACD0C77839E1F70BED685F988260319A47B6FB6A7602A36856D073AC4B7E3EA468A51A2584D29DA5E7A691E03C2DAD3CABE08719810B180DB0BBEE761A991E5|+3367

telegram=|_5E442D2C8182788240047AA00050255BA7CF5BF5678CA5A22E55D8E846D2414C581D58FF8605A7181ED7AB0C848A4AC24A61A72CECC399ABDF4E06A314E6D4CDA58A50B37D863CCC6027B798141F6432C680B36FEAC9D2415DC47B68CCA382|+3398

telegram=|_5E442D2C8182788240047AA10050255A9DB2C8E409242D2DC8433DE4EAE86510686E7096761E869476C59C013106EC9342A3A6C8A10B73FBB9A37CCA5D2FC0E2B0C5E34EEEA0A2A1D4052E87AE5CD298C5FCBBFDBF3A3B57A0D2205C463A9D|+3414

telegram=|_5E442D2C8182788240047AA3005025A0F360C51437C44BA58E3E475EB6DF9A109C41EA0206051E80738AB1A9EDB3EA6059AD2E9C9EA65905ABCD01C0C9A6BA87B51D73C889645719CE60F1DC818E5E550D066BF6AD7A9DDB77A52B17A46B77|+3445

telegram=|_5E442D2C8182788240047AA4005025776C2003CC0FA7A86EF4B1364E8875D41853FF62E78DDDC7E0488E62CDB50500B9791B1BB05FE46419A80E02E2E34AC708196D91BD7297D76B7BF45F1BEC355D5F26C357263AF37CF7704207D0463084|+3461

telegram=|_5E442D2C8182788240047AA5005025901BD72D3E8EBB45AA55B15221E637A00FB390307A3E160DCC0D27BE58DA1EB030032FABED66BF432E8AEAE98315719735142603F323769D0B6FB71988F3AE01587AD1A01C1141CA37C180CF4D550F2A|+3477
telegram=|_5E442D2C8182788240047AA70050251F96685210800C95C76872F5A9D77ACF5D36C3D9AE2E9BFEB91CEE989ED89D4B8C6B5BD1B29C09305BC37642313657CDC0EB1A158AC35AA7EB66143F777533E29CA1339E26962D2A14927CA752354119|+3509

thecem, 2022-08-19:
Ok, hard work and I need help please!

I created additional a new driver (driver_multical303.cc) additional to the meter_multical303.cc (and the adaptions in kem-import.py) the driver is partly working. i'm not able to get the values for 04FF07 & 04FF08 & 04FF22 and both dates as values 026C & 426C. In old meter format it is working with multical603 and meter_multical303.cc (PR).

I don't know how to get the date_time:
```
addStringFieldWithExtractor(
            "target_date_time",
            "The most recent billing period date and time (14/348/Date and Time logged).",
            PrintProperty::JSON | PrintProperty::FIELD,
            FieldMatcher::build()
            .set(DifVifKey("426C")) // How to Handle this ??? What is the order in FieldMatcher ??
            .set(MeasurementType::Instantaneous)
            .set(VIFRange::DateTime)
            );
```
This is the complete driver:

```
/*
 Copyright (C) 2020-2022 thecem (gpl-3.0-or-later)

 This program is free software: you can redistribute it and/or modify
 it under the terms of the GNU General Public License as published by
 the Free Software Foundation, either version 3 of the License, or
 (at your option) any later version.

 This program is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 GNU General Public License for more details.

 You should have received a copy of the GNU General Public License
 along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#include"meters_common_implementation.h"

namespace
{
    struct Multical303 : public virtual MeterCommonImplementation {
        Multical303(MeterInfo &mi, DriverInfo &di);
    };

    static bool ok = registerDriver([](DriverInfo&di)
    {
        di.setName("multical303");
        di.setMeterType(MeterType::HeatMeter);
        di.addLinkMode(LinkMode::C1);
        di.addLinkMode(LinkMode::T1);
        di.addDetection(MANUFACTURER_KAM, 0x04,  0x40);
        di.setConstructor([](MeterInfo& mi, DriverInfo& di){ return shared_ptr<Meter>(new Multical303(mi, di)); });
    });

    Multical303::Multical303(MeterInfo &mi, DriverInfo &di) : MeterCommonImplementation(mi, di)
    {
        addNumericFieldWithExtractor(
            "total_energy",
            "The total energy consumption recorded by this meter.",
            PrintProperty::JSON | PrintProperty::FIELD | PrintProperty::IMPORTANT,
            Quantity::Energy,
            VifScaling::Auto,
            FieldMatcher::build()
            .set(MeasurementType::Instantaneous)
            .set(VIFRange::AnyEnergyVIF)
            );

        addNumericFieldWithExtractor(
            "total_volume",
            "The volume of water (3/68/Volume V1).",
[... 140 lines cut ...]
            .set(DifVifKey("426C")) // How to Handle this ??? What is the order in FieldMatcher ??
            .set(MeasurementType::Instantaneous)
            .set(VIFRange::DateTime)
            );
    }
}
// Test: Heat multical303 82788281 75EDE0CBBB6E126764898645AA366568
// Comment:
// Not able to get values for 04FF07 & 04FF08 & 04FF22 and both dates as values 026C & 426C in old meter format it is working multical603 and multical303.
// telegram=|_5E442D2C8182788240047A83005025186E9C6D9815EBFC04CBE8E4B8C8A6B9949C9DAA629CD96D920F321CFBEE7AE104DD8532C5C0EE79B4CFACCFA75D3A5EB6D4493DFAFE91B15C3A3DCFCE899138B8EA02CDB609D31CF019F9E4FD04559E|
// {"media":"heat","meter":"multical303","name":"","id":"82788281","total_energy_kwh":0,"total_volume_m3":2.38,"forward_c":26.12,"return_c":26.24,"actual_flow_m3h":0,"target_energy_kwh":0,"target_volume_m3":0,"current_status":null,"timestamp":"2022-08-19T06:09:27Z"}
// | Not yet

```

Telegramms to Test:
`telegram=|_5E442D2C8182788240047ACE0050250F5E95006915A213610F6BD25EB0DB526153DF53ADB35C501C4546A6D2B917ABE426D2354F291AF2FD73661B44DC43E284719A06F5B0A8586D6A1AC7B6B43A776252DCCEE2F6F5A0DFC68CD10644C26F|
`
`telegram=|_5E442D2C8182788240047ACD005025A0A7D27CDF8F0ACCD1E55B33F4870F820D0C83DD6993713C1C885A252456060AA41479DA431707B653AB1C7BCE15F1C9C17D7E47FB8D08D7AE155786313B0956A23F22A1E1303CCF38432CF6DD626107|
`
`telegram=|_5E442D2C8182788240047ACB0050254E643F14671B0DF3DFDA47B2B065FC1B81ED045697715143528C9B91591192F88FEACEF27078915B0804E40DD592C7D0558710B6DD22805BFBF44021A10469E8C5BDAE8777E897BF4B04C8D6D2CD0E79|`

It would very nice if someone could help me out, to the catch even the last 5 fields...

Thank you!

thecem, 2022-08-28:
Hi,

I am right the „driver“ file will be the new format? Is there a possibility to get help on how to extract the values E8 (ID 97) and E9 (ID 110) in right way? 

Thanks a lot?

weetmuts, 2022-09-01:
Great work! Yes, the driver format is the new format!

weetmuts, 2022-09-01:
I took your driver here and did some fixes. It now decode everything except 04FF07 and 04FF08.

I need to know which unit they use? And if you have an example value for the telegrams, then 
we can verify the future code.... :-)

The only missing things where `.set(StorageNr(1))` for the target values.
And using ::Date instead of ::DateTime for the date values.

The status field is now found using the `set(DifVifKey("02FF22"))` and the lookup is there.

weetmuts, 2022-09-01:
Pushed your driver to github. Please test!

weetmuts, 2022-09-01:
I looked at your first driver and there it looks like the forward/return energy is simply kwh. 
So I added a forced lookup for these from their difvif keys to the Unit::KWH. 
Seems to work. Please confirm that it is indeed the correct values. :-)

thecem, 2022-09-02:
Great! Many thanks for piking this up and helping out!

**E8 // E9 -> 04FF07 and 04FF08**:  -> Unit of this kind of Energy is: °Cm³ [m³ x °C] (E8 = m 3 x t1 & E9 = m 3 x t2).

> 
> 7.1.2 Energy registers E8 and E9
> E8 and E9 are used as a basis for calculating the volume-based average temperatures of inlet and outlet, respectively. With every volume count (every 0,01 m 3 for q p 1,5 m 3/h with CCC=419), the registers are accumulated by the product of m 3 x °C, which makes E8 and E9 a suitable basis for calculating the volume-based average temperatures. E8 and E9 can be used for average calculation during any period of time as long as the volume register is read at the same time as E8 and E9.

Reference in [Tec Manual](https://koce1-kamstrup.ocecdn.oraclecloud.com/content/published/api/v1.1/assets/CONT37BE438014F34F26BC3AF0C9CC1D0712/native/55122701_E1_GB.pdf?channelToken=ed241bbb18f444908a8fc9ed97ca5d5b): Site 62/63.

**Date:**

The Manual describes that both date values are Date & Time.

Test ongoing.

thecem, 2022-09-02:
```
{
    "media": "heat",
    "meter": "multical303",
    "name": "ODGMulti303_TEST",
    "id": "82788281",
    "total_energy_kwh": 0,
    "total_volume_m3": 2.87,
    "forward_energy_kwh": 73,
    "return_energy_kwh": 73,
    "forward_c": 25.41,
    "return_c": 25.54,
    "actual_flow_m3h": 0,
    "status": "OK",
    "date_time": "2022-09-02",
    "target_energy_kwh": 0,
    "target_volume_m3": 2.87,
    "target_date_time": "2022-09-01",
    "timestamp": "2022-09-02T07:13:02Z",
    "device": "rtlwmbus[cmd_0]",
    "rssi_dbm": 136
}

```
I propose:
    "forward_energy_kwh" and  "return_energy_kwh" may change to "..._energy_m3c"

Something for /src/units.h:

```
#define LIST_OF_UNITS \
    X(M3C,m3c,"m3C",SupplyReturnEnergy,"cubic meter celsius")  \
```

What is with the time of "date_time" and "target_date_time"?

Something else what I should test analyse: 

"The only missing things where .set(StorageNr(1)) for the target values."

NICE! Rest seems to be working!

weetmuts, 2022-09-02:
Nice! I added m3c as a new unit, but I did not create a new quantity, so it belongs to Energy.

However there are no automatic conversions between this energy and kwh, since obviously we do not have all info about the heating medium, eg water.

So, the telegram only contains bits for dates.....if they say date-time in the manual then that is wrong.

I changed the json keys from "date_time" to "meter_date" and "target_datetime" to "target_date"
this is consistent how the fields are named in other meters.

thecem, 2022-09-02:
Date: if no time, then it is consistent to rename! thx.

here a actual mqtt output::

```
{
    "media": "heat",
    "meter": "multical303",
    "name": "ODGMulti303",
    "id": "82788281",
    "total_energy_kwh": 0,
    "total_volume_m3": 2.87,
    "forward_energy_m3c": 73,
    "return_energy_m3c": 73,
    "forward_c": 25.36,
    "return_c": 25.49,
    "actual_flow_m3h": 0,
    "status": "OK",
    "meter_date": "2022-09-02",
    "target_energy_kwh": 0,
    "target_volume_m3": 2.87,
    "target_date": "2022-09-01",
    "timestamp": "2022-09-02T10:10:35Z",
    "device": "rtlwmbus[cmd_0]",
    "rssi_dbm": 133
}
```

[image]

BIBOLV, 2023-01-02:
Closing issue as driver were implemented.

closed 2023-01-02 by BIBOLV