wmbusmeters/wmbusmeters #585: Home assistant addon automatically stops and watchdog does not restart

atze09, 2022-08-16:
Hello,
Since a few days the integration stops every day und does not automatically restart.

This is the log i get, i will increase the loglevel:
`telegram=|6871716808027255870032496A8804FB000000_0C78558700320406F001000082046CFFFFC284006CFFFF84040600000080C484000600000080820A6CC128840A06F00100000413EF3C040002597D0B025D140B02616900042D00000000043B00000000046D0109D0280426CF0B000002FD1700001FC216|+5
telegram=|68B3B36808007257004820C51400042F000000_047839803801040690050000041372F60700042B00000000142BE5A40000043BA3060000143B11070000025B2400025F24000261FDFF0223CA00046D0329D028440600000000441300000000426C000001FD170003FD0C05000084200600000000C420060000000084300600000000C430060000000084401300000000C44013000000008480401300000000C48040130000000084C0401300000000C4C04013000000000E16|+6
Assertion failed: v.size() == 12 (src/dvparser.cc: extractDouble: 775)
/run.sh: line 56:   142 Aborted                 (core dumped) /wmbusmeters/wmbusmeters --useconfig=$CONFIG_DATA_PATH
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped`

weetmuts, 2022-08-16:
Thanks! For mbus telegrams, the final 2 bytes, the crc and the 0x16 end marker were
not properly removed before the telegram was put into the div vif decoder.
And for some telegrams, the crc + 0x16 became a valid dv entry for a value that could 
be looked up by the driver, alas there were no actual data after the 0x16 so the assert
terminated wmbusmeters.

I have pushed a fix that properly removed the crc+0x16 before parsing.
I have also pushed a fix the makes wmbusmeters more user friendly when it gets bad telegrams.
It will print warnings instead of terminating on an assert.

Please test!

atze09, 2022-08-18:
Works again, thank you very much!

closed 2022-08-18 by atze09