wmbusmeters/wmbusmeters #588: Problem with USB-Dongle by HA on Proxmox

MrLukson, 2022-08-22:
Hello,

I add USB with RTL-SDR dongle to VM with HASS and I can see it in HASS, but the add-on doesn't't see the adapter and in log I have "No wmbus device detected, waiting for a device to be plugged in."
[image]

Could someone help me why I have trouble to run WMBUS as add-on?
Perhaps someone can share me config file.

Thanks in advance

thecem, 2022-08-22:
could you post your addon config as YAML?

MrLukson, 2022-08-22:
Config:
`data_path: /config/wmbusmeters
conf: >-
  loglevel=normal device=auto:t1 logtelegrams=false format=json
  logfile=/dev/stdout shell=/wmbusmeters/mosquitto_pub.sh
  wmbusmeters/$METER_NAME "$METER_JSON"
meters: []
mqtt: {}
`
LOG:
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Syncing wmbusmeters configuration ...
Registering meters ...
Generating MQTT configuration ... 
Broker core-mosquitto will be used.
Running wmbusmeters ...
No such log level: "normal device=auto:t1 logtelegrams=false format=json logfile=/dev/stdout shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON""
No wmbus device detected, waiting for a device to be plugged in.
No meters configured. Printing id:s of all telegrams heard!`

thecem, 2022-08-22:
try first to see you receive telegrams:
```
data_path: /config/wmbusmeters
conf: |-
  loglevel=debug
  device=rtlwmbus
  logtelegrams=true
meters: []
mqtt: {}
```

MrLukson, 2022-08-22:
@thecem is working. Should add logs and mqtt and meters?

thecem, 2022-08-22:
to save some CPU usage and produce less heat on the rtl-sdr:

```
data_path: /config/wmbusmeters
conf: |-
  loglevel=normal
  device=rtlwmbus:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a)
  logtelegrams=true
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=yourmetername
    driver=yourmeterdriverisworking
    id=meteridshowninlog
    key=nokey or hexstring
mqtt: {}
```

MrLukson, 2022-08-22:
with multiple meters should I add under first meter second?

thecem, 2022-08-22:
get the first working, if then add multi meters via:

data_path: /config/wmbusmeters
conf: |-
  loglevel=normal
  device=rtlwmbus:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a)
  logtelegrams=true
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=yourmetername
    driver=yourmeterdriverisworking
    id=meteridshowninlog
    key=NOKEY or hexstring
 - |-
    name=yourmetername2
    driver=yourmeterdriverisworking
    id=meteridshowninlog
    key=NOKEY or hexstring
mqtt: {}

MrLukson, 2022-08-22:
great it's working. Thank you very much. That was simple to configure wmbusmeters on docker but in VM was a little bit tricky.

thecem, 2022-08-22:
yes, I had same problems! Happy to help and save time!
What kind of meter do you have and which driver do you use?

MrLukson, 2022-08-22:
I have two meters: Izar and Amiplus

MrLukson, 2022-08-22:
I moved from docker to Proxmox VM

BIBOLV, 2023-01-02:
Closing as solution was provided.

closed 2023-01-02 by BIBOLV