wmbusmeters/wmbusmeters #589: need help in tuning to exact frequency of hydrus meter..

vanjapogacic, 2022-08-23:
Hi everyone,
First of all to the developer(s) -  thanks for the great piece of software.

I am running it on a small x86 linux machine in a docker. dongle is rtl, RTL2838
the meter is Diehl Hydrus 171a. Meter distance is 20 meters from my apartment. there is 5 meters in total on this floor - they are all next to each other in some small valve room (for all apartments on the floor).
The issue I have is I am getting those CRC errors (broken incomplete telegrams)
https://github.com/weetmuts/wmbusmeters/issues/378 (same issue).
I tried with original antenna of the dongle (small 7-8cm), i tried with homemade 35cm coat hanger wire (tried sticking it directly to the dongle to avoid cable loss doubts) - i am getting same results. I have ordered another antenna, but not sure antenna is the only issue..
I have tried 433.470, 433.570mhz (as I saw it mentioned by other people here for this meter). Also now i started trying 433.9, 434.1 etc...
the issue I have is that these meters seem to broadcast every 5 hours or something. what is your experience? what do you expect from 5 meters, how many messages per day should I see? At this point I dont care which one is my meter, as long as I start receiving correct messages from any of them, only then I can go to the next step...
so it will take me weeks to discover the correct setting for it. Each change on antenna/frequency/settings requires almost 1 day waiting...
I went to the meter with rf analyzer on android (and dongle plugged in otg usb) - but i dont think it was broadcasting...i couldnt figure out any signal at that time.
Do you have any advice how to speed up the process of tuning into correct settings?
Just to mention, the dongle setup is working fine, if I stop this docker and start sdr docker, it starts picking up neighbour weather stations immediately..
since I am using docker i dont know how to run wmbusmeters in command line directly, so I am just changing .config file and restarting docker every time.
here is my config:
loglevel=debug
logtimestamps=important
#device=rtl433:434.570M
#device=rtlwmbus:434.470M
device=rtlwmbus(ppm=17):433.900M
#donotprobe=
#driver=auto
listento=t1
logtelegrams=true
format=json
meterfiles=/wmbusmeters_data/logs/meter_readings
meterfilesaction=overwrite
logfile=/wmbusmeters_data/logs/wmbusmeters.log

any suggestion will be appreciated :)
thanks!

thecem, 2022-08-23:
try 

`device=rtlwmbus`

and

`device=rtl433`

megacar666, 2022-08-23:
> try
> 
> `device=rtlwmbus`
> 
> and
> 
> `device=rtl433`

Hi, 
I tried that first, rtlwmbus gives me nothing (868mhz). it seems these are configured here at 433 something, as I managed to get 10-15 broken telegrams on that freq.
with rtl433 i have some strange problem that is immediately receives one ascii and then gets stuck in checking frame (it just stays like that forever, nothing happens after that. it all happens in 1 second after start, and then no change.

here is what happens with rtl433:
`(serial) EVENT thread interrupted
(shell) return code 1
(shell) 11 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 12 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(main) found specified device (rtl433) that matches detected device (:rtl433[00000001]:0/0)
(main) opening rtl433
[2022-08-23_20:28:24] Started config rtl433[00000001] listening on t1
(rtl433) using command: /usr/bin/rtl_433  -d 0 -F csv -f 868.95M
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_433  -d 0 -F csv -f 868.95M"
(serialcmd) opened /bin/sh pid 17 fd 5 (rtl433)
(main) regular reset of rtl433  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtl433
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) received ascii "rtl_433 version 21.12-137-gb1e39852 branch master at 202208130705 inputs file rtl_tcp RTL-SDR<0A>Use -h for usage help and see https://triq.org/ for documentation.<0A>Trying conf file at "rtl_433.conf"...<0A>Trying conf file at "(null)/.config/rtl_433/rtl_433.conf"...<0A>Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...<0A>Trying conf file at "/etc/rtl_433/rtl_433.conf"...<0A><0A>New defaults active, use "-Y classic -s 250k" for the old defaults!<0A><0A>Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]<0A>time,msg,codes,model,button,id,channel,battery_ok,temperature_C,mic,subtype,humidity,state,status,rain_rate,rain_rate_mm_h,rain_rate_in_h,rain_total,rain_mm,rain_in,gust,average,direction,wind_max_m_s,wind_avg_m_s,wind_dir_deg,pressure_hPa,uv,power_W,energy_kWh,radio_clock,sequence,unit,group_call,command,dim,dim_value,dipswitch,rbutton,temperature_F,data,closed,event,tamper,esn,exception,status_hex,xactivity,xtamper1,xtamper2,setpoint_C,switch,cmd,cmd_id,tristate,wind_avg_km_h,wind_max_km_h,signal,sensor_code,uv_sensor_id,uv_status,uv_index,lux,wm,seq,current,interval,learn,message_type,sequence_num,leak_detected,wind_avg_mi_h,storm_dist,strike_count,strike_distance,active,raw_msg,rfi,wind_max_mi_h,wind_approach,flags,maybetemp,binding_countdown,depth_cm,power0_W,power1_W,power2_W,power0,node,ct1,ct2,ct3,ct4,batt_Vrms,temp1_C,temp2_C,temp3_C,temp4_C,temp5_C,temp6_C,pulse,button1,button2,button3,button4,group,transmit,moisture,type,pressure_PSI,battery_mV,pressure_kPa,pulses,pulsecount,len,to,from,payload,motion,opened,water,contact_open,reed_open,alarm,heartbeat,temperature_1_C,temperature_2_C,test,humidity_1,newbattery,heating,heating_temp,uvi,light_lux,pm2_5_ug_m3,estimated_pm10_0_ug_m3,counter,code,repeat,maybe_battery,moving,unknown,unknown_3,delay,raw_message,switch1,switch2,switch3,switch4,switch5,extradata,house_id,module_id,sensor_type,sensor_count,alarms,sensor_value,mode,version,type_string,CI,AC,ST,CW,sn,knx_ctrl,src,dst,l_npci,tpci,apci,crc,M,C,data_length,average_temperature_1h_C,average_temperature_24h_C,average_humidity_1h,average_humidity_24h,minimum_temperature_1h_C,maximum_temperature_1h_C,minimum_temperature_24h_C,maximum_temperature_24h_C,minimum_humidity_1h,maximum_humidity_1h,minimum_humidity_24h,maximum_humidity_24h,counter_0,counter_1,failed,battery,alert,secret_knock,relay,wind_dev_deg,exposure_mins,transmit_s,button_id,button_name,pulses_per_kwh,cumulative_kWh,effect,encrypted,current_A,voltage_V,pairing,connected,gap,impulses,triggered,storage,boost,ad_raw,power_idx,power_max,timestamp,rain,physical_tamper,ert_type,encoder_tamper,consumption_data,else,restore,supervised,spidernet,repeater,ProtocolID,EndpointType,EndpointID,Consumption,Tamper,PacketCRC,MeterType,from_id,to_id,msg_type,msg_type_str,extended,hops_max,hops_left,formatted,cmd_dat,msg_str,hopsmax,hopsleft,PacketTypeID,PacketLength,HammingCode,ApplicationVersion,ERTType,ERTSerialNumber,ConsumptionIntervalCount,ModuleProgrammingState,Unknown_field_1,LastGenerationCount,Unknown_field_2,TamperCounters,AsynchronousCounters,PowerOutageFlags,LastConsumptionCount,DifferentialConsumptionIntervals,TransmitTimeOffset,MeterIdCRC,rolling,fixed,remote_id,sendmode,control,retransmission,rain1,rain2,startup,light_klx,message_subtype,raw,rain2_mm,temperature_alarm,timer,timer_active,timer_alarm,meat,taste,id0,id1,switch_id,pad_id,pin,net_id,message,text,ids,time_request,flame_status,zone,setpoint,cycle_rate,minimum_on_time,minimum_off_time,proportional_band_width,device_number,failsafe_mode,ticker,heat_demand,boiler_modulation_level,datetime,domain_id,demand,zone_idx,max_flow_temp,pump_run_time,actuator_run_time,min_flow_temp,token,cmd_name,cmd_value,dst_id,mac,storm_dist_ [... cut ...] nc,temperature_1_F,temperature_2_F,facility,wheel,action,longpress,pressure,humidity_2,rain_raw,pilot,light,thermostat,power,front,fan,aux,flame,battery_level,ctr,cmac,encr,stype,value,centrifugal_acc,has_tick,tick,pm10_0_ug_m3,co2_ppm,ext_power,flags_1,volume_gal,flags_2<0A>Found Rafael Micro R820T tuner<0A>Exact sample rate is: 1000000.026491 Hz<0A>[R82XX] PLL not locked!<0A>Sample rate set to 1000000 S/s.<0A>Tuner gain set to Auto.<0A>Tuned to 868.950MHz.<0A>Allocating 15 zero-copy buffers<0`

megacar666, 2022-08-23:
I dont know how to paste it without word wrap, here is a screenshot for better visibility
[image]

megacar1, 2022-08-31:
hi, any ideas? I'm still blindly changing frequencies by 0.1mhz...also, i have got a monopole telescopic antenna.
i tried it in full wavelenght (69cm), half (34.5cm) and quarter(17cm)

the thing that puzzles me is - i get better results with dongle stock antenna (which is a random size of 8cm) si in theory it should have much worse performance then the telescopic one I am using now.
Nonetheless, I am getting better results (10 telegrams caught, but all with CRC errors) with that stock antenna of "wrong" length, then with "correct" antenna (only 1 or 2 telegrams caught, also CRC errors)
how come??

BIBOLV, 2023-01-02:
Hi,
Were you able to get it working properly?

megacar1, 2023-01-03:
Hi,
No, I gave up

BIBOLV, 2023-01-03:
Do you want to continue to work on this issue or we can close it for now?

megacar1, 2023-01-04:
Hi, I would like to continue, but since I got only one reply in 4 months, I don't anticipate any progress.....

BIBOLV, 2023-01-04:
Ok, can we start from basics - could you confirm that your meter has marking 434 OMS on meter where Interface is described?
[image]

megacar1, 2023-01-05:
it doesnt say
[image]

megacar1, 2023-01-06:
sure, i will tape it with isolator tape and see...
It seems unlikely to me they are doing this manual reading on 1500 apartments in this compound monthly, I never saw anyone...
If they are indeed coming, it would probably be easier/faster to just read the numbers off the screen, then to connect 2 naked wires to some kind of serial adapter....

these meters are located in  in-wall cabinet with metal doors...

I'm curious now to understand how they are doing exact reading, never miss a month, it may indeed be manual reading...

so, the telegrams I catch with your software, they are wmbus protocol or it can be anything? if they are wmbus protocol, then what can it be? electrical meters?

BIBOLV, 2023-01-06:
Booby trap also meter lid, so you would know if someone opens it. 
If there will be no evidences of manual reading then it is good question. You can call them and ask, maybe someone will tell you.
They are wmbus but probably too far away to get proper reading, as I imagine you did try to adjust freq but still did not get correct one. Or it is some device that does not send proper telegrams, but this is unlikely. And, yes, it could be anything, including electricity meter.

megacar1, 2023-01-10:
btw what is too far away? this cabinet is some 20-30 meters from my antenna.....

BIBOLV, 2023-01-11:
[Here](https://github.com/weetmuts/wmbusmeters/issues/580) one user provided info that he got with best receiver 20 meters within concrete walls, [here](https://radiocrafts.com/kb/range-wireless-m-bus/) is mentioned 100m, so it really depends on your specific case. Can you get your receiver closer to test if signal is emitted from that cabinet at all, or from different source?

closed 2023-04-18 by github-actions[bot]

Eddie979, 2024-09-25:
Hi, I have same issue but I think it is due to a different problem source.
My meter is a WMBus Metersit (gas meter) that broadcast @169MHz: log telegram is the following:

(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.

Do you know if such a meter brand could work?

Thank you in advance