wmbusmeters/wmbusmeters #591: the key pollinterval in /etc/wmbusmeter.conf is not recognized

almerindo, 2022-08-25:
Hello!
I'm trying to get the information every 5 minutes. But, currently, I receive it every 8 seconds. So, I tried to change the /etc/wmbusmeter.conf with the key pollinterval=XXX, but this key was not recognized

```
format=json
meterfiles=/var/lib/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
loglevel=normal
device=rtlwmbus:434.47M
logtelegrams=false
donotprobe=/dev/ttyACM2
meterfilesnaming=name-id
meterfilestimestamp=hour
ignoreduplicates=true
oneshot=true
resetafter=1h
pollinterval=500
shell=/usr/bin/mosquitto_pub -h localhost -p 1883 -u pi -P rews2002 -t mangueiras/watermeter/"$METER_ID" -m "$METER_JSON"
```

meter:
```
name=casa123
type=izar
id=XXXXXX
key=NOKEY
```

The meter information was published correctly using MQTT, but I'm receiving this message:
`Aug 25 13:36:51 raspberrypi wmbusmetersd[8240]: No such key: pollinterval`

weetmuts, 2022-08-28:
The pollinterval is set on each meter, so you add pollinterval=xyz 
in the meter file.

However, the pollinterval is ignored when you receive async radio 
telegrams from your meter, using for example rtlwmbus.

The pollinterval is currently used for mbus meters, that require polling
over wired connection. It will also be used for c2/t2/s2 meters that require 
polling over radio.

weetmuts, 2022-08-28:
I did think about having pollinterval in the wmbusmeters.conf would set a default poll interval
for all meters, however I decided against it. Overpolling will degrade the battery in your wmbus meters
so it is better if you specifify the pollinterval explicitly per meter.

weetmuts, 2022-08-28:
So, I have been thinking about a feature that would change the shell command from being executed every
time a telegram is received (or polled) to instead sending an update regularly, since the content of the meter
is already cached in wmbusmeters.

Simplest case woudl be --regularshell=5m 
to have it executed every 5 minutes. But perhaps one also wants to specify 15 past every hour?
--regularshell=%%:15 
or once per day --regularshell=14:30

schadom, 2022-11-09:
> So, I have been thinking about a feature that would change the shell command from being executed every time a telegram is received (or polled) to instead sending an update regularly, since the [...]

+1 for this feature

testuser7, 2022-11-30:
Are there any updates regarding the problem?
As a workaround, we can use the bash script:
```
while true; do ./wmbusmeters --oneshot auto:t1 MyWater izar 'xxxxxxxx' NOKEY; sleep $(( 600 - $(date +%s) % 600 )); done
```

weetmuts, 2023-02-02:
Still on the TODO, but not high in the queue! :-)

marcelstoer, 2023-04-17:
Thanks to the pointer to this issue I learned about the `--oneshot` switch; must have missed it in the README. It should allow me to now trigger wmbusmeters from cron.