wmbusmeters/wmbusmeters #593: ((wmbus) WARNING! decrypted payload crc failed check, did you use the correct decryption key?

HinrichF, 2022-08-27:
Hi

We are busy integrating Kamstrup Multical21 meters into our own AMS. 

We got a lab setup with one Multical21 and a Rx1180 currently connected via USB to and Raspberry Pi 4.

We unpacked the kem file supplied with the meter (identifiers are redacted):

python3 kem-decryptor.py multical21_12345678.zip.kem XXXX

```
Detected a zip file on input ... extracting
Found meter MC21 (02140C01894)
    number : 12345678
    serial : 12345678
    type   : VolumeCold
    driver : multical21
    config : 0100200023133
    key    : 9FB48E422263949D220DC3B763B95B90
    meter file: 12345678
```

Extracted info from kem file matches the meter at hand (serial, type, config all match)

after building wmbusmeters from code we ran:

`wmbusmeters --verbose /dev/ttyUSB0:rc1180:t1 MyTapWater auto 12345678 9FB48E422263949D220DC3B763B95B90`

resulting in the following output:

```
_(wmbusmeters) version: 1.8.0-13-ga5f0783
(config) using device: /dev/ttyUSB0:rc1180:t1
(config) number of meters: 1
(serial) tty /dev/ttyUSB0 can be accessed
(serialtty) opened /dev/ttyUSB0 fd 3 (detect rc1180)
(serialtty) closed /dev/ttyUSB0 (detect rc1180)
(rc1180) are you there? yes 78563412
Started config rc1180[78563412] on /dev/ttyUSB0 listening on t1
(rc1180) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0 fd 3 (rc1180)
(rc1180) get config to get device id
(rc1180) got device id 78563412
(main) regular reset of rc1180  on /dev/ttyUSB0 will happen every 82800 seconds
(meter) created MyTapWater multical21 12345678 encrypted
(meter) used meter template MyTapWater  auto to match 12345678
(meter) started meter 1 (MyTapWater 12345678 auto)
(meter) MyTapWater(1) multical21  handling telegram from 12345678
(telegram) DLL L=22 C=44 (from meter SND_NR) M=2c2d (KAM) A=24261145 VER=1b TYPE=16 (Cold water meter) (driver multical21) DEV=rc1180[78563412] RSSI=0
(wmbus) WARNING! decrypted payload crc failed check, did you use the correct decryption key? e38b payload crc (calculated 6035) Permanently ignoring telegrams from id: 12345678 mfct: (KAM) Kamstrup Energi (0x2c2d) type: Cold water meter (0x16) ver: 0x1b
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=00 SN=fc792b20 (AES_CTR session=12 time=178079) CRC=e38b
(meter) newly created meter (MyTapWater 12345678 multical21) did not handle telegram!
(wmbus) telegram from 12345678 ignored by all configured meters!_
```

we waited for more than 8 messages to come through from the meter but they all come with the same warning.

Before we try to get a new kem file issued, we want to make sure that we did not miss anything.

We hope somebody can push us in the right direction to sove this issue.

regards

Hinrich

weetmuts, 2022-08-27:
If you can run with --debug and email me the log or post it here, including the meter id and key, then I can take a look.

Most likely it ís something wrong with the key though.....

So your multical21 sends on t1? 

Are there more than one multical21 that you can receive?

HinrichF, 2022-08-28:
Hi Fredrik

Thanks for coming back to me so quickly! 

I ran below leaving the meter to be probed automatically to avoid the szenario where the meter would not send on T1.

`wmbusmeters --debug /dev/ttyUSB0:rc1180:t1 MyTapWater auto 12345678 9FB48E422263949D330DC3B113B95FF0`

Full wMbus frame is received after some time and the multical21 seems to be identified correctly. 
after that the CRC check of Block 0 fails.

Maybe the Input Vector is wrong?

I sent you output and supplied kem file via email. 

regards

H.

weetmuts, 2022-08-28:
Many thanks for the test data! I have found the problem! The telegram bytes that you get from your rc1180 includes an extra byte! Since it is AES_CTR this extra byte will just be politely decrypted but the payload crc will fail, as expected.

If you manually drop the last byte and change the length value of the first byte. Then it decrypts properly.

    "total_m3":1.177,
    "target_m3":0.072,
    "current_status":"",
    "time_dry":"22-31 days",

I do not have access to my rc1180 for the moment. But it looks like the rssi is not removed properly in the code.
Please test the following new code in wmbus_rc1180.cc :
```
        if (status == FullFrame)
        {
            vector<uchar> payload;
            int rssi = 0;
            bool extra_byte = device_config_.usingRssi();

            if ((!extra_byte && payload_len > 0) || (extra_byte && payload_len > 1))
            {
                if (device_config_.usingRssi())
                {
                    rssi = read_buffer_[payload_offset+payload_len-1];
                    payload_len--;
                }
                uchar l = payload_len;
                payload.insert(payload.end(), &l, &l+1); // Re-insert the len byte.                                                          
                payload.insert(payload.end(), read_buffer_.begin()+payload_offset, read_buffer_.begin()+payload_offset+payload_len);
            }
            read_buffer_.erase(read_buffer_.begin(), read_buffer_.begin()+frame_length);
            AboutTelegram about("rc1180["+cached_device_id_+"]", rssi, FrameType::WMBUS);
            handleTelegram(about, payload);
        }
```

HinrichF, 2022-08-28:
Great! Well done! You suspected correctly that the last Byte is RSSI related. I disabled RSSI on the RC1180 and it all works! 

Again thanks a lot for your quick response and help!

```

(ELL) decrypting "8D1174835D34BD5BF8F917968B40B99D4B"
(ELL) IV 2D2C451126241B1620C0AF2B20000000
(ELL) block 0 block_size 16 offset 0
(ELL) decrypted "xxxx"
(ELL) block 1 block_size 1 offset 16
(ELL) decrypted "00"
(ELL) decrypted "xxxxx"
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=19 SN=c0af2b20 (AES_CTR session=0 time=178940) CRC=ae27
(wmbus) parseNWL @19 15
(wmbus) parseAFL @19 15
(wmbus) parseTPL @19 15
(wmbus) using hard coded format for hash dd34
(dvparser) using format "02FF2004134413"
(telegram) TPL CI=79
(multical21) 000   : 21 length (33 bytes)
(multical21) 001   : 44 dll-c (from meter SND_NR)
(multical21) 002   : 2d2c dll-mfct (KAM)
(multical21) 004   : 45112624 dll-id (24261145)
(multical21) 008   : 1b dll-version
(multical21) 009   : 16 dll-type (Cold water meter)
(multical21) 010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(multical21) 011   : 20 ell-cc (slow_resp sync)
(multical21) 012   : 19 ell-acc
(multical21) 013   : c0af2b20 sn (AES_CTR)
(multical21) 017   : ae27 payload crc (calculated ae27 OK)
(multical21) 019   : 79 tpl-ci-field (EN 13757-3 Application Layer with Compact frame (no tplh))
(multical21) 020   : 34dd format signature
(multical21) 022   : 3109 data crc
(multical21) 024 C!: 7000 info codes ((dry 22-31 days))
(multical21) 026 C!: 25050000 total consumption (1.317000 m3)
(multical21) 030 C!: 48000000 target consumption (0.072000 m3)
**MyTapWater      12345678        1.317 m3        0.072 m3        0 m3/h  127 °C  127 °C  (dry 22-31 days)        2022-08-28 10:20.50**
```

HinrichF, 2022-08-28:
Solution for now: 

Disable RSSI on RC1180.

closed 2022-08-28 by HinrichF