wmbusmeters/wmbusmeters #596: Add support for Zenner EDC communication module

atze09, 2022-08-30:
I just added a Zenner EDC communication module to my wired m-bus.
I tried it with minomess but got:
{
  "media": "water",
  "meter": "minomess",
  "name": "ELW_Kaltwasser",
  "id": "xxxxxxxx",
  "total_m3": 0,
  "meter_date": null,
  "target_m3": null,
  "target_date": null,
  "status": "OK",
  "timestamp": "2022-08-30T19:38:59Z",
  "device": "MAIN",
  "rssi_dbm": 0

 Is it possible to add support for this device? I can help you with any telegrams?

weetmuts, 2022-08-30:
Yes, please provide some telegram=|....| lines  when using --logtelegrams
(logtelegrams=true)

atze09, 2022-08-30:
> Yes, please provide some telegram=|....| lines when using --logtelegrams (logtelegrams=true)

First:
telegram=|6874746808007258510051496A000712000000_0C7858510051046D2414DE280413000000000C943C000000004413FFFFFFFF426CFFFF840113FFFFFFFF82016CFFFFC40113FFFFFFFFC2016CFFFF840213FFFFFFFF82026CFFFF043B000000000422E62F000004260000000034220000000002FD1700001F|+4

Is this what you need?

atze09, 2022-08-30:
telegram=|6874746808007258510051496A000712000000_0C7858510051046D2414DE280413000000000C943C000000004413FFFFFFFF426CFFFF840113FFFFFFFF82016CFFFFC40113FFFFFFFFC2016CFFFF840213FFFFFFFF82026CFFFF043B000000000422E62F000004260000000034220000000002FD1700001F|+4
telegram=|6874746808007258510051496A000713000000_0C7858510051046D2514DE280413000000000C943C000000004413FFFFFFFF426CFFFF840113FFFFFFFF82016CFFFFC40113FFFFFFFFC2016CFFFF840213FFFFFFFF82026CFFFF043B000000000422E62F000004260000000034220000000002FD1700001F|+66
telegram=|6874746808007258510051496A000714000000_0C7858510051046D2614DE280413000000000C943C000000004413FFFFFFFF426CFFFF840113FFFFFFFF82016CFFFFC40113FFFFFFFFC2016CFFFF840213FFFFFFFF82026CFFFF043B000000000422E62F000004260000000034220000000002FD1700001F|+129

weetmuts, 2022-08-30:
Yep! Many thanks, I will take a look. It is a new meter right? Most of the values are zero.

weetmuts, 2022-08-30:
This is the content:
```
Auto driver  : minomess
Best driver  : aventieswm 22/58
Using driver : minomess 00/00
000   : 68 start
001   : 74 length (116 bytes)
002   : 74 length again (116 bytes)
003   : 68 start
004   : 08 dll-c (RSP_UD2)
005   : 00 dll-a primary (0)
006   : 72 tpl-ci (long header)
007   : 58510051 tpl-id (51005158)
011   : 496a tpl-mfct (ZRI)
013   : 00 tpl-version
014   : 07 tpl-type (Water meter)
015   : 12 tpl-acc-field
016   : 00 tpl-sts-field (OK)
017   : 0000 tpl-cfg 0000 ( )
019   : 0C dif (8 digit BCD Instantaneous value)
020   : 78 vif (Fabrication no)
021 C?: 58510051
025   : 04 dif (32 Bit Integer/Binary Instantaneous value)
026   : 6D vif (Date and time type)
027 C?: 2414DE28
031   : 04 dif (32 Bit Integer/Binary Instantaneous value)
032   : 13 vif (Volume l)
033 C!: 00000000 ("total_m3":0)
037   : 0C dif (8 digit BCD Instantaneous value)
038   : 94 vif (Volume 10⁻² m³)
039   : 3C combinable vif (BackwardFlow)
040 C?: 00000000
044   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
045   : 13 vif (Volume l)
046 C?: FFFFFFFF
050   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
051   : 6C vif (Date type G)
052 C?: FFFF
054   : 84 dif (32 Bit Integer/Binary Instantaneous value)
055   : 01 dife (subunit=0 tariff=0 storagenr=2)
056   : 13 vif (Volume l)
057 C?: FFFFFFFF
061   : 82 dif (16 Bit Integer/Binary Instantaneous value)
062   : 01 dife (subunit=0 tariff=0 storagenr=2)
063   : 6C vif (Date type G)
064 C?: FFFF
066   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
067   : 01 dife (subunit=0 tariff=0 storagenr=3)
068   : 13 vif (Volume l)
069 C?: FFFFFFFF
073   : C2 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
074   : 01 dife (subunit=0 tariff=0 storagenr=3)
075   : 6C vif (Date type G)
076 C?: FFFF
078   : 84 dif (32 Bit Integer/Binary Instantaneous value)
079   : 02 dife (subunit=0 tariff=0 storagenr=4)
080   : 13 vif (Volume l)
081 C?: FFFFFFFF
085   : 82 dif (16 Bit Integer/Binary Instantaneous value)
086   : 02 dife (subunit=0 tariff=0 storagenr=4)
087   : 6C vif (Date type G)
088 C?: FFFF
090   : 04 dif (32 Bit Integer/Binary Instantaneous value)
091   : 3B vif (Volume flow l/h)
092 C?: 00000000
096   : 04 dif (32 Bit Integer/Binary Instantaneous value)
097   : 22 vif (On time hours)
098 C?: E62F0000
102   : 04 dif (32 Bit Integer/Binary Instantaneous value)
103   : 26 vif (Operating time hours)
104 C?: 00000000
108   : 34 dif (32 Bit Integer/Binary Value during error state)
109   : 22 vif (On time hours)
110 C?: 00000000
114   : 02 dif (16 Bit Integer/Binary Instantaneous value)
115   : FD vif (Second extension FD of VIF-codes)
116   : 17 vife (Error flags (binary))
117 C!: 0000 ("status":"OK")
```

atze09, 2022-08-30:
> Yep! Many thanks, I will take a look. It is a new meter right? Most of the values are zero.

The analogue meter is old, but the EDC M-BUS module is new and i just added it to the meter, maybe that is the reason why the values are empty.

weetmuts, 2022-08-30:
So the minomess driver is working for the values it can find in the telegram.
However this mbus module uses different storage nrs compare to what is used in the current driver.
Do you have any documentation on this mbus module?

atze09, 2022-08-30:
> So the minomess driver is working for the values it can find in the telegram. However this mbus module uses different storage nrs compare to what is used in the current driver. Do you have any documentation on this mbus module?

Great, on their website i found some documentation in german.
https://zenner.de/products/sys_edc_communication_module/

atze09, 2022-08-30:
I can help you translate.

weetmuts, 2022-09-01:
Thanks! I pushed improvements to the minomess driver. 

It seems that the wireless sends a more compact telegram compared
to the wired version. Added the necessary fields. It seems like the previous month value and date is in storage 8
for the wireless but storage 1 in the wired. Odd... perhaps you can monitor your meter when it switches month to see that the target_m3 and target_date is correct.

Please test!

weetmuts, 2022-09-01:
I have pushed another fix that makes sure --logtelegrams print the whole telegram including the crc and 0x16 at the end.
Can you verify that you get telegram=|68.......16| and that you can do 
wmbusmeters --analyze 68.....16 
and it will print something nice.

atze09, 2022-09-02:
> Thanks! I pushed improvements to the minomess driver.
>
> It seems that the wireless sends a more compact telegram compared to the wired version. Added the necessary fields. It seems like the previous [...]

MQTT Output is the following;
{
  "media": "water",
  "meter": "minomess",
  "name": "ELW_Kaltwasser",
  "id": "51005158",
  "fabrication_no": "51005158",
  "operating_time_h": 0.0075,
  "on_time_h": 3.422222,
  "on_time_at_error_h": 0,
  "meter_datetime": "2022-09-02 06:33",
  "total_m3": 0.063,
  "total_backward_m3": 0,
  "volume_flow_m3h": 0,
  "meter_date": null,
  "target_m3": 4294967.295,
  "target_date": "2127-15-31",
  "status": "OK",
  "timestamp": "2022-09-02T05:47:32Z",
  "device": "MAIN",
  "rssi_dbm": 0
}
Target m³ and target date is very high, does this mean the volume/date where i should change the meter?

atze09, 2022-09-02:
> I have pushed another fix that makes sure --logtelegrams print the whole telegram including the crc and 0x16 at the end. Can you verify that you get telegram=|68.......16| and that you can do wmbusmeters --analyze 68.....16 and it will print something nice.

Here a few telegrams:
telegram=|6874746808007258510051496A00073C000000|0C7858510051046D1F06C22904133F0000000C943C000000004413FFFFFFFF426CFFFF8401132200000082016CC129C40113FFFFFFFFC2016CFFFF8402133600000082026CC229043B0000000004222030000004261B00000034220000000002FD1700001F|5D16|+5
telegram=|6874746808007258510051496A00073D000000|0C7858510051046D2006C22904133F0000000C943C000000004413FFFFFFFF426CFFFF8401132200000082016CC129C40113FFFFFFFFC2016CFFFF8402133600000082026CC229043B0000000004222030000004261B00000034220000000002FD1700001F|5F16|+67
telegram=|6874746808007258510051496A00073E000000|0C7858510051046D2106C22904133F0000000C943C000000004413FFFFFFFF426CFFFF8401132200000082016CC129C40113FFFFFFFFC2016CFFFF8402133600000082026CC229043B0000000004222030000004261B00000034220000000002FD1700001F|6116|+129
telegram=|6874746808007258510051496A00073F000000|0C7858510051046D2306C22904133F0000000C943C000000004413FFFFFFFF426CFFFF8401132200000082016CC129C40113FFFFFFFFC2016CFFFF8402133600000082026CC229043B0000000004222030000004261B00000034220000000002FD1700001F|6416|+191
telegram=|6874746808007258510051496A000740000000|0C7858510051046D2406C22904133F0000000C943C000000004413FFFFFFFF426CFFFF8401132200000082016CC129C40113FFFFFFFFC2016CFFFF8402133600000082026CC229043B0000000004222030000004261B00000034220000000002FD1700001F|6616|+253

I am currently only using home assistant addon, there i cannot execute "wmbusmeters --analyze", or is there any way?

weetmuts, 2022-09-02:
Yes, you can use https://wmbusmeters.org
Here is a direct link to decode your telegram.

I made a small cleanup so meter_date which is supplied by the wmbus version does not show if
there is a meter_datetime which is supplied by the wired mbus version.

weetmuts, 2022-09-02:
https://wmbusmeters.org/analyze/6874746808007258510051496A0007400000000C7858510051046D2406C22904133F0000000C943C000000004413FFFFFFFF426CFFFF8401132200000082016CC129C40113FFFFFFFFC2016CFFFF8402133600000082026CC229043B0000000004222030000004261B00000034220000000002FD1700001F6616

weetmuts, 2022-09-02:
As you can see there are 4 storage slots. I was just guessing that storage 1 was the most recent month,
but apparently that might not be correct....

atze09, 2022-09-02:
Aweosome thank you!

weetmuts, 2022-09-02:
Your welcome! When you have used the meter for a while, I suppose it will be obvious what the storages are used for.
Please come back here and tell us the meaning so that we can finish the driver!

BIBOLV, 2023-01-02:
Closing issue for now as solution was provided. Please reopen issue once you have collected telegrams with other storage slots being used.

closed 2023-01-02 by BIBOLV

atze09, 2023-01-13:
I just ordered a second EDC meter
And now the company wrote the water meter id to the EDC.
And I got this error message.
I checked hardware and software revisions.
And those are working with the same configuration.
I am confused
Here is my log file, thank you in advance.
(meter) ELW_Warmwasser: meter detection did not match the selected driver minomess! correct driver is: unknown!
(meter) Not printing this warning again for id: 00000000 mfct: (@@@) Unknown (0x00) type: Other (0x00) ver: 0x00
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|6874746808007248005823496A000600000000|0C7811230151046D1914ED210413000000000C943C000000004413FFFFFFFF426CFFFF840113FFFFFFFF82016CFFFFC40113FFFFFFFFC2016CFFFF840213FFFFFFFF82026CFFFF043B000000000422C635000004260000000034220000000002FD1700001F|B816|+5

Best regards,
Dominik

atze09, 2023-01-14:
> en you have used the meter for a while

After that while i wanted to check that, but analysing the telegram wmbusmeters.org gives me:

Using: wmbusmeters: 1.11.0-30-g554b91b
554b91bc6b17c2781369cecd4040aac489c04cd9

Did i do anything wrong?

atze09, 2023-01-14:
> After that while i wanted to check that, but analysing the telegram wmbusmeters.org gives me:
> 
> Using: wmbusmeters: 1.11.0-30-g554b91b [554b91b](https://github.com/weetmuts/wmbusmeters/commit/554b91bc6b17c2781369cecd4040aac489c04cd9)
> 
> Did i do anything wrong?

> Your welcome! When you have used the meter for a while, I suppose it will be obvious what the storages are used for. Please come back here and tell us the meaning so that we can finish the driver!

After that while i wanted to check that, but analysing the telegram wmbusmeters.org gives me:

Using: wmbusmeters: 1.11.0-30-g554b91b
https://github.com/weetmuts/wmbusmeters/commit/554b91bc6b17c2781369cecd4040aac489c04cd9

Did i do anything wrong?