wmbusmeters/wmbusmeters #597: New water meter - This time LoRa. End of reception?

pswiatki, 2022-08-30:
I was recently duped into replacing the meter by my local water company (the meter is their property, so they can do whatever they want - just have to seek permission to enter the house's premises). They told me their sentinels detected a possible no-count scenario, so the meter getting stuck. I believed and did not immediately check the logs gathered by `wmbusmeters`. My bad. They lied the suspicion was confirmed and proceeded to replace the meter. To this "latest-and-greatest":

[image]

As can be seen - Has a nice antenna and a transmitter from aiut.com. It uses LoRa to transmit. Does anyone know if there is any chance to pick it up directly? I am not even sure if this transmits anything on its own, or only when polled from the network. But I can see some periodic activity on a spectrogram which I suspect of coming from this box. More tests and experiments to follow.
Will let you know when I have the results.

weetmuts, 2022-08-31:
How very annoying! So, LORA uses a different radio modulation that reach much much longer distances.

Its is pretty clever, alas it is proprietary, however it has been reverse engineered: https://revspace.nl/DecodingLora
There is even a rtlsdr decode: https://github.com/myriadrf/LoRa-SDR

If you could try to capture some radio transmissions, then it would be fun to see them!

Unfortunately they are most likely encrypted. What the installer meant is that the transmissions are collected by LORA gateways managed by LORA operators that are placed all around the country and relays radio to internet. So to listen to your lora meter you only need to implement a web server.  The disadvantage is that you have to pay to the lora gateway operator for this service. And alas, the web server is already implemented by your water company, or some cloud
company that implements this service.

However you do not to configure your own meter to use LORA-wan (ie the gateway etc) it is possible to listen to your own lora meter using a lora dongle or raspberry pi addon board. I got some of those to play around with this.
There might be use cases where people thinks it is a bit overkill to send the meter data all the way to some cloud and then get it back. Just talk to the meter locally using wmbusmeters. But again you have to own your meter to be able to have the encryption keys for it.

pswiatki, 2022-08-31:
Yes, I know LoRa is a completely different modulation technique and a different system (as a whole), too. 
Thanks for the links - the Dutch link I found before, but LoRa-SDR is new to me. I have to investigate it and try
if I can receive any LoRa packets at my location. We will see.
As far as encryption is concerned: the previous, `wmbus` meter did not use any encryption. So, I am still able to pick up many of the water meters in the neighbourhood. I wonder if it is at all possible to use LoRa for some JSON (or whatever other readable format) transfer to the central database of the water company here, in Wrocław POLSKA. Time will tell.

gstrzelecki, 2022-09-18:
@pswiatki, I'm in the same situation, with the new LoRa meter. Have you tried to sniff these messages?

pswiatki, 2022-09-18:
Not yet, but soon. Stay tuned...

grest, 2022-10-14:
> Not yet, but soon. Stay tuned...

and? :)

weetmuts, 2023-01-21:
Still interested in LoRa captures! :-)

grest, 2023-02-06:
> Not yet, but soon. Stay tuned...

Hi, I guess you stopped working on that?

pswiatki, 2023-02-13:
I got busy at work and could not do much with the above in the past months. Chances are I will have more time now, so maybe I could come back to this topic.

andrazek, 2023-02-20:
I am also interested. There is a qr code on my meter, could this be a key to decrypt the

pswiatki, 2023-02-20:
No, I doubt this is the meter's key.

pswiatki, 2023-06-05:
Can this be re-opened. I have new information in regards to LoRa here.

pswiatki, 2023-06-05:
Thanks a lot for re-opening. A technician from the water company visited me this morning announcing a need to replace the transmitter (again!). This came slightly less than a year from the last replacement (WMBus --> LoRa). I asked why and got the answer they experienced range issues with the first transmitter and came back to replace it. So, I got interested, took some pictures, and spoke with the buy a bit. Turns out the problem was an unreliable **wakeup** of the device due to reduced range with the original antenna. So, as you can see, I heard the keyword. No transmission unless the transceiver gets woken up (instructed to TX its accumulated data, I presume). 

This is a big surprise because I thought the primary reason for going to LoRa was to be able to pick up transmissions from a city-wide LoRa network which would make those trips to get the readings not necessary anymore. It turns out this is not the case. Especially so, at my location - it is a challenge to get the proper reading here. 
It was very similar with WMBus - even I got a very low RF level from my own device and at the same time much stronger signal from the neighboring readers. I thought my device was faulty, but it now seems new devices behave similarly (which I could not test myself as it now turned out - no wakeup command for LoRa TX).

Finally, the latest transmitter is exactly of the same type - the only difference is a piece of regular coaxial cable instead of the previous antenna (which was not removable) and poorly protected RF connectors that attach a longer piece of coax with an antenna on a magnet mount at the end of it. Looks like a makeshift solution, we'll see if it works at all. They reasoned the higher (closer to the ground) the antenna will be the better the signal.

If you are interested, I can attach pictures of these devices and some LoRa test equipment tomorrow.

pswiatki, 2023-09-02:
A new project that may help with further investigation:
https://github.com/tapparelj/gr-lora_sdr

closed 2023-12-02 by github-actions[bot]

nefh, 2024-04-01:
Did you have any further progress since?

bartekmotyl, 2024-09-30:
@pswiatki any news? I am in exactly the same situation (MPWIK just replaced my good-old wmbus meter into this LoRa thing and I am no longer able to get any readings). 
I have a LoRa dongle and can try to sniff some packets, but I am afraid that the whole communication is encrypted anyway, so rather unlikely we can get any data from these packets...

pswiatki, 2024-09-30:
I do not have the appropriate LoRa hardware now to sniff. Try sniffing and report back. It will be interesting to see any communication coming out of the meter.