wmbusmeters/wmbusmeters #598: Missing driver for Apator E-ITN 30.51 HCA, A lot of Unknown tpl-ci-field b6 in logs

kitor, 2022-08-31:
Hello,
I live in a multi story building with a lot of wmbus devices. Those are:
- `apator162` water meters
- `amiplus` power meters
- `Apator E-ITN 30.51` HCAs (not supported yet [1])

I use rtl-sdr, wmbusmeters config is (frequency was fine-tuned with spectrum analysis):
```
loglevel=normal
device=rtlwmbus:868.993M:t1
logtelegrams=false
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=append
logfile=/var/log/wmbusmeters/wmbusmeters.log
```

`wmbusmeters` is spamming a ton of `(wmbus) Unknown tpl-ci-field b6` messages, doesn't matter if I have any sensor configured in `/etc/wmbusmeters.d/*` or not (running in `logtelegrams=true` mode). 

What causes this message? Is this related to any of non-supported devices? I guess so, as those usually prepend HCA messages [1]

There's a sensor documentation (Polish only) but it doesn't tell anything about the protocol itself.
http://www.apator.com/uploads/files/Produkty/Podzielnik_kosztow_ogrzewania/i-pl-021-2016-e-itn-30-51-30-6.pdf
At least it lists that 30.51 meter transmits:
* serial number
* season start date
* read date
* electronic seal break date
* previous season average temperature
* this season average temperature
(there's also 30.6 with a bit different data transmitted, but I don't have access to that sensor)

I already partially reverse engineered protocol for `Apator E-ITN 30.51`:
Looks to be a proprietary protocol, which seems to be a theme for Apator devices. At least data is not encrypted in any way.

```
  ?  season      ts?    ?     ?   last     ?  this    current avg   ? 
     start?                       season      season? date    temp
F545 0186   F41B A62B  A0A1 0000  1002   0000  0000   192D    9815 B81A
F545 0186   F41B 985D  A0A1 0000  0000   0000  0000   192D    7415 881A
F545 0186   F41B A358  A0A1 0000  0000   0000  0000   192D    AC15 BC19
F545 0186   F41B 9F3F  A0A1 0000  6400   0000  0000   192D    D015 1C1B
```

`last season` is just BE uint16_t with reading.

`ts?` seems to be some kind of uint16_t timestamp / message identifier. It changes with every message from one sensor, even when other data is not changed.

`current date` is date in BE uint16_t, my guess on format based on more data collected is:
```
2D19h
0010 1101 0001 1001
             1 1001  D   -> 25
        1 000        M   -> 8
  10 110             Y   -> 22
```
Unfortunately that doesn't work for what I guess is season start date (which in all meters is 1.05)

Temperature is BE uint16_t, where upper byte is full degree celcius, and lower is fractional part.
LCD on meter rounds that to full 1/10 deg down, eg:
```
1598h -> 21,(152/256) -> 21,593750 -> 21,5 on LCD
1574h -> 21,(116/256) -> 21,453125 -> 21,4 on LCD
15ACh -> 21,(172/256) -> 21,671875 -> 21,6 on LCD
15D0h -> 21,(208/256) -> 21,812500 -> 21,8 on LCD
```

I'm not sure which field is reading for `current season` (it's still summer to all HCAs that I have access to show 0), as well as for `tamper date` See [2] for more messages with my comments on that.

[1] logtelegrams log example
[... 212 lines cut ...]
F545 0186 F41B A958 A0A1 0000 0000 0000 0000 192D FC 12 D418
F545 0186 F41B A43F A0A1 0000 0000 0000 0000 192D 08 12 8818
F545 0186 F41B 9A67 A0A1 0000 0000 0000 0000 192D 7C 16 F818
F545 0186 F41B A14E A0A1 0000 0000 0000 0000 192D 20 14 F418
F545 0186 F41B A758 A0A1 0000 0000 0000 0000 192D F8 13 D419
F545 0186 F41B A044 A0A1 0000 EF11 0000 0000 192D E0 15 081A
F545 0186 F41B A65D A0A1 0000 D30E 0000 0000 192D 24 16 901B
F545 0186 F41B A658 A0A1 0000 3300 0000 0000 192D 74 15 AC19
F545 0186 F41B 9B3A A0A1 0000 7C01 0000 0000 192D 6C 13 9019
F545 0186 F41B 9553 A0A1 0000 0B03 0000 0000 192D 04 13 9019
F545 0186 F41B A953 A0A1 0000 0300 0000 0000 192D 80 14 1019
F545 0186 F41B 955D A0A1 0000 2200 0000 0000 192D E0 14 8819
F545 0186 F41B 9258 A0A1 0000 CC01 0000 0000 192D 58 14 301A
F545 0186 F41B 993A A0A1 0000 1A01 0000 0000 192D C0 12 2C19
F545 0186 F41B A153 A0A1 0000 1000 0000 0000 192D F0 13 1818
F545 0186 F41B A03A A0A1 0000 6316 0000 0000 192D 1C 16 2019
F545 0186 F41B A03F A0A1 0000 EF11 0000 0000 192D E0 15 081A
F545 0186 F41B 9E5D A0A1 0000 0700 0000 0000 192D E8 13 F818
F545 0186 F41B A662 A0A1 0000 D30E 0000 0000 192D 24 16 901B
						 
F545 0186 F41B 956C A0A1 0000 0000 0000 0000 192D 38 13 F817
F545 0186 F41B A353 A0A1 0000 610C 0000 0000 192D 40 13 E01A
F545 0186 F41B A458 A0A1 0000 7809 0000 0000 192D 64 16 E819
F545 0186 F41B 9658 A0A1 0000 0B03 0000 0000 192D 04 13 9019
```

kitor, 2022-09-01:
I just noticed that there are also some unsupported water meters visible, and those messages are also prepended with that `tpl-ci-field b6` warning, so I guess this might be not related, or both use similar protocol (as APT is also apator):

```
(wmbus) Unknown tpl-ci-field b6
(wmbus) Unknown tpl-ci-field b6
Received telegram from: 00127580
          manufacturer: (APT) Unknown (0x8614)
                  type: Water meter (0x07)
                   ver: 0x05
                device: rtlwmbus[00000001]
                  rssi: 88 dBm
                driver: unknown! 
telegram=|_5E441486807512000507B60EFFFFF5450106C11B9A7BF41B98767A8F0040850797FC0A11A0588DD2A25E6325F5CCCF9ECB62F14996E72ECA269FC9179E20C0F22836702977820F47E3F046CB482DA6B8EFBD2A55719F008CEB3A7BDEEF51AA|+172
Received telegram from: 00143215 
          manufacturer: (APA) Apator, Poland (0x601)
                  type: Water meter (0x07) encrypted
                   ver: 0x05 
                device: rtlwmbus[00000001] 
                  rssi: 34 dBm   
                driver: apator162 
```

weetmuts, 2022-09-01:
Thanks! It turns out that all ci-fields from 0xa0 to 0xb6 seems to be manufacturer specific.
I have now fixed the code so it accepts b6 as well. 

I also add two stub drivers: apatoreitn (the hca) and apt_07_05 (the water)
you can find them in src/driver_apatoreitn.cc and src/driver_apt_07_05.cc

Please hack them to add any decoded data that you can reverse engineer!

kitor, 2022-09-08:
Thank you, this will have to wait a bit until heating starts (as the most interesting value is this season reading). So I'll probably update this issue in a month or two, when there will be some data to work with.

mmmalina, 2022-09-18:
I confirm the EITN 30.51 allocator reading by analyzing your data.

For me, the display shows 89 units for the last heating season and the telegram shows "5900" and 122 units on the allocator display in the current heating season, which is in the telegram "7A00". I checked the telegram reading with the display and everything is correct.

I also checked the temperatures and corrected to the previous season's average and the current average.

This is how it looks according to your table.

[image]

Please update src/driver_apatoreitn.cc

Thanks.

kitor, 2022-09-18:
Radiators just got warm yesterday, so I'll collect a bit of data to confirm the "actual season" field and then make a PR with changes.

Thank you for your inputs!

kitor, 2022-09-20:
Just a small update: Looks like I have a driver working, still requires a bit of polish - I'll make a PR probably later today.
Fields for actual season reading and actual average temperature are confirmed indeed.

```
{'current_date': '2022-9-19T01:00:00Z',
 'current_hca': 0,
 'device': 'rtlwmbus[00000001]',
 'id': 'xxxxxxxx',
 'media': 'heat cost allocation',
 'meter': 'apatoreitn',
 'name': 'heat',
 'previous_hca': 528,
 'rssi_dbm': 11,
 'temp_room_avg_c': 23.609375,
 'temp_room_prev_avg_c': 21.59375,
 'timestamp': '2022-09-20T08:46:00Z'}
```

```
{'current_date': '2022-9-19T01:00:00Z',
 'current_hca': 3,
 'device': 'rtlwmbus[00000001]',
 'id': 'yyyyyyyy',
 'media': 'heat cost allocation',
 'meter': 'apatoreitn',
 'name': 'heat',
 'previous_hca': 0,
 'rssi_dbm': 9,
 'temp_room_avg_c': 23.515625,
 'temp_room_prev_avg_c': 21.671875,
 'timestamp': '2022-09-20T08:50:14Z'}
```

kitor, 2022-09-20:
Yep, see the PR.

mmmalina, 2022-09-20:
The allocator always sends three telegrams with different telegram lengths:
e.g:

1. - 19440186********0408A0A10000590000000100322DE413B415- wmbusmeters doesn't read this so no problem.

2. - 25441486********0408B60AFFFFF2460186EC1B934EA0A10000590000000100322DE413B415 - wmbusmeters reads this telegram well and gives the correct data.

    "media":"heat cost allocation",
    "meter":"apatoreitn",
    "name":"",
    "id":"********",
    "current_hca":1,
    "previous_hca":89,
    "current_date":"2022-09-18T02:00:00Z",
    "season_start_date":"2016-05-01T02:00:00Z",
    "esb_date":"",
    "temp_room_avg_c":21.703125,
    "temp_room_prev_avg_c":19.890625,
    "timestamp":"2022-09-20T23:18:52Z"

3. - 29441486********0408B60EFFFFF1460186EC1B934EE91BA57BA0A10000590000000100322DE413B415 - wmbusmeters reads this telegram and gives invalid data.

    "media":"heat cost allocation",
    "meter":"apatoreitn",
    "name":"",
    "id":"********",
    "current_hca":89,
    "previous_hca":41376,
    "current_date":"",
    "season_start_date":"2020-08-27T02:00:00Z",
    "esb_date":"",
    "temp_room_avg_c":45.195312,
    "temp_room_prev_avg_c":0.003906,
    "timestamp":"2022-09-20T23:20:09Z"

Is it possible for wmbusmeters to read only one telegram? I have this data imported to domoticz, which constantly enters incorrect data from the third telegram.

kitor, 2022-09-21:
Sorry, but I was collecting telegrams from 20+ meters for over 24 hours and never saw a single telegram that was different length than the one that is decoded properly.

You can change 
```
        if (content.size() < 26)
```
in my implementation to fail if it is not exactly 26 in length. 

If you compare 2nd and 3rd telegram, you have some noise inside ( `BA57BA0A` ) which is not even 8 bit aligned (!)
First one is missing half of the telegram. What do you use to receive those? Maybe @weetmuts can comment on this?

kitor, 2022-09-21:
@mmalina This shorter telegram is actually correct. I started investigation of #612, and it turs out that 
`B60AFFFFF2460186EC1B934E` is part that sometimes does exist, sometimes it doesn't on apator162 water meter. Size matches, on water meters this part also starts with `B60AFFFFF`.

I'll try to implement that later today.

Third format requires some more investigation.

mmalina, 2022-09-21:
@kitor I believe you meant to tag @mmmalina and not me.

kitor, 2022-09-21:
Yep, sorry for that :)

kitor, 2022-09-21:
Okey, I understand at least a part of this B6 header. B6 is followed by number of header bytes.

1st example contains no B6 header at all.
2nd contains 0xA which checks out for `FFFFF2460186EC1B934E`
3rd contains 0xE, which checks out for `FFFFF1460186EC1B934EE91BA57B`

All three cases end with "expected" `A0A10000590000000100322DE413B415`

Whatever this part means I don't think it matters. So I'll implement just a way to skip it.

kitor, 2022-09-21:
@weetmuts since `B6` is `b6 tpl-ci-field (Mfct specific)`, shall this be somehow implemented in driver or elsewhere? `B6` itself is missing in extracted payload.

mmmalina, 2022-09-22:
Hello Kitor,

I am using the MegaCUL project on atmega1284p and the CC1101 receiver to receive telegrams. This hardware can receive long telegrams. After your update, it still decodes a long telegram with incorrect data "0".

e.g:
29441486********0408B60EFFFFF1460186EC1B934EE91BA57BA0A10000590000000100322DE413B415

    "media":"heat cost allocation",
    "meter":"apatoreitn",
    "name":"",
    "id":"********",
    "current_hca":0,
    "previous_hca":0,
    "current_date":"",
    "season_start_date":"",
    "esb_date":"",
    "temp_room_avg_c":0,
    "temp_room_prev_avg_c":0,
    "timestamp":"2022-09-22T22:43:12Z"

And I save these incorrect data in domoticz logs.

Tomorrow I will do tests to see if the RTLWMBUS receiver also receives these three telegrams.

kitor, 2022-09-23:
> After your update, it still decodes a long telegram with incorrect data "0".

Yes, I don't know how to return with "error" state from decoding function. I started using wmbusmeters only about a month ago, I don't know the internals...

Anyway, this won't be fixed until @weetmuts reply to my B6 tpl-ci-field question, as implementing this header support is the proper solution for different telegram lengths (and would also benefit apator162 and maybe other apator devices)

weetmuts, 2022-09-24:
>@weetmuts since B6 is b6 tpl-ci-field (Mfct specific), shall this be somehow implemented in driver or elsewhere?
>If this should be implemented in driver, how can I access it from Telegram, since it is missing in extracted payload?

Yes, from within processContent(Telegrm *t) it is available as t->tpl_ci 

What are the other expected bytes at that position? We have to check that the other ones also count as manufacturer specific.

kitor, 2022-09-24:
> Yes, from within processContent(Telegrm *t) it is available as t->tpl_ci

Great, that should be enough to handle this properly.

> What are the other expected bytes at that position? 

I never seen anything else than `B6`. In case of `apatoreitn` you either have `B6<header_len><header><sensor_data>` or nothing (starts with sensor data `A0A1...`). where sensor data is last 16 bytes of a telegram.

In case of this "new" `apator162` format, all telegrams I got so far started with `B6` and header so far always had the same length.

Unless `A0A1` has some special meaning -> so far I assumed this *may* be season start date, as this is the only non-zero field that had no clear meaning in telegram, and taken with a gain of salt (with no byte swapping) yielded correct date in case of my sensors.

kitor, 2022-09-27:
@mmmalina please check my latest commit, it should work with all message lengths.

kitor, 2022-09-27:
#617 was merged, so I'm closing the issue :)

closed 2022-09-27 by kitor