wmbusmeters/wmbusmeters #603: Driver for Multical 62

TorbenHansen, 2022-09-05:
I have a Multical 62 with WMbus.  

When i set wmbusmeters in Home Assistant to driver:auto, I get the following.
[image]

is it possible to make the multical 62 work in home assistant.

here is the pdf on the wmbus 
[58101075_GB.pdf](https://github.com/weetmuts/wmbusmeters/files/9491760/58101075_GB.pdf)

weetmuts, 2022-09-05:
Yes, please provide a long and a short telegram from your device. The look like 
telegram=....
in the debug log.

TorbenHansen, 2022-09-05:
Is this useable?...In the first picture there is 2 lines called telegram..with the meter number 78489982. 

[image]

[image]

TorbenHansen, 2022-09-06:
**Here is the debug log i plain text:**

(meter) UdendørsHane: for me? 78489982 in 78489982
(meter) UdendørsHane: yes for me
(meter) UdendørsHane(3) auto  handling telegram from 78489982
(meter) UdendørsHane 78489982 "23442D2C8299487833168D204B5C4185220319B525E1C9380BA3ED28484157911064EC2C"
(wmbus) parseDLL @0 36
(telegram) DLL L=23 C=44 (from meter SND_NR) M=2c2d (KAM) A=78489982 VER=33 TYPE=16 (Cold water meter) (driver unknown!) DEV=amb8465[00000000] RSSI=-52
(wmbus) parseELL @10 26
(ELL) decrypting "0319B525E1C9380BA3ED28484157911064EC2C"
(ELL) IV 2D2C829948783316205C418522000000
(ELL) block 0 block_size 16 offset 0
(ELL) decrypted "8F1D79ABB3DDF5D99600000000000000"
(ELL) block 1 block_size 3 offset 16
(ELL) decrypted "005046"
(ELL) decrypted "8F1D79ABB3DDF5D99600000000000000005046"
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=4b SN=5c418522 (AES_CTR session=12 time=2642965) CRC=8f1d
(wmbus) parseNWL @19 17
(wmbus) parseAFL @19 17
(wmbus) parseTPL @19 17
(dvparser) found remembered format for hash b3ab
(dvparser) using format "041402FD17043B0259"
(telegram) TPL CI=79
(auto) 000   : 23 length (35 bytes)
(auto) 001   : 44 dll-c (from meter SND_NR)
(auto) 002   : 2d2c dll-mfct (KAM)
(auto) 004   : 82994878 dll-id (78489982)
(auto) 008   : 33 dll-version
(auto) 009   : 16 dll-type (Cold water meter)
(auto) 010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(auto) 011   : 20 ell-cc (slow_resp sync)
(auto) 012   : 4b ell-acc
(auto) 013   : 5c418522 sn (AES_CTR)
(auto) 017   : 8f1d payload crc (calculated 8f1d OK)
(auto) 019   : 79 tpl-ci-field (EN 13757-3 Application Layer with Compact frame (no tplh))
(auto) 020   : abb3 format signature
(auto) 022   : ddf5 data crc
(auto) 024 C?: D9960000
(auto) 028 C?: 0000
(auto) 030 C?: 00000000
(auto) 034 C?: 5046
(shell) exec "/bin/sh"
(shell) arg "-c"
(shell) arg "/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON""
(shell) env "METER_JSON={"media":"cold water","meter":"auto","name":"UdendørsHane","id":"78489982","timestamp":"2022-09-06T07:27:49Z","device":"amb8465[00000000]","rssi_dbm":-52}"
(shell) env "METER_ID=78489982"
(shell) env "METER_NAME=UdendørsHane"
(shell) env "METER_MEDIA=cold water"
(shell) env "METER_TYPE=auto"
(shell) env "METER_TIMESTAMP=2022-09-06T07:27:49Z"
(shell) env "METER_TIMESTAMP_UTC=2022-09-06T07:27:49Z"
(shell) env "METER_TIMESTAMP_UT=1662449269"
(shell) env "METER_TIMESTAMP_LT=2022-09-06 09:27.49"
(shell) env "METER_DEVICE=amb8465[00000000]"
(shell) env "METER_RSSI_DBM=-52"
(shell) waiting for child 10316 to complete.
(shell) /bin/sh: return code 0

TorbenHansen, 2022-09-15:
(telegram) DLL L=23 C=44 (from meter SND_NR) M=2c2d (KAM) A=78489982 VER=33 TYPE=16 (Cold water meter) (driver unknown!) DEV=amb8465[00000000] RSSI=-120

(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=cb SN=dc9a8822 (AES_CTR session=12 time=2656685) CRC=c96e

(telegram) TPL CI=79

These are the 3 lines with telegram in it.

TorbenHansen, 2022-09-15:
The above telegrams is from here.

(meter) UdendørsHane: for me? 78489982 in 78489982
(meter) UdendørsHane: yes for me
(meter) UdendørsHane(3) auto  handling telegram from 78489982
(meter) UdendørsHane 78489982 "23442D2C8299487833168D20CBDC9A88227C51934B1D0662603DD1A5BC2FD79B03CBABA3"
(wmbus) parseDLL @0 36
(telegram) DLL L=23 C=44 (from meter SND_NR) M=2c2d (KAM) A=78489982 VER=33 TYPE=16 (Cold water meter) (driver unknown!) DEV=amb8465[00000000] RSSI=-120
(wmbus) parseELL @10 26
(ELL) decrypting "7C51934B1D0662603DD1A5BC2FD79B03CBABA3"
(ELL) IV 2D2C82994878331620DC9A8822000000
(ELL) block 0 block_size 16 offset 0
(ELL) decrypted "C96E79ABB3DDF1DB9600000000000000"
(ELL) block 1 block_size 3 offset 16
(ELL) decrypted "005046"
(ELL) decrypted "C96E79ABB3DDF1DB9600000000000000005046"
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=cb SN=dc9a8822 (AES_CTR session=12 time=2656685) CRC=c96e
(wmbus) parseNWL @19 17
(wmbus) parseAFL @19 17
(wmbus) parseTPL @19 17
(dvparser) found remembered format for hash b3ab
(dvparser) using format "041402FD17043B0259"
(telegram) TPL CI=79
(auto) 000   : 23 length (35 bytes)
(auto) 001   : 44 dll-c (from meter SND_NR)
(auto) 002   : 2d2c dll-mfct (KAM)
(auto) 004   : 82994878 dll-id (78489982)
(auto) 008   : 33 dll-version
(auto) 009   : 16 dll-type (Cold water meter)
(auto) 010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(auto) 011   : 20 ell-cc (slow_resp sync)
(auto) 012   : cb ell-acc
(auto) 013   : dc9a8822 sn (AES_CTR)
(auto) 017   : c96e payload crc (calculated c96e OK)
(auto) 019   : 79 tpl-ci-field (EN 13757-3 Application Layer with Compact frame (no tplh))
(auto) 020   : abb3 format signature
(auto) 022   : ddf1 data crc
(auto) 024 C?: DB960000
(auto) 028 C?: 0000
(auto) 030 C?: 00000000
(auto) 034 C?: 5046
(shell) exec "/bin/sh"
(shell) arg "-c"
(shell) arg "/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON""
(shell) env "METER_JSON={"media":"cold water","meter":"auto","name":"UdendørsHane","id":"78489982","timestamp":"2022-09-15T20:06:25Z","device":"amb8465[00000000]","rssi_dbm":-120}"
(shell) env "METER_ID=78489982"
(shell) env "METER_NAME=UdendørsHane"
(shell) env "METER_MEDIA=cold water"
(shell) env "METER_TYPE=auto"
(shell) env "METER_TIMESTAMP=2022-09-15T20:06:25Z"
(shell) env "METER_TIMESTAMP_UTC=2022-09-15T20:06:25Z"
(shell) env "METER_TIMESTAMP_UT=1663272385"
(shell) env "METER_TIMESTAMP_LT=2022-09-15 22:06.25"
(shell) env "METER_DEVICE=amb8465[00000000]"
(shell) env "METER_RSSI_DBM=-120"
(shell) waiting for child 2204 to complete.
(shell) /bin/sh: return code 0

TorbenHansen, 2022-09-16:
Here is the telegrams I found. Hope they can be used.

Received telegram from: 78489982
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Cold water meter (0x16)
                   ver: 0x33
                device: amb8465[00000000]
                  rssi: -88 dBm
                driver: unknown!
telegram=||28442D2C8299487833168D203104F2882237EAA46F1BCC844982ACADD7B84BF60708311F4911A5ECE3|+54

Received telegram from: 78489982
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Cold water meter (0x16)
                   ver: 0x33
                device: amb8465[00000000]
                  rssi: -117 dBm
                driver: unknown!
telegram=||23442D2C8299487833168D20333CF2882206A5F5B7F00B0483C3D3A0957ED2512FF105AA|+120

TorbenHansen, 2022-09-20:
pi@raspberrypi:~/GIT/wmbusmeters $ wmbusmeters --analyze simulation_udendors.txt Wasser multical21 7XXXXXX2 9XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX6
000   : 28 length (40 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 2d2c dll-mfct (KAM)
004   : 82994878 dll-id (7XXXXXX2)
008   : 33 dll-version
009   : 16 dll-type (Cold water meter)
010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
011   : 20 ell-cc (slow_resp sync)
012   : 31 ell-acc
013   : 04f28822 sn (AES_CTR)
017   : 5ce6 payload crc (calculated 5ce6 OK)
019   : 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
020   : 04 dif (32 Bit Integer/Binary Instantaneous value)
021   : 14 vif (Volume 10⁻² m³)
022 C!: DB960000 total consumption (386.190000 m3)
026   : 02 dif (16 Bit Integer/Binary Instantaneous value)
027   : FD vif (Second extension FD of VIF-codes)
028   : 17 vife (Error flags (binary))
029 C?: 0000
031   : 04 dif (32 Bit Integer/Binary Instantaneous value)
032   : 3B vif (Volume flow l/h)
033 C!: 00000000 max flow (0.000000 m3/h)
037   : 02 dif (16 Bit Integer/Binary Instantaneous value)
038   : 59 vif (Flow temperature 10⁻² °C)
039 C!: 5046 flow temperature (180.000000 °C)
No automatic driver selection could be found based on mfct/type/version!
The driver which matches most of the content is multical21 with 10/12 content bytes understood.
000   : 23 length (35 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 2d2c dll-mfct (KAM)
004   : 82994878 dll-id (7XXXXXX2)
008   : 33 dll-version
009   : 16 dll-type (Cold water meter)
010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
011   : 20 ell-cc (slow_resp sync)
012   : 33 ell-acc
013   : 3cf28822 sn (AES_CTR)
017   : c96e payload crc (calculated c96e OK)
019   : 79 tpl-ci-field (EN 13757-3 Application Layer with Compact frame (no tplh))
020   : abb3 format signature
022   : ddf1 data crc
024 C!: DB960000 total consumption (386.190000 m3)
028 C?: 0000
030 C!: 00000000 max flow (0.000000 m3/h)
034 C!: 5046 flow temperature (180.000000 °C)
No automatic driver selection could be found based on mfct/type/version!
The driver which matches most of the content is multical21 with 10/12 content bytes understood.
pi@raspberrypi:~/GIT/wmbusmeters $ 

I can see that the Multical21 driver is almost correct 10/12. Is this any help

weetmuts, 2022-09-21:
Thanks, but I need the telegram=|....| when you have used the key.
The previous telegram=|| that you posted are still encrypted.

TorbenHansen, 2022-09-25:
Is this any help?

**Short telegram**
https://wmbusmeters.org/analyze/23442d2c8299487833168d20333cf2882206a5f5B7f00B0483c3d3a0957ed2512ff105aa:auto:93757B0A070B7F34282CEBCD524426C6 

**Long Telegram**
https://wmbusmeters.org/analyze/28442d2c8299487833168d203104f2882237eaa46f1Bcc844982acadd7B84Bf60708311f4911a5ece3:auto:93757B0A070B7F34282CEBCD524426C6

closed 2023-04-05 by github-actions[bot]